FreeCAD/FreeCAD #9432: [Feature Request] working with GIT with unzipped source code for long-term maintenance

nex0ma, 2023-04-28:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecad.org/search.php?keywords=git

### Subproject(s) affected?

Core

### Idea description

working with GIT with unzipped source code for long-term maintenance

the development of parts and the process of their processing is no less different than the development of programs.

I propose the idea of versioning FREECAD projects in version control systems, for example, git . This will lead to a better process for production.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

GreenCappuccino, 2023-04-30:
Consider taking a look at the [WebTools workbench](https://wiki.freecad.org/WebTools_Workbench). It has [Git support](https://wiki.freecad.org/WebTools_Git).

I think that this solution should work pretty well for most people. FreeCAD files don't really benefit from certain version control features like merging, so saving full copies of them should be fine.

If you really want to have more fine-grained version control for some reason, I think that realthunder's fork of the codebase has some support for saving FreeCAD files uncompressed, which can then be version-controller. There's a forum post mentioning it [here](https://forum.freecad.org/viewtopic.php?t=38353) if you want to look further.

luzpaz, 2023-04-30:
Im sure we have another feature request for this but where is it ?

yorikvanhavre, 2023-05-03:
In Realthunder's branch there is I think the possibility to save a FreeCAD file as an unzipped directory... I found the implementation a bit weird though, but I don't remember exactly why. Might be good to have a new look at it

luzpaz, 2023-05-03:
@realthunder hi! Do you mind sharing where your git related file saving functionality is located in Link ? TIA!

Jookia, 2023-05-08:
I would like to see this. A zip file is just a collection of compressed files just like a git repository. Git having the advantage of being able to check out older versions due to deduplication is very helpful too.

yorikvanhavre, 2023-05-08:
Indeed the task of managing files and syncing them to remote services is something that can have a lot of different forms (git, dropbox, or a plethora of private corporate server systems), and it's unrealistic to try to manage that all with FreeCAD.

However I think it makes sense to be able to save your FreeCAD work in a manner that is more friendly to text-based version control systems such as Git. And since a .FCStd file is a zipped folder containing mostly text files, it would make sense to be able to save it as an unzipped folder. For me the option could co-exist besides the .FCStd format.

Now in an ideal world, git and other VCS would seamlessly handle zip files, and everything would be solved :)

nex0ma, 2023-05-08:
save as unzipped in menu?

realthunder, 2023-05-10:
The original patch for saving and open unzipped directory is https://github.com/realthunder/FreeCAD/commit/1dfd9b93c9f. There are a few changes here and there after that. I actually included most of the changes in the toponaming branch.

yorikvanhavre, 2023-05-11:
What about this:

* In the Save/Save as dialog, in the file type drop down, below "FreeCAD Document (*.FCStd)" you would also have "FreeCAD unzipped folder (*.FCStd.d)" (.d folders is common on linux, not sure if there is something better out there? .FCDir? Unfortunately it's not common to have such extensions on folder names, but why not?)
* In the Open/Insert dialog, you could select a .FCStd.d/.FCDir folder too, or the Document.xml inside, and it would seamlessly open

nex0ma, 2023-05-11:
if it does not disrupt the flow of the document

1. menu - save - save unzipped, but in work we must remember that 'save' is 'save unzipped'

2. menu - 'save unzipped', menu - 'load unzipped'      **it is easier**

3. more complex. separate menu for working with version control systems. NO NO NO!

4. about names of unzipped dirs - can as it is, only the name without the extension

ezorzin, 2024-01-02:
I think the "save unzipped - load unzipped" could be a simple and good strategy. Maybe it is possible to create a Freecad plugin doing this? I believe that could be enough for those who want to have a git repository for their projects, and it would not require modifications in the mainstream development of Freecad.

recursivenomad, 2024-07-10:
I also support "save unzipped" and "open unzipped".

Alternatively, it could be nice to simply have the FCStd file format be accepted as a zip with zero compression, and having a setting to save uncompressed files.

For reference, [Blender offers such an option](https://docs.blender.org/manual/en/latest/editors/preferences/save_load.html) ("Compress File").

nex0ma, 2024-07-10:
there's also the aspect of using Freecad for use in neural network training when someone's thought about it comes up.

a) part description text or drawing -> neural network -> 3D model -> neural network -> CNC program
b) photo or photos -> neural network -> drawing

johnsonm, 2024-07-10:
I think that "unzipped" needlessly foists internals on the UX. "Save as directory" as @realthunder called it is user-centric wording.

I was around for creating the "`.d`" pattern in Linux and it was intended for configuration snippets so that we could get out of the business of having tools edit configuration files. It hasn't really been a pattern for user programs that I've been aware of.

If you want a standardized extension for directories, I would suggest `.FCdir` — then whatever pattern is chosen for backup files, make sure it's something that's trivial to put into `.gitignore`  so we can just `git add .` ☺

ezorzin, 2024-07-11:
Maybe a possibility is to modify the default behaviour of Freecad and simply make it saving its files as plain, uncompressed, human readable text in a project folder, as it already is before the compression.
In the electronics sector, there is [Kicad](https://www.kicad.org/): they do exactly that and everybody can happily manage the versioning of the projects by using Git (or similar methods).

Do we really need a Freecad compressed folder by default?
In case somebody really need it, everybody is eventually able to compress it by their own.

ezorzin, 2024-07-11:
P.S. a possible implementation of the solution I am describing could be to add a checkbox in the "Settings" of Freecad, something that can enable the user to select the compressed/uncompressed behaviour.

johnsonm, 2024-07-11:
@ezorzin:

> Maybe a possibility is to modify the default behaviour of Freecad and simply make it saving its files as plain, uncompressed, human readable text in a project folder, as it already is before the compression.

Well, that's what realthunder's save as directory *is,* ignoring the point of whether it is the default behavior.

Making this a configuration for what "save" means would hurt interoperability, bug reporting, etc. I would expect that to be a bridge too far right now.

Making it a run-time option that people can choose one way or another depending on what they are working on seems to me like a small change in UX for existing FreeCAD users.

------------------------------------------------------------------------------

Incidentally, folks in general (reasonably!) keep thinking that Git is a "source code control" tool, and forget that it is a versioned filesystem that has an (optional to use) source code control layer on top of it. So they keep worrying about whether divergence can be merged (focusing on the source code control side) as *sine qua non* for git being useful for FreeCAD files. But that's not so. When Linus first released git, it didn't even have much in the way of source code control capacity. It just managed relationships between versioned states. And the versioned filesystem part is very useful even without feature-centric merging.

XML isn't human-centric but it's human-readable, and I've used `git diff` on "save as directory" in realthunder's fork plenty of times to remind myself what I changed in order to write a useful commit message for myself.

recursivenomad, 2024-07-12:
@johnsonm:

> Making this a configuration for what "save" means would hurt interoperability, bug reporting, etc. I would expect that to be a bridge too far right now.

For interoperability, I really want to reemphasize doing what (I think) Blender does - still saving as a file with .FCStd extension, but simply applying 0 compression to the zip.  That way, there is full parity/interoperability between compressed and non-compressed files, bit it's trackable in VCS. *(Also would be great to be able to disable B-rep export when saving uncompressed)*

I have manually extracted my .FCStd contents, deleted all files except for `Document.xml` and `GuiDocument.xml`, and rezipped them with 0 ("store") compression.  Fully legible file contents (after a recompute), impressively portable file size, and fully deterministic (at least for my parametric workflow) when recomputed!

johnsonm, 2024-07-12:
@recursivenomad, oh, just have the option of setting the compression level in configuration! I had missed that, even though you said it. I have used .fodt ("flat xml document") with libreoffice exactly the same way, where `git diff` helped me do a good commit message and I was never trying to `git merge`.

I no longer remember whether it was the forum or a github issue where the idea of leaving out recomputable elements was discussed in detail and rejected.

I guess that with my `.FCDir` you could `.gitignore` the recomputable elements and they would still be on the filesystem for normal use, and you could take the need to recompute as your own choice for what you commit to source code control and just expect to have to recompute on load in the rare case of reverting or using a remote repository as backup?

recursivenomad, 2024-07-12:
> I guess that with my `.FCDir` you could `.gitignore` the recomputable elements and they would still be on the filesystem for normal use, and you could take the need to recompute as your own choice [...]

Oh, wow - yes, directories offering granular per-project control via `.gitignore` is something that did not occur to me!  That's a very nice prospect.  I've been playing with writing a `pre-commit` githook to prune what I do not need and repackage the file with zero compression, committing the slim diff-able file, and then restoring the full original file to the working tree... but after the many headaches associated with such an obscure workflow, the prospect of a simple `.gitignore` acting on a directory structure is *really* appealing now 😅

That said, I think a zero-compression (and in my opinion, without B-reps) option is some very low-hanging fruit!

johnsonm, 2024-07-12:
A nice thing about save-as-directory / ".FCdir" is that the names are visible in the `git diff` and those can be correlated without too much trouble to the model. A zero-compression ZIP file will still have a binary table of contents, which will make the `git diff` for writing log messages less useful.

I suspect that configuring zero compression ZIP as a configuration will get a lot further quicker than FreeCAD adding a feature for dropping the BREPs from its output (either in ZIP or save-as-directory); see @yorikvanhavre's [comments on dropping BREPs](https://github.com/FreeCAD/FreeCAD/issues/11936#issuecomment-1887462554).

Then the question of not storing the BREPs could be left out of FreeCAD entirely; the filesystem would become a *cache* for them if you chose not to commit them. Personally, I'd almost certainly say "meh, disk space is cheap" and commit the BREPs to the repository. You could do differently. And FreeCAD wouldn't need any explicit support for the difference.

recursivenomad, 2024-07-14:
I definitely agree that a directory has some nice developer-facing advantages.  The only disadvantages of a directory structure I see would be file size (again, mostly the B-reps) and portability (directory contents can be separated).

In a [highly parametric project](https://gitlab.com/recursivenomad/ki-lime-pi-pico/-/blob/main/models/RaspberryPi_Pico_Variations.FCStd) I'm working on with a lot of geometric variants, the B-reps inflate an otherwise 7 MB decompressed directory to 105 MB (!)  So for sufficiently complex (parametric) projects, I still think there's a case for excluding B-reps when considering uncompressed storage.

In the meantime, I managed to get that `pre-commit` git hook I was working on to be functional:

<details> <summary> pre-commit git hook bash script </summary>

```bash
#!/bin/bash

#  SPDX-License-Identifier: MIT-0 OR CC0-1.0  #
#     Authored 2024 by Peter S. Hollander     #

# Pre-commit git hook for committing reduced, usable, & trackable .FCStd files
# ----------------------------------------------------------------------------

#   This was made primarily with parametric PartDesign workflows in mind.
#   It "should" work out-of-the-box on all operating systems.

#   Recomputing from reduced .FCStd files appears to non-deterministically
#   regenerate their structure, essentially costing a commit the size of the
#   entire file if you commit after a checkout/clone.

#   Continuously committing from the same non-reduced local source document
#   does not encounter this issue.

# Single-colour file exclusion:
#   Uncomment if you want to exclude (hopefully) recomputable colour files

#ExcludeSingleColourFiles=true

# Total file exclusion:
#   Only uncomment file types you're 100% confident you don't need tracked!
#   For large files, you may be forced to exclude most file types, as
#   `git archive` will approach an argument list limit during .FCStd repacking.

ExcludeFiles=""

ExcludeFiles+=" *.brp"
#ExcludeFiles+=" PointColorArray*"
#ExcludeFiles+=" LineColorArray*"
#ExcludeFiles+=" DiffuseColor*"
#ExcludeFiles+=" ScaleList*"
#ExcludeFiles+=" PlacementList*"

# --show-toplevel should throw an error if there is no working tree (untested)
GitRoot=$(git rev-parse --show-toplevel)

# Get an array of the paths of each staged file
StagedFiles=($(git diff --cached --name-only))

FCStdErrorCount=0
for file in "${StagedFiles[@]}"; do
  # For every .FCStd staged for commit:
  if [[ "${file^^}" == *\.FCSTD ]]; then
    #echo "Identified staged" $file

    if [[ ! `git rev-parse --verify HEAD 2>&-` ]]; then
      echo "FCStd pre-commit hook cannot find HEAD! Unstage .FCStd before making an initial commit."
      (( FCStdErrorCount+=1 )); continue
    fi

    if [[ ! -s "${GitRoot}"/"${file}" ]]; then
      echo $file "does not exist - assuming this is a deletion commit."
      continue # No fixup necessary
    fi

    #echo "Saving temporary copy of unmodified .FCStd"
    mv "${GitRoot}"/"${file}" "${GitRoot}"/"${file}".bak
[... 65 lines cut ...]
  fi

done

if (( FCStdErrorCount > 0 )); then
  echo ".FCStd pre-commit hook encountered" "${FCStdErrorCount}" "error(s); aborting commit"
  exit 1
fi

if [[ `git rev-parse --verify HEAD 2>&-` ]]; then
  # HEAD exists
  if [[ ! `git diff-index --cached HEAD` ]]; then
    echo "No changes to commit."
    exit 1
  fi
fi

exit 0
```

</details>

It outputs a zero-compression .FCStd upon commit with various files excluded based on the top configuration variables, but preserves the original non-reduced default-compression file locally so you don't need to recompute every commit.

I'm not very experienced in writing bash scripts, but this solution seems to suit my use-case well enough - I don't know if it will be much help to anyone outside of a PartDesign workflow, but feel free to make use of it as you please :)

yorikvanhavre, 2024-07-15:
Don´t forget that anything that is not parametric (ex. imported STEP files) do need the Brep files. If the Brep file is not there, the object has no shape anymore even after recomputing. So it's a tricky game to determine which Breps are needed and which aren't.

johnsonm, 2024-07-15:
@recursivenomad any reason not to use `zip -0` to create the archive? (I see you mention an argument list limit. I'd think that you could pipe the list of names to `zip -0 -@` to avoid creating the git commit.)

@yorikvanhavre  I wouldn't delete any BREP files myself, but now I'm curious... Would keeping `.brp` files that don't have a corresponding `Map.txt` file answer that need? I can see that there **are** cases of a brp without a Map.txt that are parametric; e.g. chamfers. But does any **non-parametric** object have a corresponding Map.txt file?

yorikvanhavre, 2024-07-15:
The Map.txt is if I'm not mistaken the TNP's element map. It is not a good indicator, as you can have parametric objects that don't have an element map. Hopefully that'll vanish in the future, but then at that time you could also have "dumb" imported shapes with an element map.

I see no real way to determine this from the outside. If we want to pursue that path, I think a way to "mark" objects with "disacardable" shapes will be needed

recursivenomad, 2024-07-15:
> @recursivenomad any reason not to use `zip -0` to create the archive? (I see you mention an argument list limit. I'd think that you could pipe the list of names to `zip -0 -@` to avoid creating the git commit.)

@johnsonm yes, I wanted this to work "out-of-the-box" on most platforms, and Git Bash on Windows does not appear to have the `zip` command 🙃  `git archive` was the only default zipping option I could find.  Also, the reason I run into the argument list limit is because it would seem FreeCAD looks for `Document.xml` at the top of the archive, but `git archive` sorts alphabetically, so by default it will place `DiffuseColor` at the top of the archive.  So to get around this, I can only tell it to zip the .xml files, and then append all others so they are listed below Document.xml in the archive.  I'll admit it is janky, but it works "enough" !

> Don´t forget that anything that is not parametric (ex. imported STEP files) do need the Brep files.

@yorikvanhavre I almost exclusively work in PartDesign workflows, and work that I version-control via git tends not to distribute imported assets for copyright reasons.  That said, I am certainly noting this warning for the future!  Is there a way to have FreeCAD reference an external file to be able to regenerate the geometry from, rather than embedding it in the B-reps of the archive itself?

ezorzin, 2024-07-15:
zips with zero-compression: never heard about this, there is always something new to learn! 😅

yorikvanhavre, 2024-07-15:
@recursivenomad yes, you could do that with AppLinks or BIM References... But it would make the whole workflow heavier than it should be... Plus, you'd need to version those files too anyway, it wouldn't really be a big advantage

recursivenomad, 2024-07-15:
@yorikvanhavre the lightweight repository size advantage of being able to version the recomputable zero-compression .FCStd far outweighs the benefits of a file with pre-computed B-reps for me (again, exclusively PartDesign workflow here).  AppLinks sound like a good place to start if I ever need a non-parametric reference, thank you for the suggestion!

johnsonm, 2024-07-16:
Wait, how long has this been here?

[image]

I've been hoping for something that, and completely missed that it exists. I think it would be awesome some day for it to be a default for new documents (or documents not expressing the preference) and then the general Document preference would be when creating a new file (or adding it to an existing file).

@nex0ma git should actually handle this setting very well. There is still a small binary header at the beginning and table of contents at the end, so you definitely don't want to tell git that they are text files, but there's a way to see changes.

```
echo '*.FCStd diff=strings' >> .gitattributes
git config --global diff.strings.textconv strings
```

Now `git diff` looks meaningful. E.g. "I renamed this object":

[image]

Not all of the diff is useful for helping write a `git commit` message (e.g. the .brp files are not so human-readable) but it's better than nothing, and it will let git do its own compression nicely.

recursivenomad, 2024-07-16:
@johnsonm 😱

Wow!  In hindsight I totally remember seeing this setting when first setting up FreeCAD!  Thank you so much for re-discovering and sharing it!

The diffs are both helpful for identifying what changed, and it makes for very lightweight version control, allowing the tracking of (mostly) only what changes between small edits.  I'll still be using the pre-commit hook above for stripping B-reps in my parametric projects, but I'm really glad to see zero-compression already integrated in FreeCAD 😊

Great find! 🎉

ezorzin, 2024-07-16:
@johnsonm nice finding!

Could you please better explain your method? As far as I understood, Freecad normally uses a compressed file format: basically a folder with many things in it, then compressed into a single file. But you found an option for setting the compression level to zero: therefore you get a single file, but being not compressed it is still somehow text readable? And this plays nice with git?

Am I understanding correctly?

johnsonm, 2024-07-16:
@ezorzin A zip file, which is what FreeCAD uses (with further restrictions, at least involving which file is the first file in the archive) can have multiple compression levels. Level 0 is **not compressed at all.** There is still a binary header at the front, and a binary table of contents at the end, but in between are the individual files. The `strings` command prints out everything in a file that could be a text string. My suggestion configures git to use the `strings` command to compare the text parts of the contents.

When Git "packs" files, its compression is applied to multiple file versions together, which means that unmodified parts of files share most of their storage space between versions. When you check in a ZIP file, the compression is re-started for every file included in the archive. This means that git's compression can be effective at not storing duplicate copies of files that don't change. But the internal compression of each file in the zip archive means that it can't take much advantage of that storage optimization for files that only changed a little bit. Storing them in the ZIP uncompressed means that git can take advantage of the similarity between versions of the individual files in the ZIP, substantially reducing the total storage required for the git repository.

Similarly, if FCStd files are stored uncompressed and then compressed with ZIP (foo.FCStd compressed into foo.FCStd.zip), they will be smaller than if the FCStd was stored with compression, because the compression can take advantage of self-similarity between flies in the archive, which it can't do with a normal ZIP file because each included file is compressed individually.

dgalli1, 2024-10-25:
> Wait, how long has this been here?

> ```
> echo '*.FCStd diff=strings' >> .gitattributes
> git config --global diff.strings.textconv strings

Nice find this seems to work perfectly. I made a quick bash script to create a GitHub repository and set it up for FreeCAD. Might be useful for some of you:
https://gist.github.com/dgalli1/5d34b322b6e120ae383ef3a9c5f89c2b

Mikestriken, 2025-11-09:
> What about saving as [TAR](https://en.wikipedia.org/wiki/Tar_(computing)#File_format) ? It's a unique file but uncompressed so it has a text content which is version-control-compatible.

Simply set the compression level to 0, [as was mentioned earlier](https://github.com/FreeCAD/FreeCAD/issues/9432#issuecomment-2229790551), and then the same thing is accomplished