FreeCAD/FreeCAD #9435: [Feature Request] Transform should be available for all objects having a Placement or an AttachmentOffset

PLChris, 2023-04-29:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=77980#p678705

### Subproject(s) affected?

Core

### Idea description

The Transform tool should be applicable to all objects having a Placement, especially to sketches.
If an object is attached, the Transform tool should work on the AttachmentOffset.

According to forum user mfro there are already some issues, which may affect this:
https://github.com/FreeCAD/FreeCAD/issues/5546
https://github.com/FreeCAD/FreeCAD/issues/5611
https://github.com/FreeCAD/FreeCAD/issues/5527
https://github.com/FreeCAD/FreeCAD/pull/5265

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-05-02:
Technical detail: most PartDesign features do have placement properties but they are hidden, these features SHOULD NOT have the transform tool available, even when setting the default edit mode to transform

adrianinsaval, 2023-05-02:
Second detail, if I'm not mistaken the transform tools is not currently able to operate on the attachment offset property, transformations performed on attached objects are reverted upon recompute

maxwxyz, 2024-02-26:
Related: https://github.com/FreeCAD/FreeCAD/issues/11355 https://github.com/FreeCAD/FreeCAD/issues/12451
I'm going to use this issue for the comment on the FreeCAD Day 2024 Complaint Session: 
Why do we need separate transform, attachment, and placement dialogues? It should all be in the same dialogue.

maxwxyz, 2024-10-30:
@kadet1090 this cold be next after https://github.com/FreeCAD/FreeCAD/pull/17564 , for placement first :)