FreeCAD/FreeCAD #9448: [Issue]

boud123, 2023-04-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
Animation for results is not working for version 0.20.2
```

### Subproject(s) affected?

None

### Issue description

Animation for results is not working for version 0.20.2

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

d1nuc0m, 2023-04-30:
Please review your issue, there are a lot of missing information, first of all there is no title.
Then:

> ### Full version info
> 
> ```shell
> Animation for results is not working for version 0.20.2
> ```
Go to Help/About and "Copy to clipboard", it is something like this:

```
OS: Artix Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: makepkg
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.9, Qt 5.15.8, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Dracula 0.0.2
  * QuickMeasure 2022.10.28
```

> ### Issue description
> 
> Animation for results is not working for version 0.20.2

Please **add details**: What happened? Where should these animation have been? What results are you talking about?  What did you expect to happen? Etc.

adrianinsaval, 2023-05-02:
Unacceptably vague and therefore unactionable, closing. You may add appropriate information about the problem, full version info and change the title to something meaningful and then you can reopen the issue

closed 2023-05-11 by adrianinsaval