FreeCAD/FreeCAD #9449: [Issue] Text on navigation cube is very small

K4LCIFER, 2023-04-30:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: KDE Flatpak runtime (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.6, Qt 5.15.8, Coin 4.0.0, Vtk 8.2.0, OCC 7.6.3
Locale: English/Canada (en_CA)
[/code]
```

### Subproject(s) affected?

None

### Issue description

The text on the navigation cube appears to be very tiny (unreadable) as seen in the image below:
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-05-01:
@K4LCIFER, you created another issue, but the second time you don't mention your operating system. Could provide some information what you are running? What KDE are you using?

K4LCIFER, 2023-05-01:
> @K4LCIFER, you created another issue, but the second time you don't mention your operating system. Could provide some information what you are running? What KDE are you using?

Sorry about that. I updated the issue.

xtemp09, 2023-05-01:
@K4LCIFER, have your tried [this](https://github.com/FreeCAD/FreeCAD/issues/8250#issuecomment-1529111612)?

adrianinsaval, 2023-05-02:
You can change the font size in preferences, in 0.21 Thai should not be a problem anymore as the navicube has been vastly improved and now auto sizes the font. Please test the 0.21 version (it is also available in flatpak-beta repo)

Roy-043, 2023-07-16:
Closing as fixed.

closed 2023-07-16 by Roy-043