FreeCAD/FreeCAD #9459: [Problem] The user cannot set the anchor point of a Draft Shapestring intuitively.

sliptonic, 2023-05-02:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Linux Mint 20 (i3/i3)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33029 +1 (Git)
Build type: Unknown
Branch: (HEAD detached at upstream/pr/9418)
Hash: bfcc5d029f2dfbce0be1c240363ef015dbd7d3c4
Python 3.8.10, Qt 5.12.8, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
Installed mods: 
  * DynamicData 2.46.0
  * Assembly4 0.12.5
  * fasteners 0.4.55
  * Behave-Dark-Colors 0.1.1
  * 3DfindIT 1.2.0
  * Render 2023.2.2
  * ProDarkThemePreferencePack 1.0.0
  * sheetmetal 0.2.59
  * QuickMeasure 2022.10.28
  * Nodes 0.1.36
  * FeedsAndSpeeds 0.5.0
  * Assembly3 0.11.4
  * ExtremeProDark 2.6.4
  * FeedsAndSpeeds.bak
  * Curves 0.6.5
  * A2plus 0.4.60k
  * Dracula 0.0.2
[/code]
```

### Subproject(s) affected?

Draft

### Problem description

When a user creates a shapestring and sets the position to 0,0,0, it is placed like this:

[image]

The reference corner or 'anchor' is at the bottom left.  The user often wants to orient the shapestring more logically so that changes in scale are justified.  

The user should be able to select the anchor point to any of the nine points. (top, center, bottom  X  left, center, and right) .
Anchoring on center, center would cause the string center position to remain fixed in both horizontal and vertical positions as the scale is increased.

The anchor points should be reflective of the shape boundbox.  If the shape is rotated, it will not affect placement anchoring.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Roy-043, 2023-05-02:
The current alignment is not bottom left, but baseline left. IMO we should also include baseline left, center and right. And maybe "cap height" left, right and center.

onekk, 2023-05-03:
If you intend fontstring as a Shape you are right, but fonts, are usually made to have "baseline" as a reference, and it depends, on how fonts are made, as example a monospaced font is different from "variable width font".

It is not easy to set a rule, as it depends on what is the "intended behaviour", what user is expecting could be wrong, and usually is not even correct.

It is not a case that typesetting programs are so complicated, and if you make drawings with fonts, it is hard to plan "word "positions if you change text, left, right an center in horizontal and vertical positions referred to the baseline are the correct way to go, but probably is not what the user is expecting. 

If a user write "Dominion" and then he change the word in Empire, what is the predictable position, as it change the "height" of the boundingbox as the 'p' will enlarge it to the bottom, so if the user put the  baseline near the bottom of the face, the 'p' is going out of the face, same if the user will make the face to fit perfectly first world, and then it change the word using some 't' or 'p' what could be a correct "expected behaviour"?

Some documentation about this sort of things, without writing a "typesetting treatise" should warn the user about the fact that he could made some "correct assumptions" when designing things, if the text is subject to changes, his whole work could be trashed.

And this is not imputable to a fault in FreeCAD.

Regards

sliptonic, 2023-05-03:
I use f-engrave to layout engraving strings.  It lets me set basic horizontal and vertical alignment.  Yes, it's very primitive. but it's VASTLY better than the current shapestring functionality.  
I would suggest we start by improving the current mess to be more usable.  Then keep refining it until it's perfect.

onekk, 2023-05-03:
Ok probably some metrics, could be derived.

Some idea as 1pt is 1/72 of inch even some idea of height could be used to adapt at least char height to available space? and make it usable maybe setting the "boundingbox" height  to something eyeballed to available space, maybe some adjustment factor could be added, but it is only a rough idea.

Roy-043, 2023-06-10:
Attached is a first proposal.
Unpack the attached zip and place the py file in `Mod/Draft/draftobjects/` and restart FreeCAD.

The new properties can not be set via the task panel for now.

There are 9 justification options, from "Top-Left" to "Bottom-Right".

The box used for the justification depends on the `JustificationReference` parameter:
* "Capital M": Use the distance between the baseline and the top of the capital M for the height. For Left aligned text: keep the left kerning margin (the small distance between the anchor point and the first character).
* "Capital M no left margin": Idem, but remove the left 'kerning' margin.
* "Shape": Use the actual bounding box for the height.

Questions:
1. Maybe use 2 parameters for the justification (horizontal and vertical)?
2. "Top-Left", "Top Left", "Top left" or "TL"?
3. `JustificationReference` perhaps too complex?
4. How to get the right 'kerning' margin? If that is possible the "Capital M no left margin" option could be removed.

[shapestring.zip](https://github.com/FreeCAD/FreeCAD/files/11712596/shapestring.zip)

sliptonic, 2023-06-10:
This is wonderful!   The strings for the Justificationreference parameter will be a little confusing.  Tooltips would help.  But that is just picking nits.  It's really great.

Answers:
1.  It would meet all my needs as-is.  Two parameters might be needed in some edge cases but I can't think of any.
2. "Top-Left"
3. As noted above.  Tool tips go a long way to make it understandable.
4.  I've got nothing.

WandererFan, 2023-06-10:
The left side bearing (left kerning margin) can be retrieved using FreeType's ft_glyph_metrics on the first glyph.  Might need some trickery to get that value back to the python routine, but the value is easily obtained around line 186 in FT2FC.cpp.

onekk, 2023-06-12:
>     * "Capital M": Use the distance between the baseline and the top of the capital M for the height. For Left aligned text: keep the left kerning margin (the small distance between the anchor point and the first character).

About font metrics, usually the 'm' is used to be the max width, while it is difficult to have a measure of the max "bounding box" of text.

usually capital M should be slightly less than "m", but a simple test of the "bounging box" will reveal the dimensions.

In Tex "em"  that is the width of the 'm' and ex that is the dimension of the 'x' are used as "font metrics" when typesetting things, 

https://stackoverflow.com/questions/2385829/what-is-the-difference-between-px-em-and-ex

https://tex.stackexchange.com/questions/8260/what-are-the-various-units-ex-em-in-pt-bp-dd-pc-expressed-in-mm

About kerning, usually is the dedent as example when you have AV in capitals to reduce the spacing as the the shape of the two mating Characters, other kerning there are around some guides, but if you have to manage TTF it is not so easy to "place boxes" (in Tex Typesetting terminology), 

When coding I'm used to evaluate some space requirements, using as example "mmmmmmmmm" 10 'm' as they are the max width that usually a field could have, but it could result in a too conservative things, some suggest as example "mxmxmxmxmx" five times 'mx' for a ten character fields and variable width fonts, as 'x' is a decent mean value, as 'i' 'l' 'j' 't' sometimes 'f' are usually smaller in width than a 'x' but it depends heavily even on the language used. 

Sorry if this could seem unrelated.

Regards

Carlo D.

Roy-043, 2023-06-13:
Attached is a 2nd proposal.

Compared to the previous proposal several properties have been added. And the type of an existing property has been changed.

Overview of all new or changed properties:

**Justification:**
Tooltip: Horizontal and vertical alignment
9 options: "Top-Left" to "Bottom-Right"
Default: "Bottom-Left"

**JustificationReference:**
Tooltip: Height reference used for justification
2 options: "Cap Height" and "Shape Height"
Default: "Cap Height"

**KeepLeftMargin:**
Tooltip: Keep left margin and leading white space when justification is left
Default: False (when updating old objects: True)
A separate property to handle this seems clearer than more options for JustificationReference.

**ScaleToSize:**
Tooltip: Scale to ensure cap height is equal to size
Default: True (when updating old objects: False)
Currently the actual height of the text typically does not match the Size property. Depending on the font it is sometimes bigger, sometimes smaller than Size.

**Fuse:**
Tooltip: Fuse faces if faces overlap, usually not required (can be very slow)
Default: False
Not directly related to this issue. This is mainly for 'calligraphy' fonts were characters overlap intentionally.

**Tracking:**
Changed the type of this existing property to "App::PropertyDistance" so that negative values are possible.

[shapestring_20230613.zip](https://github.com/FreeCAD/FreeCAD/files/11739236/shapestring_20230613.zip)

Roy-043, 2023-06-13:
@WandererFan

It would be interesting to get the true 'typographical' width of a string. And also the internal cap height. Any white space (from 'kerning' but also from white space characters) is currently lost when alignment is Right. Probably only of importance if multiple ShapeStrings are placed one below the other (there is no multi-line option).

Roy-043, 2023-06-13:
@onekk 

The information you have given is a bit general, but please test the code if you have the time. Thanks.

sliptonic, 2023-06-22:
> Attached is a 2nd proposal.
> 
I installed the python file dated june 13.  Everything looks the same.  I don't see the fuse property and the options are as before.  Did you zip up the right one?

Roy-043, 2023-06-22:
Try again please, shapestring_20230613.zip contains the correct file.

Roy-043, 2023-06-22:
@onekk We are in a feature freeze for V0.21. You will have to use the zip file. Installation instruction as given before:

> Unpack the attached zip and place the py file in `Mod/Draft/draftobjects/` and restart FreeCAD.

sliptonic, 2023-06-22:
Indeed.  Creating a new shapestring shows the new properties and works correctly.  Previously created shapestrings don't.  

I see the logic in set_properties to add missing properties.  This function isn't firing on document restored though.  So existing shapestrings aren't getting updated.

sliptonic, 2023-06-22:
FYI,  You are more than welcome to create a PR for this.  The PR won't get merged until after V021 is released but you can still create it now

Roy-043, 2023-06-27:
> This function isn't firing on document restored though.

The code checks if a key property (Justification) is missing and only then is the `set_properties` function fired for existing objects. This should work fine when opening a file with ShapeStrings created with the current dev version. But indeed does not work for ShapeStrings created with the code from the initial proposal as they already have the Justification property.

I'll update the code to check for the existence of all added properties. It can't hurt to have that list in the code.

MantaRayDeeJay, 2023-08-09:
Feature Request Link from FreeCAD Forum

[FC v0.21.0 [Feature Request] - Draft - Shape String - Add text Alignment feature](https://forum.freecad.org/viewtopic.php?t=80305)](https://forum.freecad.org/viewtopic.php?p=698554#p698554l)

sliptonic, 2023-08-11:
@Roy-043  Please make a PR for your fix to the issue.  This is already such an improvement that I'm sad we didn't get it into 0.21

Roy-043, 2023-08-13:
I will try to do a PR this week. Note that I will probably submit the code from the 2nd proposal (shapestring_20230613.zip). You have tested that code with files created with the code from the 1st proposal. But files created with the 1st code will not exist 'in the wild'.

closed 2023-11-08 by Roy-043