FreeCAD/FreeCAD #9465: [Problem]  Transforming a plane causes FreeCAD to crash

edwilliams16, 2023-05-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.0.33040 (Git)
Build type: Release
Branch: master
Hash: 1e830a8800156097ccb3ef6124dced30d4002e76
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.0
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * DynamicData 2.46.0
  * lattice2.backup1682566094.626239 1.0.0 (Disabled)
  * fcgear 1.0.0
  * workfeature
  * Pyramids-and-Polyhedrons
  * QuickMeasure-main
  * GDML 1.8.0
  * offline-documentation
  * Manipulator 1.5.0
  * fasteners 0.4.56
  * lattice2 1.0.0
  * CurvedShapes 1.0.4
  * toSketch 1.0.1
  * Help 1.0.3
  * Curves 0.6.8
[/code]
```

### Subproject(s) affected?

None

### Problem description

Create a plane with Part|Create Primitives
Translate it  with the Transform Tool
FreeCAD crashes when the dialog is closed.
Seems to be MacOS specific.

### Anything else?

Forum link [https://forum.freecad.org/viewtopic.php?t=78077](url)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

edwilliams16, 2023-05-05:
What happens if you build it with the prebuilt
`Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.0` ?
If that breaks it, the problem wasn't in FreeCAD and presumably the next development version will be fixed.

Jolbas, 2023-05-05:
I have problems building with python 3.11 that I don't dare to handle right now. Something with shiboken.

luzpaz, 2023-05-15:
Is it Qt or VTK that is the problem ?

xtemp09, 2023-05-16:
@Jolbas, if you can reproduce the problem, you can run FreeCAD in _gdb_ or _lldb_. Could you upload the stacktrace?

edwilliams16, 2023-05-16:
Is Apple's crashlog in [https://forum.freecad.org/viewtopic.php?p=679811#p679811](url) any use?

edwilliams16, 2023-05-16:
If I'm doing this right MallocZeroOnFree seems to have no effect

```
[52714:31747:0516/082933.405987:ERROR:command_buffer_proxy_impl.cc(141)] ContextResult::kTransientFailure: Failed to send GpuChannelMsg_CreateCommandBuffer.
zsh: segmentation fault  env MallocZeroOnFree=0 ./freecad --write-log
```
`[52808:31747:0516/083317.954605:ERROR:command_buffer_proxy_impl.cc(141)] ContextResult::kTransientFailure: Failed to send GpuChannelMsg_CreateCommandBuffer.
zsh: segmentation fault  ./freecad --write-log`

edwilliams16, 2023-05-16:
Also I get the same failure in High Sierra

```
OS: macOS High Sierra (10.13)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33110 (Git)
Build type: Release
Branch: master
Hash: 18e40941ceb36c52701a7c74ea26043ef3385701
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * CurvedShapes 1.0.4
  * Curves 0.6.4
  * Curves.backup1667610597.862062 0.5.12 (Disabled)
  * DynamicData 2.46.0
  * fasteners 0.4.25
  * fcgear 1.0.0
  * GDML 1.8.0
  * Help 1.0.3
  * lattice2 1.0.0
  * Manipulator 1.4.9
  * MeshRemodel 1.8919.0
  * offline-documentation 1.0.0-alpha
  * offline-documentation.backup1668887250.997013 1.0.0-alpha (Disabled)
  * offline-documentation.backup1670387333.286728 1.0.0-alpha (Disabled)
  * Pyramids-and-Polyhedrons
  * QuickMeasure-main
  * toSketch 1.0.1
  * Trails 2022.1.0

```

xtemp09, 2023-05-17:
edwilliams16, yes, it points out the problem ― Coin3D or its usage in FreeCAD.

MattN92705, 2023-06-01:
I can confirm that this bug still exists with all versions through the latest.  Simple to reproduce - create any two bodies and translate one of them with the Transform Tool.  FreeCAD crashes when the dialog is closed.

[code]
OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 0.21.0.33345 (Git)
Build type: Release
Branch: master
Hash: a3c4f4bb01f5201daf666852444a8f9c5adbda8d
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.60n
  * fasteners 0.4.56
  * Help 1.0.3
  * Assembly4 0.50.2
  * Assembly3 0.12.0
[/code]

MattN92705, 2023-06-23:
Sounds like a Mac problem.  I've been testing this on every weekly build, and it's definitely still there.  Just create any body and translate it from the Combo View | Model pane, and *boom* crash.

[code]
OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 0.21.0.33439 (Git)
Build type: Release
Branch: master
Hash: 976ecb3754700dfe6059699b77f896f9610891e3
Python 3.11.4, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.60n
  * fasteners 0.4.56
  * Help 1.0.3
  * Assembly4 0.50.2
  * Assembly3 0.12.0
[/code]

edwilliams16, 2023-07-01:
The transform bug has gone - but if I open an FCStd file and then immediately try to save it, FreeCAD crashes. I'm awaiting feedback in the forum on this.

MattN92705, 2023-07-01:
> Interesting - I can't reproduce this.
```

OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 0.21.0.33458 (Git)
Build type: Release
Branch: master
Hash: 8b89dd29d2382c72d738c52e8974216130ffa9db
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.60n
  * fasteners 0.4.56
  * ExtremeProLight 1.6.0
  * Help 1.0.3
  * Assembly4 0.50.2
  * Assembly3 0.12.0

```

MattN92705, 2023-07-16:
The transform crash seems to have come back to life in the last couple of versions.  same old thing - create any body in Part Design, transform it from the context menu in the Model combo view, and poof.

```
OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 0.21.0.33593 (Git)
Build type: Release
Branch: master
Hash: 15371a04f38bbeb5a04edc5a91020ada1e30ef88
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.60n
  * fasteners 0.4.56
  * ExtremeProLight 1.6.0
  * Help 1.0.3
  * Assembly4 0.50.2
  * Assembly3 0.12.0

```

edwilliams16, 2023-07-16:
33515  was fixed for me.  33593 has the transform crash.

`Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Codes:       0x0000000000000001, 0x0000000000000010

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [9539]

VM Region Info: 0x10 is not in any region.  Bytes before following region: 140722985926640
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      mapped file              7ffc9f968000-7ffcaea9c000 [241.2M] r-x/r-x SM=COW  ...t_id=f74aedd1

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   libCoin.4.0.0.dylib           	       0x113b6ba90 SoCallbackAction::~SoCallbackAction() + 144
1   libCoin.4.0.0.dylib           	       0x113b6bbfe SoCallbackAction::~SoCallbackAction() + 14
2   libCoin.4.0.0.dylib           	       0x113be50c2 SoDragger::~SoDragger() + 194
3   libFreeCADGui.dylib           	       0x10e50e21e Gui::TDragger::~TDragger() + 14
4   libCoin.4.0.0.dylib           	       0x113d43aac SoBase::destroy() + 172
5   libCoin.4.0.0.dylib           	       0x113d3727a SoBaseList::truncate(int) + 74
6   libCoin.4.0.0.dylib           	       0x113d481b1 SoChildList::~SoChildList() + 17
7   libCoin.4.0.0.dylib           	       0x113d9e925 SoGroup::~SoGroup() + 37
8   libCoin.4.0.0.dylib           	       0x113dbadce SoSeparator::~SoSeparator() + 14
9   libCoin.4.0.0.dylib           	       0x113d43aac SoBase::destroy() + 172
10  libCoin.4.0.0.dylib           	       0x113d3727a SoBaseList::truncate(int) + 74
11  libCoin.4.0.0.dylib           	       0x113d481b1 SoChildList::~SoChildList() + 17
12  libCoin.4.0.0.dylib           	       0x113d9e925 SoGroup::~SoGroup() + 37
13  libCoin.4.0.0.dylib           	       0x113dbf008 SoSwitch::~SoSwitch() + 56
14  libCoin.4.0.0.dylib           	       0x113d43aac SoBase::destroy() + 172
15  libCoin.4.0.0.dylib           	       0x113d3727a SoBaseList::truncate(int) + 74
16  libCoin.4.0.0.dylib           	       0x113d481b1 SoChildList::~SoChildList() + 17
17  libCoin.4.0.0.dylib           	       0x113d9e925 SoGroup::~SoGroup() + 37
18  libCoin.4.0.0.dylib           	       0x113d8752e SoAnnotation::~SoAnnotation() + 14
19  libCoin.4.0.0.dylib           	       0x113d43aac SoBase::destroy() + 172
20  libCoin.4.0.0.dylib           	       0x113d3727a SoBaseList::truncate(int) + 74
21  libCoin.4.0.0.dylib           	       0x113d481b1 SoChildList::~SoChildList() + 17
22  libCoin.4.0.0.dylib           	       0x113d9e925 SoGroup::~SoGroup() + 37
23  libCoin.4.0.0.dylib           	       0x113dbadce SoSeparator::~SoSeparator() + 14
24  libCoin.4.0.0.dylib           	       0x113d43aac SoBase::destroy() + 172
25  libCoin.4.0.0.dylib           	       0x113d3727a SoBaseList::truncate(int) + 74
26  libCoin.4.0.0.dylib           	       0x113d481b1 SoChildList::~SoChildList() + 17
27  libCoin.4.0.0.dylib           	       0x113d9e925 SoGroup::~SoGroup() + 37
28  libCoin.4.0.0.dylib           	       0x113dbadce SoSeparator::~SoSeparator() + 14
29  libCoin.4.0.0.dylib           	       0x113d43aac SoBase::destroy() + 172
30  libCoin.4.0.0.dylib           	       0x113d3727a SoBaseList::truncate(int) + 74
31  libCoin.4.0.0.dylib           	       0x113d481b1 SoChildList::~SoChildList() + 17
32  libCoin.4.0.0.dylib           	       0x113d77215 SoBaseKit::~SoBaseKit() + 37
33  libFreeCADGui.dylib           	       0x10e51622e Gui::SoFCCSysDragger::~SoFCCSysDragger() + 14
34  libCoin.4.0.0.dylib           	       0x113d43aac SoBase::destroy() + 172
35  libFreeCADGui.dylib           	       0x10e35d260 Gui::TaskCSysDragger::~TaskCSysDragger() + 32
36  libFreeCADGui.dylib           	       0x10e35d2ee Gui::TaskCSysDragger::~TaskCSysDragger() + 14
37  libFreeCADGui.dylib           	       0x10e56ea89 Gui::TaskView::TaskView::removeDialog() + 313
38  libFreeCADGui.dylib           	       0x10e56f12a Gui::TaskView::TaskView::accept() + 202
39  libQt5Core.5.15.8.dylib       	       0x112c0cf3a 0x112ac1000 + 1359674
`
`
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33593 (Git)
Build type: Release
Branch: master
Hash: 15371a04f38bbeb5a04edc5a91020ada1e30ef88
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * DynamicData 2.46.0
  * fcgear 1.0.0
  * workfeature
  * Pyramids-and-Polyhedrons
  * QuickMeasure-main
  * GDML 1.8.0
  * offline-documentation
  * Manipulator 1.5.0
  * fasteners 0.4.56
  * lattice2 1.0.0
  * CurvedShapes 1.0.4
  * Alternate_OpenSCAD 1.0.0
  * toSketch 1.0.1
  * Help 1.0.3
  * Curves 0.6.12
`
`
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33515 (Git)
Build type: Release
Branch: master
Hash: 8d87eaac22b844105960f22817f55c8b81d51e60
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * DynamicData 2.46.0
  * fcgear 1.0.0
  * workfeature
  * Pyramids-and-Polyhedrons
  * QuickMeasure-main
  * GDML 1.8.0
  * offline-documentation
  * Manipulator 1.5.0
  * fasteners 0.4.56
  * lattice2 1.0.0
  * CurvedShapes 1.0.4
  * Alternate_OpenSCAD 1.0.0
  * toSketch 1.0.1
  * Help 1.0.3
  * Curves 0.6.12
`

edwilliams16, 2023-07-20:
0.21.0.33593   (Conda)  crashes on Transform in both the Mac-X86 and Mac-arm versions.

chennes, 2023-07-20:
I can't reproduce in a self-compiled version on my M2:
[code]
OS: macOS 13.3.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.32240 (Git)
Build type: Release
Branch: pr8826
Hash: 75808785c909acebd8a0283668ce683673f0d266
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.5.3
[/code]

edwilliams16, 2023-07-20:
@chennes  But the downloaded packages do crash?  My Mac is an M1.

chennes, 2023-07-20:
Yes, testing on the same machine but with the latest Conda build, I reproduce the crash:
```
OS: macOS 13.3.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33617 (Git)
Build type: Release
Branch: master
Hash: ae215724dd6c5336dcbb263733c6830a24917963
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
```

edwilliams16, 2023-07-20:
@Jolbas  back on May 4th also reported that he couldn't reproduce the crash with his self-compiled version.

Jolbas, 2023-07-21:
I collected versions from the Libraries tab in About FreeCAD in some downloaded freecad versions vs transform crash or ok
Stripped away libraries that didn't differ and Boost 1_74 seems suspicious.

```
ok
----------
FreeCAD 0.20.2.29603
Boost 1_81
FreeType 2.13.0
Open CASCADE Technology 7.7.0
PyCXX 7.1.7
Python 3.11.3
Qt for Python (PySide) 5.15.5
Qt 5.15.8
Salome SMESH 7.7.1.0
Qt for Python (Shiboken) 5.15.5
vtk 9.2.6
zlib 1.2.11

ok
----------
FreeCAD 0.20.2.29603
Boost 1_74
FreeType 2.12.1
Open CASCADE Technology 7.6.3
Point Cloud Library 1.12.1
PyCXX 6.2.8
Python 3.10.8
Qt for Python (PySide) 5.15.5
Qt 5.15.4
Salome SMESH 
Qt for Python (Shiboken) 5.15.5
vtk 9.1.0
zlib 1.2.13

crash
---------
FreeCAD 0.21.0.33263
Boost 1_78
FreeType 2.12.1
Open CASCADE Technology 7.7.1
Point Cloud Library 1.13.0
PyCXX 7.1.7
Python 3.11.3
Qt for Python (PySide) 5.15.8
Qt 5.15.8
Salome SMESH 9.8.0.2
Qt for Python (Shiboken) 5.15.8
vtk 9.2.2
zlib 1.2.13

ok
-------
FreeCAD: 0.21.0.33515
Boost 1_74
FreeType 2.12.1
Open CASCADE Technology 7.6.3
Point Cloud Library 1.12.1
PyCXX 7.1.7
Python 3.10.12
Qt for Python (PySide) 5.15.8
Qt 5.15.8
Salome SMESH 9.8.0.2
Qt for Python (Shiboken) 5.15.8
vtk 9.1.0
zlib 1.2.13

crash
---------
FreeCAD: 0.21.0.33626
Boost 1_78
FreeType 2.12.1
Open CASCADE Technology 7.6.3
Point Cloud Library 1.13.0
PyCXX 7.1.7
Python 3.10.12
Qt for Python (PySide) 5.15.8
Qt 5.15.8
Salome SMESH 9.8.0.2
Qt for Python (Shiboken) 5.15.8
vtk 9.2.5
zlib 1.2.13
```

0penBrain, 2023-07-23:
I made an attempt to fix here : #9975 
Hopefully will work :neutral_face:

0penBrain, 2023-07-23:
Mac users, please test when next weekly version is available and reopen issue if problem still here. Thx

adrianinsaval, 2023-07-23:
I triggered a new build now so this can be tested, since there's no caching and it's done in only two cores it'll take a couple hours but it will be up today.

adrianinsaval, 2023-07-24:
new builds are now available, please test

Jolbas, 2023-07-24:
Still crashes :(

```
crash
--------
FreeCAD: 0.21.0.33636
Boost 1_78
Coin3D 4.0.0
Eigen 3.4.0
FreeType 2.12.1
Open CASCADE Technology 7.6.3
Point Cloud Library 1.13.0
PyCXX 7.1.7
Python 3.10.12
Qt for Python (PySide) 5.15.8
Qt 5.15.8
Salome SMESH 9.8.0.2
Qt for Python (Shiboken) 5.15.8
vtk 9.2.5
Xerces-C 3.2.4
zlib 1.2.13
```

Jolbas, 2023-07-24:
Is it possible this can be an issue of Boost 1_78? I'm trying to build with this version but have failed so far to find the right combination of python linking and configuration in cmake... There is a boost@1.76 package in brew but no 1.78.

Jolbas, 2023-07-24:
I finally managed to build it with Boost 1.78 but did not get the crash. So it doesn't seem to be involved.

MattN92705, 2023-07-24:
Latest build still crashes for me :(
```
OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 0.21.0.33636 (Git)
Build type: Release
Branch: master
Hash: a5e350ad469f4030b46baa6fd3d1770375f1ddd1
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.60n
  * fasteners 0.4.56
  * ExtremeProLight 1.6.0
  * Help 1.0.3
  * Assembly4 0.50.2
  * Assembly3 0.12.0
```

0penBrain, 2023-07-24:
OK, I comitted a more complete PR #9987 
@chennes may you check it please ?
@adrianinsaval may you be kind to trigger another build when it's merged ?

adrianinsaval, 2023-07-24:
Of course, just ping me. For the record the process for that is as follows:
Trigger https://github.com/FreeCAD/freecad-feedstock/actions/workflows/increase_build_number.yml
Once the builds are finished trigger https://github.com/FreeCAD/FreeCAD-Bundle/actions/workflows/freecad_bundle.yml for x86_64 and https://cirrus-ci.com/github/FreeCAD/FreeCAD-Bundle/master for arm

edwilliams16, 2023-07-24:
@adrianinsaval  @OpenBrain @chennes 

Unfortunately - we don't appear to have a fix...

`-------------------------------------
Translated Report (Full Report Below)
-------------------------------------

Process:               freecad [86728]
Path:                  /Applications/FreeCAD-X86/FreeCAD.app/Contents/Resources/bin/freecad
Identifier:            freecad
Version:               ???
Code Type:             X86-64 (Translated)
Parent Process:        Exited process [86703]
User ID:               501

Date/Time:             2023-07-24 07:44:29.4537 -1000
OS Version:            macOS 13.4.1 (22F82)
Report Version:        12
Anonymous UUID:        D6C571A8-2D94-0C29-A3D0-5B218F76898B

Sleep/Wake UUID:       2BD06E9D-A1FC-4CD6-A667-4B007BA3A67A

Time Awake Since Boot: 830000 seconds
Time Since Wake:       1744 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Codes:       0x0000000000000001, 0x0000000000000010

Termination Reason:    Namespace SIGNAL, Code 11 Segmentation fault: 11
Terminating Process:   exc handler [86728]

VM Region Info: 0x10 is not in any region.  Bytes before following region: 140722985926640
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      mapped file              7ffc9f968000-7ffcaea9c000 [241.2M] r-x/r-x SM=COW  ...t_id=f74aedd1

Kernel Triage:
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   libCoin.4.0.0.dylib           	       0x1115a9a90 SoCallbackAction::~SoCallbackAction() + 144
1   libCoin.4.0.0.dylib           	       0x1115a9bfe SoCallbackAction::~SoCallbackAction() + 14
2   libCoin.4.0.0.dylib           	       0x1116230c2 SoDragger::~SoDragger() + 194
3   libFreeCADGui.dylib           	       0x10bf4c1be Gui::TDragger::~TDragger() + 14
4   libCoin.4.0.0.dylib           	       0x111781aac SoBase::destroy() + 172
5   libCoin.4.0.0.dylib           	       0x11177527a SoBaseList::truncate(int) + 74
6   libCoin.4.0.0.dylib           	       0x1117861b1 SoChildList::~SoChildList() + 17
7   libCoin.4.0.0.dylib           	       0x1117dc925 SoGroup::~SoGroup() + 37
8   libCoin.4.0.0.dylib           	       0x1117f8dce SoSeparator::~SoSeparator() + 14
9   libCoin.4.0.0.dylib           	       0x111781aac SoBase::destroy() + 172
10  libCoin.4.0.0.dylib           	       0x11177527a SoBaseList::truncate(int) + 74
11  libCoin.4.0.0.dylib           	       0x1117861b1 SoChildList::~SoChildList() + 17
12  libCoin.4.0.0.dylib           	       0x1117dc925 SoGroup::~SoGroup() + 37
13  libCoin.4.0.0.dylib           	       0x1117fd008 SoSwitch::~SoSwitch() + 56
14  libCoin.4.0.0.dylib           	       0x111781aac SoBase::destroy() + 172
15  libCoin.4.0.0.dylib           	       0x11177527a SoBaseList::truncate(int) + 74
16  libCoin.4.0.0.dylib           	       0x1117861b1 SoChildList::~SoChildList() + 17
17  libCoin.4.0.0.dylib           	       0x1117dc925 SoGroup::~SoGroup() + 37
18  libCoin.4.0.0.dylib           	       0x1117c552e SoAnnotation::~SoAnnotation() + 14
19  libCoin.4.0.0.dylib           	       0x111781aac SoBase::destroy() + 172
[... 54 lines cut ...]

VM Region Info: 0x10 is not in any region.  Bytes before following region: 105553518919664
      REGION TYPE                    START - END         [ VSIZE] PRT/MAX SHRMOD  REGION DETAIL
      UNUSED SPACE AT START
--->  
      MALLOC_NANO (reserved)   600018000000-600020000000 [128.0M] rw-/rwx SM=NUL  ...(unallocated)

Kernel Triage:
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage
VM - (arg = 0x0) pmap_enter retried due to resource shortage

Thread 0 Crashed:: CrBrowserMain Dispatch queue: com.apple.main-thread
0   libCoin.4.0.0.dylib           	       0x106beb704 SoCallbackAction::~SoCallbackAction() + 124
1   libCoin.4.0.0.dylib           	       0x106beb834 SoCallbackAction::~SoCallbackAction() + 12
2   libCoin.4.0.0.dylib           	       0x106c49598 SoDragger::~SoDragger() + 192
3   libFreeCADGui.dylib           	       0x10420ed4c Gui::TDragger::~TDragger() + 12
4   libCoin.4.0.0.dylib           	       0x106d8d1c8 SoBase::destroy() + 212
5   libCoin.4.0.0.dylib           	       0x106d8d1c8 SoBase::destroy() + 212
.....
509 libCoin.4.0.0.dylib           	       0x106d8d1c8 SoBase::destroy() + 212
510 libCoin.4.0.0.dylib           	       0x106d8d1c8 SoBase::destroy() + 212
`

edwilliams16, 2023-07-27:
This crashes as before. I tried running _freecad --log-file_  and saw no Assertion error.  Is that where they are supposed to show up, or do I need a debug build?  If you can point me at one, I'd be happy to try it out.

`[code]
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33656 (Git)
Build type: Release
Branch: master
Hash: 9f1e2c9f37dda46a10840095d0a044ca2e4d89cf
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * MeshRemodel 1.8919.0
  * DynamicData 2.46.0
  * fcgear 1.0.0
  * workfeature
  * Pyramids-and-Polyhedrons
  * QuickMeasure-main
  * GDML 1.8.0
  * offline-documentation
  * Manipulator 1.5.0
  * fasteners 0.4.56
  * lattice2 1.0.0
  * CurvedShapes 1.0.4
  * Alternate_OpenSCAD 1.0.0
  * toSketch 1.0.1
  * Help 1.0.3
  * Curves 0.6.12
[/code]
`

edwilliams16, 2023-07-27:
Latest PR isn't merged yet.

edwilliams16, 2023-08-02:
Released 0.21 without a fix!   Doesn't give a good impression when you get crashes with a core tool.

MattN92705, 2023-08-02:
> Released 0.21 without a fix! Doesn't give a good impression when you get crashes with a core tool.

yeah, 33658 still crashes on me.  I tried to test this bug on the 0.21 release, but it crashes on launch (Termination Reason:    DYLD, [0x1] Library missing).

frustrating.  but i get it.  free software, more on the todo list than there are people who can work it...

chennes, 2023-08-02:
> yeah, 33658 still crashes on me. I tried to test this bug on the 0.21 release, but it crashes on launch (Termination Reason: DYLD, [0x1] Library missing).

Intel or ARM?

MattN92705, 2023-08-02:
> > I tried to test this bug on the 0.21 release, but it crashes on launch (Termination Reason: DYLD, [0x1] Library missing).
> 
> Intel or ARM?

sorry - i've posted my config so many times i just forgot this time... Intel, MacOS 10.15 (that 10 year old iMac just keeps chugging along...)

chennes, 2023-08-02:
No worries, I think signing the Intel binary on my M2 corrupted it. I will re-sign and upload a new version. If this PR fixed the issue I'll also hot fix it in the release

chennes, 2023-08-03:
@0penBrain valiant effort, but I am still seeing this crash in the Conda build. Basically the same backtrace as above:
```
0   libCoin.4.0.0.dylib           	       0x106b9b704 SoCallbackAction::~SoCallbackAction() + 124
1   libCoin.4.0.0.dylib           	       0x106b9b834 SoCallbackAction::~SoCallbackAction() + 12
2   libCoin.4.0.0.dylib           	       0x106bf9598 SoDragger::~SoDragger() + 192
3   libFreeCADGui.dylib           	       0x1041b98dc Gui::TDragger::~TDragger() + 12
4   libCoin.4.0.0.dylib           	       0x106d3d1c8 SoBase::destroy() + 212
...
```

adrianinsaval, 2023-08-03:
Could it be a problem in coin itself?

jacprog, 2023-08-03:
tested latest arm and intel on apple M1
no change yet
still crash after transform when click OK
Crash report similar to edwilliams16
[code]
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.21.0.33679 (Git)
Build type: Release
Branch: master
Hash: 205c0e73459088308812858282ff51f09dfd2676
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * freecad.gears-master 1.0.0
  * FreeCAD_Assembly4-master 0.12.4
  * freecad-slic3r-tools-master
  * fasteners 0.3.40
[/code]
Now played around a little
Part , Part Design, Fastener, issue is the same
To be more precise:
in Part:
Made a sketch to have a reference
Made a Cube
select cube, right click, select transform
if no interaction, no move, no translation, click ok, no crash
A simple click on transform handle, even with no movement will crash at press of OK
if make a move, check property value in Combo view, the position value is updated
click save, the new position is saved
click ok, freecad crash
Click reopen in Apple crash report  or open freecad by selecting the saved file
the new position of the cube has been saved.
It is the process that happen when clicking OK in tasks in combo view that cause the crash and only when the transform handle (3 color arrows)has Benn clicked on, activated, moved or not
One more thing I noticed, could related or not
and that happen too in freeCAD 0.20.2 
Select the cube right click select transform
the transform handle show up at the corner of cube
Not closing the transform I go to model and change the position with the property value
the transform handle stays where it is, but the cube moves according to the position I enter in x, y, z
when clicking and moving the transform handle, the cube and the handle sychronize
and de-sychronize when releasing the mouse
If I click ok to close the transform tool, then reopen, the transform and cube are in synch again

Jolbas, 2023-08-03:
The issue was present at build 0.21.0.33458 (tested by @MattN92705 23-06-23) and then temporarily resolved on 0.21.0.33458 (tested 23-06-30) and it's back on 0.21.0.33593 (tested 23-07-16) All of them seems to be using coin3D 4.0.0. The crashing versions uses VTK 9.2.2 or 9.2.5 while the working ones uses VTK 9.1.0 or 9.2.6. Could that be relevant? Could there be some interference between coin3D and VTK? And maybe in combination with Boost 1.78.
Would it bi possible to use Boost 1.81 and or VTK 9.2.6 in the distributed builds?

0penBrain, 2023-08-03:
> @0penBrain valiant effort, but I am still seeing this crash in the Conda build.

Will be hard to catch. :neutral_face:

Jolbas, 2023-08-03:
> Could it be a problem in coin itself?

I replaced FreeCAD.app/Contents/Resources/lib/libCoin.4.0.0.dylib with the same dylib of FreeCAD build 33515 and then there is no crash!
The Coin3D versions are different. Both are 4.0.0 but the working one is marked as build 1011 in FreeCAD 33515.app/Contents/Resources/conda-meta/ and the failing one is build 1012 and they differ in size
Here's the working library if someone want to try and confirm:
[libCoin.4.0.0.dylib.zip](https://github.com/FreeCAD/FreeCAD/files/12253540/libCoin.4.0.0.dylib.zip)

edwilliams16, 2023-08-03:
> I replaced FreeCAD.app/Contents/Resources/lib/libCoin.4.0.0.dylib with the same dylib of FreeCAD build 33515 and then there is no crash! 

Great catch.  I just tried this with the X86 0.21.0-33675 and no crash.

adrianinsaval, 2023-08-03:
Good to know, I'll try to see what went wrong with the newer coin build and how we can use the working one or fix the library.

chennes, 2023-08-03:
So to be clear, if I just repackage the Mac OS builds with the 33535 dylib the problem is resolved?

edwilliams16, 2023-08-03:
> So to be clear, if I just repackage the Mac OS builds with the 33515 dylib the problem is resolved?

That appears to be the case for X86 - not verified AFAIK for the corresponding arm library.

Jolbas, 2023-08-03:
> So to be clear, if I just repackage the Mac OS builds with the 33515 dylib the problem is resolved?

Yes, at least for x86 build. It's not tested on arm build.

chennes, 2023-08-03:
Are those bundles still available someplace?

adrianinsaval, 2023-08-03:
https://anaconda.org/conda-forge/coin3d/files

Jolbas, 2023-08-03:
If someone are building a non crashing version on Mac with M1 or M² a working coin should be located at /usr/local/Cellar/coin3d/4.0.0_4/lib/libCoin.4.0.0.dylib

edwilliams16, 2023-08-03:
I tried just blindly replacing  /Applications/FreeCAD/Contents/Resources/lib/libCoin*  with the corresponding files from the anaconda.org
[https://anaconda.org/conda-forge/coin3d/4.0.0/download/osx-arm64/coin3d-4.0.0-h21c6956_1011.tar.bz2](url)

It didn't work..., but I don't know what I'm doing here...

`bin % ./freecad --log-file ~/Desktop/logfile_arm.txt
dyld[63802]: Library not loaded: @rpath/libCoin.80.dylib
  Referenced from: <34A7699C-CBD0-335B-9F23-A98BD19B529D> /Applications/FreeCAD_arm/FreeCAD.app/Contents/Resources/lib/libFreeCADGui.dylib
  Reason: tried: '/Applications/FreeCAD_arm/FreeCAD.app/Contents/Resources/lib/libCoin.80.dylib' (code signature in <C9D9FBC8-839C-3C1B-BB9F-5A7DC2895262> '/Applications/FreeCAD_arm/FreeCAD.app/Contents/Resources/lib/libCoin.4.0.0.dylib' not valid for use in process: mapped file has no Team ID and is not a platform binary (signed with custom identity or adhoc?)), '/Applications/FreeCAD_arm/FreeCAD.app/Contents/Resources/bin/../lib/libCoin.80.dylib' (code signature in <C9D9FBC8-839C-3C1B-BB9F-5A7DC2895262> '/Applications/FreeCAD_arm/FreeCAD.app/Contents/Resources/lib/libCoin.4.0.0.dylib' not valid for use in process: mapped file has no Team ID and is not a platform binary (signed with custom identity or adhoc?)), '/Applications/FreeCAD_arm/FreeCAD.app/Contents/Resources/bin/../lib/libCoin.80.dylib' (code signature in <C9D9FBC8-839C-3C1B-BB9F-5A7DC2895262> '/Applications/FreeCAD_arm/FreeCAD.app/Contents/Resources/lib/libCoin.4.0.0.dylib' not valid for use in process: mapped file has no Team ID and is not a platform binary (signed with custom identity or adhoc?)), '/usr/lib/libCoin.80.dylib' (no such file, not in dyld cache)
zsh: abort      ./freecad --log-file ~/Desktop/logfile_arm.txt`

chennes, 2023-08-03:
With a signed app you can't replace parts of it, the signature validation will fail (which is what you are seeing there). I'll have them updated in an hour or two.

chennes, 2023-08-04:
@edwilliams16 try the arm64 download now, it should have the "updated" Coin library (the Intel is still chugging away).

Jolbas, 2023-08-04:
The new Intel download works.

edwilliams16, 2023-08-04:
The Mac-arm download also works.  Great job everybody!  Probably can close.

Jolbas, 2023-08-04:
But this isn't yet resolved in the automatic weekly builds, is it?

adrianinsaval, 2023-08-04:
> But this isn't yet resolved in the automatic weekly builds, is it?

Correct. The issue should remain open

MattN92705, 2023-08-06:
Unfortunately, the latest build crashes.

```
OS: macOS 10.15
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33719 (Git)
Build type: Release
Branch: master
Hash: 2a88c7c7dfed699d2a4944bda13bf8626ed39315
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * A2plus 0.4.60n
  * fasteners 0.4.56
  * ExtremeProLight 1.6.0
  * Help 1.0.3
  * Assembly4 0.50.2
  * Assembly3 0.12.0
```

```
Process:               freecad [60227]
Path:                  /Applications/FreeCAD.app/Contents/Resources/bin/freecad
Identifier:            org.freecadweb.FreeCAD
Version:               ??? (0.22.0-33719)
Code Type:             X86-64 (Native)
Parent Process:        ??? [60221]
User ID:               504

Date/Time:             2023-08-06 06:33:27.603 -0700
OS Version:            Mac OS X 10.15.7 (19H2026)
Report Version:        12
Anonymous UUID:        5D59B88E-1E33-C625-CBB6-884D97888AA1

Sleep/Wake UUID:       731A3000-E0A9-4504-90C1-0E1E264BA14C

Time Awake Since Boot: 2700000 seconds
Time Since Wake:       13000 seconds

System Integrity Protection: enabled

Crashed Thread:        0  CrBrowserMain  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_BAD_ACCESS (SIGSEGV)
Exception Codes:       KERN_INVALID_ADDRESS at 0x0000000000000010
Exception Note:        EXC_CORPSE_NOTIFY

Termination Signal:    Segmentation fault: 11
Termination Reason:    Namespace SIGNAL, Code 0xb
Terminating Process:   exc handler [60227]

VM Regions Near 0x10:
--> 
    __TEXT                 0000000106c3f000-0000000106c60000 [  132K] r-x/r-x SM=COW  /Applications/FreeCAD.app/Contents/Resources/bin/freecad

Thread 0 Crashed:: CrBrowserMain  Dispatch queue: com.apple.main-thread
0   libCoin.4.0.0.dylib           	0x0000000109f8aa90 SoCallbackAction::~SoCallbackAction() + 144
1   libCoin.4.0.0.dylib           	0x0000000109f8abfe SoCallbackAction::~SoCallbackAction() + 14
2   libCoin.4.0.0.dylib           	0x000000010a0040c2 SoDragger::~SoDragger() + 194
3   libFreeCADGui.dylib           	0x0000000107001f5e Gui::TDragger::~TDragger() + 14
4   libCoin.4.0.0.dylib           	0x000000010a162aac SoBase::destroy() + 172
5   libCoin.4.0.0.dylib           	0x000000010a15627a SoBaseList::truncate(int) + 74
6   libCoin.4.0.0.dylib           	0x000000010a1671b1 SoChildList::~SoChildList() + 17
7   libCoin.4.0.0.dylib           	0x000000010a1bd925 SoGroup::~SoGroup() + 37
8   libCoin.4.0.0.dylib           	0x000000010a1d9dce SoSeparator::~SoSeparator() + 14
9   libCoin.4.0.0.dylib           	0x000000010a162aac SoBase::destroy() + 172
10  libCoin.4.0.0.dylib           	0x000000010a15627a SoBaseList::truncate(int) + 74
11  libCoin.4.0.0.dylib           	0x000000010a1671b1 SoChildList::~SoChildList() + 17
12  libCoin.4.0.0.dylib           	0x000000010a1bd925 SoGroup::~SoGroup() + 37
13  libCoin.4.0.0.dylib           	0x000000010a1de008 SoSwitch::~SoSwitch() + 56
14  libCoin.4.0.0.dylib           	0x000000010a162aac SoBase::destroy() + 172
15  libCoin.4.0.0.dylib           	0x000000010a15627a SoBaseList::truncate(int) + 74
16  libCoin.4.0.0.dylib           	0x000000010a1671b1 SoChildList::~SoChildList() + 17
17  libCoin.4.0.0.dylib           	0x000000010a1bd925 SoGroup::~SoGroup() + 37
18  libCoin.4.0.0.dylib           	0x000000010a1a652e SoAnnotation::~SoAnnotation() + 14
19  libCoin.4.0.0.dylib           	0x000000010a162aac SoBase::destroy() + 172
20  libCoin.4.0.0.dylib           	0x000000010a15627a SoBaseList::truncate(int) + 74
21  libCoin.4.0.0.dylib           	0x000000010a1671b1 SoChildList::~SoChildList() + 17
22  libCoin.4.0.0.dylib           	0x000000010a1bd925 SoGroup::~SoGroup() + 37
23  libCoin.4.0.0.dylib           	0x000000010a1d9dce SoSeparator::~SoSeparator() + 14
24  libCoin.4.0.0.dylib           	0x000000010a162aac SoBase::destroy() + 172
25  libCoin.4.0.0.dylib           	0x000000010a15627a SoBaseList::truncate(int) + 74
26  libCoin.4.0.0.dylib           	0x000000010a1671b1 SoChildList::~SoChildList() + 17
27  libCoin.4.0.0.dylib           	0x000000010a1bd925 SoGroup::~SoGroup() + 37
28  libCoin.4.0.0.dylib           	0x000000010a1d9dce SoSeparator::~SoSeparator() + 14
29  libCoin.4.0.0.dylib           	0x000000010a162aac SoBase::destroy() + 172
30  libCoin.4.0.0.dylib           	0x000000010a15627a SoBaseList::truncate(int) + 74
31  libCoin.4.0.0.dylib           	0x000000010a1671b1 SoChildList::~SoChildList() + 17
32  libCoin.4.0.0.dylib           	0x000000010a196215 SoBaseKit::~SoBaseKit() + 37
33  libFreeCADGui.dylib           	0x000000010700a06e Gui::SoFCCSysDragger::~SoFCCSysDragger() + 14
34  libCoin.4.0.0.dylib           	0x000000010a162aac SoBase::destroy() + 172
35  libFreeCADGui.dylib           	0x0000000106e50ea0 Gui::TaskCSysDragger::~TaskCSysDragger() + 32
36  libFreeCADGui.dylib           	0x0000000106e50f2e Gui::TaskCSysDragger::~TaskCSysDragger() + 14
37  libFreeCADGui.dylib           	0x00000001070628c9 Gui::TaskView::TaskView::removeDialog() + 313
38  libFreeCADGui.dylib           	0x0000000107062f6a Gui::TaskView::TaskView::accept() + 202
39  libQt5Core.5.15.8.dylib       	0x0000000109c35f3a 0x109aea000 + 1359674
40  libQt5Widgets.5.15.8.dylib    	0x0000000108d8de7b 0x108c7f000 + 1109627

```

[30 more replies]

PR #9975 (2023-07-23): [Bugfix]Gui: detach field sensor of draggers and handle post-call of callbacks
This is an attempt to fix #9465 

I can't test at all (no Mac) but AFAIK nobody really can as this only happens on prebuilt versions, and not on own build ones.
So at some time, only way to test may be to merge and see. :smile:

I did this because the reported framestack makes me think to symptoms of a Coin sensor that wouldn't be correctly detached.
Looking at the code, I indeed found the `fieldSensor` of `TDragger` and `RDragger` to not be detached.*They now are detached in instance dtor.

As sensor data is instan [... cut]
commit ca093646 (2023-07-23): [Bugfix]Gui: detach field sensor of draggers and handle post-call of callbacks, fixes #9465
PR #9987 (2023-08-02): [Bugfix]Gui: in draggers, remove/detach all callbacks + add checks and asserts
This is continuation of #9975 in an attempt to fix #9465 

I found sensor callbacks that hadn't been covered in #9975 and fix them.

Also detach all "standard" callbacks in dtor and add asserts so at least if they are culprit for the crash we have a trace.
commit 205c0e73 (2023-08-02): [Bugfix]Gui: in draggers, remove/detach all callbacks + add checks and asserts, hopefully fixes #9465
closed 2023-10-27 by wwmayer