FreeCAD/FreeCAD #9466: [Feature Request] Improvements for the tool library UI

knipknap, 2023-05-03:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=78010

### Subproject(s) affected?

Path

### Idea description

As a user, I want to see some improvements in the tool library:

### List view changes

- Show information about tool parameters in the tool list, because currently the parameters can only be seen one-by-one which is tedious and slow if you have a big tool library (see https://forum.freecad.org/viewtopic.php?t=78010).
- It de-facto forces me to repeat myself, editing the data not just once, but in fact two or even three times:
    - Once in the parameter list of the tool.
    - Once in the name of the tool, just to make the data visible in the tool list.
    - Once in the file name - which brings its own problems (see below).

The proposal is to rework the UI to support a view like this:

[image]

This would eliminate the need for redundant information in parameters+name and would be relatively fast to use.

### Filename handling changes

Forum link: https://forum.freecad.org/viewtopic.php?t=78030

In addition, I also propose to consider the tool file name to be an internal data structure not exposed to the user. Reasoning:

- These file names are (and have to remain) immutable, because otherwise it could break other tool libraries using the same file. So if we ask the user, then he has two choices:
    1. Use a meaningful name: In this case, the name will be wrong as soon as the tool is edited in the UI.
    2. Use a meaningless name (e.g. an ID). In this case, it doesn't make sense to ask the user, FreeCAD could create the file automatically.

### Summary

Together, these changes would
- Eliminate the repeating of attributes by users.
- Allow the user to quickly see the tool parameters in the UI.

### Anything else?

Qt supports these kinds of cell formatting: https://stackoverflow.com/questions/25187444/pyqt-qlistwidget-custom-items

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2023-05-04:
Thanks for creating the issue.  It's usually helpful to avoid 'feature request' and instead talk about a problem to be solved.  In this case, you've alluded to at least three different problems:
problem 1 - You don't know which tool to pick because there isn't enough information shown.
problem 2 -  Seeing tool details is slow if you have very large collections of tools
problem 3 - You have to repeat information multiple times to create a tool. 

The original design intent of the dialog was to allow the user to add tool controllers without interrupting the workflow of creating an op.  As such, the user may wish to keep the dock open for a long time, even the whole time they are using Path.  So it should occupy as little screen real estate as possible.

@knipknap's idea of adding a row  of meta-data below the description as shown in the image is interesting.  Realistically, the icons won't look that good.  More like:
[image]

I'm not sure how useful that is.  It adds to the dock width but I can see some value.

The problem is that while this solves problem 1, it doesn't do anything for problem 2.  You can't sort based on the metadata like you could with a table of sortable columns.
In some ways it makes the problem of large tool collections worse because each item occupies more vertical space.  So you'll be scrolling a lot more to find items far down the list.

One possibility would be some kind of filter widget at the top of the list.  Something similar to the filters in sketcher
[image]

A two part filter widget [filter type] [filter value] where the type is either shape, diameter, flute count, etc.  and the value is an input

knipknap, 2023-05-04:
I think that breaks down the problems quite well, Thank you!

**About problem 1:**

I agree the dock should be kept as small as possible. Honestly I mostly look for the tool in the library editor, then select in the dock based on the tool ID.
From my point of view it may be sufficient to show some of the details only in the opened library editor and not in the dock - IMO the dock doesn't need things like the icon as long as the tool is recognizable.

Regarding the tool images I had in mind to show the tool shape, though this probably requires some sort of caching. Spiffing this up by actual images is not really needed IMO.

**About problem 2:**

I agree a filter would be a really good addition to help with problem 2. I just want to note that resolving problem 1 for me would also help with problem 2, as I haven't yet arrived at the point of "gigantic" tool collection (i.e. >100), so if the attributes were visible I could find the tool relatively quickly by scrolling through the list without having to clicking on multiple tools to examine them.

But yeah, additional filtering would be the icing on the cake, and for larger tool collections is probably pretty much essential - just without showing the parameters this also wouldn't work. At least I usually go about it in the other way: I look for a tool of a certain type (e.g. "up-down cut endmill") and then check the sizes/lengths/flutes/... available to choose a tool. Relatively rarely I look for a specific parameter first.

**About problem 3:**

Is also partially addressed by 1.

knipknap, 2023-05-05:
I like this approach - just a minor addition: In this case maybe it makes sense to add an "edit tool" button in area 4 as these actions are all related (btw. could also rename "toolbit" to "tool" for some more space, also since not all tools are necessarily bits.

The "double-click-action" on each row could be equivalent to clicking the "add to job" button in area 3.

This way the dialog becomes the "Add tool to job" dialog, with the (more rarely used) option to add/edit tools "in place".

knipknap, 2023-05-05:
In general I would think that UIs should try to avoid "context switching" - i.e. a user action should mean the same thing in every situation. Maybe that is what you mean when you say that it "may be confusing".

Because of that reason, IMO it is good practice to avoid different "modes" for one user interface.

sliptonic, 2023-05-05:
Exactly.  
Any thoughts on how we do that?

knipknap, 2023-05-05:
Well, that's what I meant by this:

- Leave area 4 visible all the time, no "edit library" button needed in area 6.
- The "double-click-action" on each row could be equivalent to clicking the "add to job" button in area 3.
- To edit a toolbit you would then need a new "Edit toolbit" button in area 4.

sliptonic, 2023-05-05:
Alright.  I can live with it.  Give it a try.

knipknap, 2023-05-05:
I am still not quite sure what to do to show the parameters in the list. I understand that removing the dock gives us a little more freedom to use more space e.g. for additional columns, but the problem that not all tools share common attributes remains and we have to decide how to use that space effectively.

The way I see it, there are two options:
1. Agree on a set of "likely common" parameters that are used by a majority of tools and add columns for them. This is what I tried in my initial patch that I posted in the forum. But in some environments (e.g. someone running a lathe) this will lead to several cells showing "n/a", and it may not scale in the future.
2. Go with the two-line UI from my initial posting. In this case I would define a priority for all known parameters and dynamically show only the first four or so. The UI wouldn't allow for ordering based on parameters, but filters could still be implemented.

I also thought of introducing the concept of "library types" - e.g. one for lathe tools, one for probes, ... and then show different columns depending on that. But I think in reality machine types differ too much to cover everything - there are lathes that use endmills, CNCs that support sawblades, etc. So a user would then have to choose tools from multiple libraries, depending on the tool type. Another drawback is that FreeCAD could not ensure that tool IDs are unique from this user.

So in summary, I still think option 2 (combined with the other UI and button changes that we discussed) would be best.

sliptonic, 2023-05-05:
I think idea 2 sounds the best.  As long as the logic for deriving the second line of information is in just one place, we can revise it or even make it configurable in the future.  
Especially if there's an info panel at the bottom of the left pane, the user can always see more if needed.

knipknap, 2023-05-20:
Nothing to show for yet, I am still fighting some other issues at the moment. I have locally started, but it'll probably be a while.

knipknap, 2023-05-20:
[Here is the branch.](/knipknap/FreeCad/tree/toolbit-library-9466)

This is actually half way there already, some parts of what we discussed is finished: The dock is gone, everything works from directly within the library dialog.

Yet to be done:
- Replace table columns by the dynamic two-line table rows.
- Auto-file-naming

In any case, I think the current implementation is pretty much ready to be merged if you are happy with partial progress.

Not sure what the policy is on updating translation files, as some strings are now obsolete and I think two or three were changed or added.

knipknap, 2023-05-21:
One problem with the "tool icons/thumbnails" I noticed during the implementation: The only way to display them is by creating a temporary tool in an active document. That is a bit ugly, because:

1. It is slow and prevents us from loading many icons at the same time. Definitely cannot be used to show a preview of each tool in a list.
2. It prevents the tool editor sub-dialog from being looked at when no document is open. And in the current implementation even the library dialog is blocked from being opened (which makes sense because of the aforementioned problem). That has bugged me numerous times when organizing/researching stuff in the workshop.
3. I think the library editor in general shouldn't manipulate a document, unless explicitly asked by clicking "Add toolbit to job". A document shouldn't need to be recompute just because a shape in the tool library was edited/added/removed. So long as it is still in the job, of course. 

So I am thinking of the following plan to address these issues:
1. Decouple the tool editor from the open document by rendering the toolbit in a separate, headless FreeCAD instance. I have done something similar using "freecadcmd". It feels hacky though, not sure if there is a more direct way using the official API while preventing the user from seeing it. This would resolve the issue of not being able to open the library editor while no document is open.
2. Create static icons for the tool list and show those there. Even better would be if FreeCAD files had embedded thumbnails...

knipknap, 2023-05-21:
@sliptonic I am mighty confused by "class _TableView" in BitLibrary.py. This [class implementation has references to self.toolModel()](https://github.com/knipknap/FreeCAD/blob/48ef2cf37de98b8ab2a039f79344212d00cf7197/src/Mod/Path/Path/Tool/Gui/BitLibrary.py#L190), but I don't see it being set anywhere. Is this monkey-patched somewhere?

I also can't seem to use drag&drop in the UI, so I am wondering if this is all dangling code that can just be removed.

knipknap, 2023-07-22:
Since this issue was stalled, I implemented all features [as an Addon](https://github.com/knipknap/better-tool-library), In case somebody want to still do something like it in FreeCAD directly, maybe there is something useful to copy. Also not sure if I am supposed to close the issue now, as I requested the feature but no longer need it.

JohnOCFII, 2023-12-25:
> Since this issue was stalled, I implemented all features [as an Addon](https://github.com/knipknap/better-tool-library), In case somebody want to still do something like it in FreeCAD directly, [...]

It seems a shame not to integrate this into mainline FreeCAD instead of having it as an Add-on.

Unlogic-dot-se, 2024-05-08:
I agree this, this should really be considered for merging into the FreeCAD main branch.

sliptonic, 2024-05-09:
The addon provides some nice utility but it isn't suitable for merging upstream for a few reasons.
First, it's re-incorporating some of the existing tool shapes but with a changed license.  I'm not sure why this is.  For example, the slitting saw and others are included with 'All rights reserved'  The original tools are included with the FreeCAD source as 'CC-BY-4'.   I expect that this is just an oversight and the included files haven't been updated since the licenses changed.  I don't know why the addon adds another copy of those toolshapes rather than using the ones already installed with CAM.

A second problem is that the addon is using data from machiningdoctor.com.  It's unclear if the author has permission to use the data in this way.  If the data can be used legally, the logic needs to be reconsidered to use the new materials system being implemented now.

The addon also uses 2D SVG images of the tools to give a visual correspondence between attributes like 'length' and what they physically correspond to in the tool shape.  This is very attractive visually but it creates two major challenges which I raised with the author before the addon was written:

First, the images aren't translatable.
Second, the images have to be created and maintained manually.  

The existing tool system relies on FreeCAD functionality to generate the thumbnails automatically.  This means someone can create their own toolshape and a thumbnail will automatically be created.  The thumbnails aren't as attractive but we get them for free with no maintenance.

I encouraged the author to work on improving the visual attractiveness of the automatic images rather than relying on hand-drawn images. 

I want to see the tool system improved.  I'm keenly aware of its current limitations.  However, I'm unwilling to include functionality that makes the application less translatable or requires more manual intervention.

Unl0gic, 2024-05-10:
Thanks for taking the time to write such a detailed reply @sliptonic. Everything you mention makes full sense.

From a usability perspective this add-on has great potential but it's clear that it needs to be somewhat reworked before it can be considered for merging into the main branch.

knipknap, 2024-07-27:
> The addon provides some nice utility but it isn't suitable for merging upstream for a few reasons. First, it's re-incorporating some of the existing tool shapes but with a changed license.

I never made any license change. I double-checked the files that you mentioned and re-downloaded them from the FreeCAD repository - when I open them locally, they all have "All rights reserved" licenses, not CC-BY-4.

https://github.com/FreeCAD/FreeCAD/tree/main/src/Mod/CAM/Tools/Shape

Copying them was necessary as they weren't included in all versions of the AppImage, so the tool library initialization wouldn't work for all users.

As for the other points I really don't understand why you are reiterating them, this was previously discussed:

> The addon also uses 2D SVG images of the tools to give a visual correspondence between attributes like 'length' and what they physically correspond to in the tool shape. This is very attractive [...]

This was discussed:
- Firstly, BTL uses SVG images and only falls back to pixmaps if no SVG is present. If you want, you can translate them - they are arguably easier to translate then any existing format, because open standards exists for SVG.
- Second, if neither SVG nor Pixmaps are present, BTL already falls back to using an auto-generated image.
- Third, even though you could already translate them if you wanted, IMO it would be absurd, because no semantics are required here. Would it be accepted if the all letters were replaced by other geometric shapes? I think that would be putting in effort for arguably no benefit.
- And lastly, the current FreeCAD internal library relies on parameter names in the shape files. These are entirely untranslatable, which is why the FreeCAD Tool Editor is currently untranslatable as well.

In short: There is currently no better solution. If one ever pops up, the BTL approach and/or SVGs would not stand in the way.

As for the feeds taken from machining doctor: These can be removed if needed. I think in general, facts are not copyrightable, especially miniscule amounts of data isn't, but if there are doubts the feeds could be slightly changes. It's just a handful of integers.

phaseloop, 2025-03-14:
I opened another discussion on this topic (before I found this one)

https://github.com/FreeCAD/FreeCAD/issues/20187

I agree with @knipknap arguments about loading order of an SVG, then pixmap, etc.

As for machining data, this can probably be copied as fair use from machining tables or other source too (maybe public domain)

Also I would suggest too to standardize dimension labels - I see people complaining that FC tool conventions in shapes do not match the industry ones and are confusing.

Let's be honest - there is a pretty limited number of tool shapes and we should focus on good experience for those. So I don't buy an argument about need to create and maintain SVG files being an obstacle - we maintain tons of SVG's and icons in the project.

sliptonic, 2025-03-16:
I've initiated discussion with the DWG about this. https://discord.com/channels/870877411049357352/1162044910535450665/1350149329242816593

Based on their feedback and knipknap's arguments I'm withdrawing my opposition.  @knipknap, if you want to prepare a PR to bring BTL in as the default tool management system, you have my support.

phaseloop, 2025-03-16:
That's great news! Thanks for bringing it up to DWG

knipknap, 2025-03-16:
Cool, I am happy to hear it, thanks! I am currently focusing on another project, but hope to start trying to integrate it soon.

I am guessing it is a week worth of code changes to prepare a PR, but want to plan being available in case any fixes are needed after a release. Any release milestones I should watch out for?

knipknap, 2025-03-16:
I tried to open the link, but it seems that I do not have access.

phaseloop, 2025-03-17:
Yeah, you need to be logged to discord account, then join the FreeCAD server using link below, then a bot will contact you to answer some additional captcha and then bot will unlock you the access to chat channels. Tried that yesterday, lmao.

https://discord.gg/uh85ZRNcfk

or this one

 https://discord.gg/w2cTKGzccC

knipknap, 2025-03-17:
Thanks, your second invitation link worked!

knipknap, 2025-04-14:
I have started with the integration, the branch is here:

https://github.com/knipknap/FreeCAD/tree/btl

Turns out integrating it is way harder than expected, due to overlapping functionality. So it might take a while to sort all this out.

knipknap, 2025-04-15:
@sliptonic I am at a decision point where I need to ask. The issue:

Currently BTL [defines its own ToolBit class](https://github.com/knipknap/better-tool-library/blob/main/btl/tool.py), and converts it to the FreeCAD tool class when adding it to the job.

In addition, BTL [defines an explicit Shape class](https://github.com/knipknap/better-tool-library/blob/main/btl/shape.py), which is essential for the image generation and Feeds & Speeds simulation.

This means there are two options when doing the integration:
1. Either extend the existing CAM.Tool class to add the functionality needed by BTL, effectively merging the two Tool classes.
2. Or, keep the models separate as if BTL were a separate extension, use conversion functions where applicable.

I think option 1 is cleaner, but it has far reaching consequences:
- Functionality extension of the ToolBit class is non-trivial.
- Many tests (tool mocks, etc.) and UI functions need to be changed - it is far from straight forward.
- Factories for creating Tool bits need to be changed
- The changes will almost certainly break any extensions referencing CAM tools (I am not sure that are any such extensions)

Option 2 on the other hand comes with its own problems:
- Long term maintainability is questionable if CAM has two types of ToolBit classes. It certainly wouldn't be good design if these functions were written from scratch.
- Duplication of serialization/deserialization of tools. In BTL I implemented a [Serialization protocol](https://github.com/knipknap/better-tool-library/blob/main/btl/serializers/serializer.py). The CAM internal serialization is instead just a function in the UI (view) code.

My proposal would be to go with option 1. Any thoughts?

knipknap, 2025-04-15:
Ouch, yeah, that PR definitely conflicts with my integration branch.

I agree that the introduction of shape-specific schemas makes sense.  So does the separation of Shape and ToolBit, because there are tool attributes that IMO should not be defined in the shape file, such as material, vendor info, hardness, tool notes etc etc. In other words, I would limit the use of properties in the Shape file strictly to geometric information.

The only issue: This would for now block any attempt to integrate BTL and would rather turn this into a re-development effort.

knipknap, 2025-04-17:
I agree that it makes sense to migrate from PropertyBag to VarSet.

Since this breaks compatibility, it may be an opportunity to refactor the toolbit design. In the current implementation, the cyclomatic complexity is IMO just too high, and the root cause is a lack of separation between toolbit and shape. This also causes some real problems with compatibility (see below).

This is the current signature of the ToolBit:

[image]

One problem is that the ToolBit needs to keep four inter-dependent properties consistent: BitShape, ShapeName, BitBody, and File, further complicated by the fact that some of those attributes depend on the environment (i.e. files may disappear, models may be edited, ...)

This complexity may rise even further by re-introducing the Tool Type property as @phaseloop suggested.

My proposal: Separate the ToolBitShape from the ToolBit as such:

- ToolBit should primarily manage the ToolBitShape (enum) and the link (BitBody) to the actual parametric shape object.
- Remove inter-dependent properties (BitShape, ShapeName, BitPropertyNames) from ToolBit.
- ToolBitShape handles loading the file, and applying geometric parameters to it (those are also the parameters that are specified in the VarSet)
- ToolBit handles non-geometric parameters relevant for the specific tool instance, such as material, hardness, etc.

[image]

Additional advantage:

- By encapsulating each shape type in a proper class future migrations are simplified, as the class can act as a bridge/compatibility layer.
- It solves a problem I had with the implementation of Fusion 360 tool compatibility: some tools cannot be mapped to existing FreeCAD tools without some translation logic. Having the ability to subclass existing tools could solve this.

knipknap, 2025-04-18:
> will there be one ToolBitShape class per shape (like vbit, endmill, etc) or one class to handle all shapes? I suppose if we plan to keep schema there there should be a ToolBitShapeVBit, etc with relevant info.

Yes, thats how I would propose it.

> if someone wants to create custom shape, will they need to add relevant class? Like having a custom endmill, etc.

Yes, if it is really a completely new type of shape - but I think in that case CAM operations and post processors need to "understand" the ToolBitShape anyway - just knowing the geometry is not enough for most operations.
We could also introduce a special ToolBitShapeCustom, but I think that is only useful for CAM operations that are completely shape-agnostic, and I think none really are, considering the high diversity of tooling (think lathe tools, probes, endmills, drills, saws, etc.)

> I can retract my PR to not bring more mess and I can work on it once BTL is integrated.

I can make a PR similar to yours that focuses on the requirements you had for the post processors, but introducing the explicit Shape class in addition. I would be grateful for your review, it's useful to understand early if this introduces any issues.

sliptonic, 2025-04-18:
> We could also introduce a special ToolBitShapeCustom, but I think that is only useful for CAM operations that are completely shape-agnostic, and I think none really are, considering the high diversity of tooling (think lathe tools, probes, endmills, drills, saws, etc.)

Think about the huge number of router bits with odd profiles.  While the toolpath generation itself is shape-agnostic, the resulting shape from the simulator isn't.  I think we need to allow this as it's an intuitive way to verify the toolpath.  Even if the simulator isn't up to the job in all cases now, it's likely to improve.

phaseloop, 2025-04-18:
Exactly - to be honest I think fcstd files are not needed for anything except simulation in our case.

@knipknap - I'm happy to help with reviews or some dev work

knipknap, 2025-04-18:
Thanks. Nothing to review yet, but I have been progressing a lot with the TollBitShape introduction, today finally all the tests pass.

The WIP branch is here https://github.com/knipknap/FreeCAD/tree/add-toolbitshape

But lots of stuff yet to do though before this is even ready for a review:

- [ ] Should ToolBitShape even store the default values? It would be redundant to what is in the shape file...
- [ ] Define a ToolBitShape hierarchy. Example: Right now torus (bullnose) bits directly inherit from ToolBitShape, but probably should inherit from Endmill so common operations like FaceMill would not need to special case them. Alternatively, define interfaces such as CanChamfer, CanBackChamfer, to allow Ops and processors to easily identify "compatible" bits. But that may go into overengineering territory quick, so... not yet, I think.
- [ ] Replace ToolBitFactory by class methods and remove redundant code
- [ ] Align ToolBitShape.name, .alias, and .label with BTL terminology to avoid headaches in the future
- [ ] Clean up handling of non-shape-properties, such as material and chipload
- [ ] Clean up handling of 'flutes' param and cross-check with issues seen in BTL
- [ ] Identify dead code and remove it
- [ ] Fix: ToolBitShape enum contains duplicates
- [ ] Fix: ToolBitShape is added to active document, but not under the tool controller
- [ ] Fix: ToolBit is also under the tool controller (separately), but does not seem to have a body, nothing visible
- [ ] ...
- [ ] Final clean up

knipknap, 2025-04-19:
I am starting to wonder if it wouldn't be better to completely ignore the FCStd files, once imported, and keep the parameterizeable Body around in the active document. That would solve the issue of sharing a file that then cannot easily be edited because the tool file isn't there.

I guess the idea was that we should also support Solids (e.g. STEP import), but this could still optionally be done - solids would be just another type of ToolBitShape (which then wouldn't be editable, of course).

phaseloop, 2025-04-19:
I think this is a good idea. Right now it seems we store full path to shape in file so even if you have same tool shapes you need to fix the paths to point them to your local FC installation dir.

Is there any benefit from importing STEP files (like importing tools from other CAM suites)?

knipknap, 2025-04-19:
Yes, I think some tool suppliers provide .step files of their tools.

knipknap, 2025-04-19:
Ahh, the idea does not work. The problem is that the propertybag (or varset) is not part of the body. Unless we to store both, the varset and the body...

But then what if there are multiple shapes, the varset has the same name in all shape files... renaming for conflict avoidance, but this is getting ugly.

Shame that FreeCAD has no object-local variables that can be used in constraints 😕

phaseloop, 2025-04-20:
Well, we have named constraints themselves

knipknap, 2025-04-20:
True, named constraints may be a great option from a user XP point of view. no jumping back&forth between the tool sketch and the values.

knipknap, 2025-04-20:
Alright, I made tons of progress on the new ToolBitShape introduction. All tests pass (including tons of new ones I added), and the UI appears to mostly work. Most of the rest is testing, debugging, and cleanup, which I will continue and hopefully finish tomorrow.

One more question, or rather a proposal:

The ToolBit still has some code for handling a **change of shape on an existing tool**. I propose to remove that, because:
  
1. I don't see how making this change could make any sense, as you would also likely want to change all other tool parameters.
2. It would simplify the code base, remove some of the most "special cased" code.
3. Instead of "promoting an existing tool to a different type" I think it is better to introduce a "replace tool" feature.

But for now, I propose to remove the option to change the shape of an already added tool. Any objections?

knipknap, 2025-04-22:
The first PR is now ready. Details of what is included are described in the PR.

Please let me know if this can be accepted - I will take a break from working on BTL integration for now.

knipknap, 2025-04-23:
I am still wrecking my brain trying to come up with an _iterative_ way to integrate BTL, but no matter which end I start from there I always end up in a dependency hell.
I am tempted to throw the complete Tool/ folder out (except for the refactored ToolBit and ToolBitShape and the Controller) and integrate the existing BTL more or less as-is.

This would of course make it hard to track any changes. Would it be ok anyway from your point of view?

knipknap, 2025-04-24:
Alright, Thanks for the direction!
I think I finally came up with possible iterative next steps.

1. [ ] Get PR #20924 approved 😏 

2. [ ] Implement the following class structure:

  ```mermaid
  classDiagram
      direction LR
  
      class ToolBitShapeBallEnd
      class ToolBitShapeBullnose
      class ToolBitShapeChamfer
      class ToolBitShapeDovetail
      class ToolBitShapeDrill
      class ToolBitShapeEndMill
      class ToolBitShapeProbe
      class ToolBitShapeReamer
      class ToolBitShapeSlittingSaw
      class ToolBitShapeTap
      class ToolBitShapeThreadMill
      class ToolBitShapeVBit
  
      class ToolBit
      class ToolBitBallEnd
      class ToolBitBullnose
      class ToolBitChamfer
      class ToolBitDovetail
      class ToolBitDrill
      class ToolBitEndMill
      class ToolBitProbe
      class ToolBitReamer
      class ToolBitSlittingSaw
      class ToolBitTap
      class ToolBitThreadMill
      class ToolBitVBit
  
      ToolBit <|-- ToolBitBallEnd : is
      ToolBit <|-- ToolBitBullnose : is
      ToolBit <|-- ToolBitChamfer : is
      ToolBit <|-- ToolBitDovetail : is
      ToolBit <|-- ToolBitDrill : is
      ToolBit <|-- ToolBitEndMill : is
      ToolBit <|-- ToolBitProbe : is
      ToolBit <|-- ToolBitReamer : is
      ToolBit <|-- ToolBitSlittingSaw : is
      ToolBit <|-- ToolBitTap : is
      ToolBit <|-- ToolBitThreadMill : is
      ToolBit <|-- ToolBitVBit : is
  
      ToolBitBallEnd *-- ToolBitShapeBallEnd : has
      ToolBitBullnose *-- ToolBitShapeBullnose : has
      ToolBitChamfer *-- ToolBitShapeChamfer : has
      ToolBitDovetail *-- ToolBitShapeDovetail : has
      ToolBitDrill *-- ToolBitShapeDrill : has
      ToolBitEndMill *-- ToolBitShapeEndMill : has
      ToolBitProbe *-- ToolBitShapeProbe : has
      ToolBitReamer *-- ToolBitShapeReamer : has
      ToolBitSlittingSaw *-- ToolBitShapeSlittingSaw : has
      ToolBitTap *-- ToolBitShapeTap : has
      ToolBitThreadMill *-- ToolBitShapeThreadMill : has
      ToolBitVBit *-- ToolBitShapeVBit : has
  ```
  
3. [ ] Introduce a shape registry
4. [ ] Add the shape browser UI

I have step 2 mostly ready, but I think it does not make sense to continue before step 1 is merged.

knipknap, 2025-04-26:
I am wondering about the tool preferences: Currently, there are a ton of preferences about the tool path that IMO are not very useful and at the same time make things hard to maintain. The full list:

- LastPathToolLibrary = "LastPathToolLibrary"
- LastPathToolShape = "LastPathToolShape"
- LastPathToolTable = "LastPathToolTable"
- LastFileToolBit = "LastFileToolBit"
- LastFileToolLibrary = "LastFileToolLibrary"
- LastFileToolShape = "LastFileToolShape"
- UseAbsoluteToolPaths = "UseAbsoluteToolPaths"

I would really like to simplify this, leaving only two settings:

- ToolDir, which should point to the root of the Tools directory.
- LastLibrary, which would contain the name or ID of the last opened library.

Why?
- Much easier to maintain, fewer special cases, easier to understand for users.
- Clearer semantics for tool files. (i.e. they always contain the name of the shape, no path needed because it's always the same).
- Predicting the future: In BTL I already had someone asking for supporting thousands of tool files, at which point a database start to become necessary. Using file names is a hindrance in that case - the semantics become even harder if you have to think about multiple directories.

Would it be ok? I would also add migration code that checks for the old settings / removes the obsolete ones. So from a user point of view, not much would change, but there would be fewer settings.

knipknap, 2025-04-28:
Good news, I finished the integration of the ShapeSelector!

Since #20924 is not yet merged, this does raise the question: Do you prefer...

- processing PR #20924 first, then process a follow-up PR?
- or one merge for all changes (i.e. that is this branch: https://github.com/knipknap/FreeCAD/tree/toolbit-subclasses)

phaseloop, 2025-04-30:
I vote for one merge for all - then let's gather people for testing session using weekly build :)

knipknap, 2025-04-30:
Thanks. I created the new PR #21078. The old PR is therefore obsolete.

knipknap, 2025-05-17:
Since the next PR (#21425) is now ready, I think I am now taking a longer break (except for bugfixes if any are requested and anything directly related to the PR).

Just one last question: I would really like to rename as follows:

   git mv src/Mod/CAM/Tools src/Mod/CAM/Assets

(And the same in all preferences.)
This would align with the terminology used in the new tool management, and align with potential future extensions (such as adding machine profiles, fixtures etc.).

But I know that names sometimes have emotional value 😄, so...  is this ok?

knipknap, 2025-06-02:
With the main integration now landed, there is one major piece of the puzzle missing to close this issue: The main library editor window.

Here I see now two options:
1. Either we adopt the mainwindow from BTL, more or less identically.
2. Or we make the dock more powerful and make the dedicated dialog window obsolete.

Personally I have come to slightly prefer option 2, and even implemented it mostly. It would look like this:

[image]

Any preference?

sliptonic, 2025-06-03:
Nice!.  Yeah, the existing icons don't work well with the dark theme.

[image]

knipknap, 2025-06-03:
Oh, I think that is something where I the theme will need to configure a white background on the cell. I am not sure how images could be "universal" for every theme otherwise.

The icons I requested are rather for the buttons. Discord thread: https://discord.com/channels/870877411049357352/1379548715412361216/1379548715412361216

closed 2025-09-15 by sliptonic