FreeCAD/FreeCAD #9467: [Problem] Occasional Exception in Path Workbench

cam72cam, 2023-05-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Path

### Problem description

When opening jobs, I occasionally get a traceback in the console and the Job panel does not function

```python
08:21:45  Traceback (most recent call last):
  File "/tmp/.mount_FreeCAxTCU3R/usr/Mod/Path/PathScripts/PathOpGui.py", line 110, in setEdit
    TaskPanel(vobj.Object, self.deleteObjectsOnReject(), page, selection)
  File "/tmp/.mount_FreeCAxTCU3R/usr/Mod/Path/PathScripts/PathOpGui.py", line 1133, in __init__
    page.initPage(obj)
  File "/tmp/.mount_FreeCAxTCU3R/usr/Mod/Path/PathScripts/PathOpGui.py", line 933, in initPage
    self.form.finishDepthLabel.hide()
<class 'RuntimeError'>: Internal C++ object (PySide2.QtWidgets.QTabWidget) already deleted.
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

sliptonic, 2023-05-04:
Is it related to an older job?   Does it happen consistently on the same job or is it less predictable?
If related to a job, can you provide a reference file?

cam72cam, 2023-05-05:
It is intermittent, on a relatively new file/job.  I suspect it is a race condition somewhere.  I'll try to record a video of the problem.

cam72cam, 2023-05-05:
https://user-images.githubusercontent.com/892136/236469497-77360cf8-fb91-4a5f-8640-cb439d015beb.mp4

mlampert, 2023-05-12:
I've seen these occasionally. I always thought there's a race somewhere in shiboken2 - with my understanding of the python reference counting this should not happen. In above example - how could the `form` or `form.finishDepthLabel` be released when the proxy is still holding a reference in self?
'probably need the input from someone with a deeper understanding of the python framework.

luzpaz, 2023-08-13:
bumping for visibility

xtemp09, 2023-10-01:
Can anyone provide steps to reproduce and an example file?

cam72cam, 2023-10-16:
I can try to provide an example file today.  It happens on most files I work with eventually, with multiple jobs.

luzpaz, 2023-11-11:
@cam72cam ping

cam72cam, 2023-11-11:
[bug_9467.zip](https://github.com/FreeCAD/FreeCAD/files/13325905/bug_9467.zip)

Here's a example file and a video demonstrating the issue.  It takes a bit of clicking back and forth to cause the race condition

cam72cam, 2023-11-11:
Sorry for the delay @luzpaz!

cam72cam, 2023-11-11:
```
10:48:18  Traceback (most recent call last):
  File "/tmp/.mount_FreeCAShmFIk/usr/Mod/Path/Path/Op/Gui/Base.py", line 108, in setEdit
    TaskPanel(vobj.Object, self.deleteObjectsOnReject(), page, selection)
  File "/tmp/.mount_FreeCAShmFIk/usr/Mod/Path/Path/Op/Gui/Base.py", line 1133, in __init__
    page.initPage(obj)
  File "/tmp/.mount_FreeCAShmFIk/usr/Mod/Path/Path/Op/Gui/Profile.py", line 50, in initPage
    self.updateVisibility()
  File "/tmp/.mount_FreeCAShmFIk/usr/Mod/Path/Path/Op/Gui/Profile.py", line 145, in updateVisibility
    self.form.processCircles.show()
<class 'AttributeError'>: 'PySide2.QtWidgets.QTableWidgetItem' object has no attribute 'show'
```

xtemp09, 2023-11-11:
> Here's a example file and a video demonstrating the issue.

> When opening jobs

I don't get anything when opening the job.

<details><summary>(1)</summary>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34935 (Git)
Build type: Debug
Branch: main
Hash: e235a1b795567a99d67e1887a1c7151e60ebe395
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: Russian/Russia (ru_RU)
```

</details>

<details><summary>(2)</summary>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34959 (Git) AppImage
Build type: Release
Branch: main
Hash: 640018bfa7878a79e1b3c220ddd488d66464dc70
Python 3.10.13, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

</details>

If I open Job → Operations → Profile, I get:

```
<Path.Area> Area.cpp(1504): hit bottom -0,0,1e-06
<Path.Area> Area.cpp(1504): hit bottom 0,0,1e-07
```

cam72cam, 2023-11-12:
If you watch the video in the zip file, its a race condition.  I have to click back and forth a number of times rapidly to cause the error to appear.   It tends to happen more on my slow netbook than on the desktop used in the video

xtemp09, 2023-11-12:
Indeed, decrease in number of processors for the virtual machine and going rapidly cause this messages to appear.

<details><summary>Traceback (most recent call last):</summary>

```

  File "/home/xtemp09/Downloads/FreeCAD/build/Mod/Path/Path/Main/Gui/Job.py", line 167, in setEdit
    self.openTaskPanel()
  File "/home/xtemp09/Downloads/FreeCAD/build/Mod/Path/Path/Main/Gui/Job.py", line 171, in openTaskPanel
    self.taskPanel = TaskPanel(self.vobj, self.deleteObjectsOnReject())
  File "/home/xtemp09/Downloads/FreeCAD/build/Mod/Path/Path/Main/Gui/Job.py", line 681, in __init__
    self.setupOps = PathSetupSheetGui.OpsDefaultEditor(
  File "/home/xtemp09/Downloads/FreeCAD/build/Mod/Path/Path/Base/Gui/SetupSheet.py", line 225, in __init__
    [
  File "/home/xtemp09/Downloads/FreeCAD/build/Mod/Path/Path/Base/Gui/SetupSheet.py", line 226, in <listcomp>
    OpTaskPanel(self.obj, name, op)
  File "/home/xtemp09/Downloads/FreeCAD/build/Mod/Path/Path/Base/Gui/SetupSheet.py", line 138, in __init__
    self.form.setWindowTitle("Op {}".format(self.name))
<class 'RuntimeError'>: Internal C++ object (PySide2.QtWidgets.QWidget) already deleted.
Traceback (most recent call last):
  File "/home/xtemp09/Downloads/FreeCAD/build/Mod/Path/Path/Main/Gui/Job.py", line 167, in setEdit
    self.openTaskPanel()
  File "/home/xtemp09/Downloads/FreeCAD/build/Mod/Path/Path/Main/Gui/Job.py", line 174, in openTaskPanel
    self.taskPanel.setupUi(activate)
  File "/home/xtemp09/Downloads/FreeCAD/build/Mod/Path/Path/Main/Gui/Job.py", line 1466, in setupUi
    self.setFields()
  File "/home/xtemp09/Downloads/FreeCAD/build/Mod/Path/Path/Main/Gui/Job.py", line 890, in setFields
    self.form.postProcessorOutputFile.setText(self.obj.PostProcessorOutputFile)
<class 'AttributeError'>: 'PySide2.QtWidgets.QWidget' object has no attribute 'setText'
Traceback (most recent call last):
  File "/home/xtemp09/Downloads/FreeCAD/build/Mod/Path/Path/Main/Gui/Job.py", line 167, in setEdit
    self.openTaskPanel()
  File "/home/xtemp09/Downloads/FreeCAD/build/Mod/Path/Path/Main/Gui/Job.py", line 174, in openTaskPanel
    self.taskPanel.setupUi(activate)
  File "/home/xtemp09/Downloads/FreeCAD/build/Mod/Path/Path/Main/Gui/Job.py", line 1544, in setupUi
    self.form.modelMoveDown.clicked.connect(
<class 'AttributeError'>: 'PySide2.QtWidgets.QWidget' object has no attribute 'clicked'
```

</details>

cam72cam, 2023-11-12:
👍, sorry it took a while to provide enough details.

phaseloop, 2024-05-24:
Bumping the issue - I'm seeing same errors when opening job settings or directly tool settings on latest master branch (0.22dev).

```
16:49:03    File "E:\release\Mod\CAM\Path\Base\Gui\SetupSheet.py", line 225, in __init__
16:49:03      [
16:49:03    File "E:\release\Mod\CAM\Path\Base\Gui\SetupSheet.py", line 226, in <listcomp>
16:49:03      OpTaskPanel(self.obj, name, op)
16:49:03    File "E:\release\Mod\CAM\Path\Base\Gui\SetupSheet.py", line 138, in __init__
16:49:03      self.form.setWindowTitle("Op {}".format(self.name))
16:49:03  RuntimeError: Internal C++ object (PySide2.QtWidgets.QWidget) already deleted.
```

luzpaz, 2024-08-27:
@xtemp09 any way forward ?

xtemp09, 2024-08-27:
@luzpaz, my Pythonic experience is so scarce, so I can't have a say in anything like this.

phaseloop, 2025-03-12:
This still happens on 1.1.0dev - sometimes when creating a new job and task panel fails to open. Sometimes when clicking "edit" on existing job.

Last year I did a lot of debugging of this issue, race condition checks, etc - without success :(

luzpaz, 2025-05-29:
@johnocfii could you test https://github.com/FreeCAD/FreeCAD/issues/9467#issuecomment-1806849586 ?

JohnOCFII, 2025-05-30:
> [@JohnOCFII](https://github.com/JohnOCFII) could you test [#9467 (comment)](https://github.com/FreeCAD/FreeCAD/issues/9467#issuecomment-1806849586) ?

Following the very nicely documented example from @cam72cam I was not able to experience the race condition.  I opened and canceled/closed the task panel for the `Profile` operation 35 times.  Of course, this is a fresh build from tonight running on a Mac. Probably lots of differences.  I did have the following (unrelated?) messages open opening this older file.

```
19:04:15  v1.1, Model-Cylinder, added 'ForceCompound' property
19:04:15  New property added to "Profile": ['NumPasses', 'Stepover']. Check its default value.
19:04:15  Stock Material property is deprecated. Removing the Material property. Please use native material system to assign a ShapeMaterial
```

```
OS: macOS 15.5
Architecture: arm64
Version: 1.1.0dev.41904 (Git)
Build date: 2025/05/29 20:43:47
Build type: Unknown
Branch: main
Hash: 864449840005def4821bf597e5f3fd062badee91
Python 3.11.12, Qt 6.7.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.7.3
shiboken 6.7.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: Touchpad/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/
Logical DPI/Physical DPI/Pixel Ratio: 72/127.5/2
Installed mods: 
  * OpenTheme 2025.5.20
  * AddonManager 2025.5.1a
  * Telemetry 1.0.1
```

[image]