FreeCAD/FreeCAD #9468: [Problem] Traceback when opening path operations

cam72cam, 2023-05-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git) AppImage
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

Path

### Problem description

Opening operations occasionally produces the following traceback:

```
08:26:58  Traceback (most recent call last):
  File "/tmp/.mount_FreeCAxTCU3R/usr/Mod/Path/PathScripts/PathJobGui.py", line 170, in setEdit
    self.openTaskPanel()
  File "/tmp/.mount_FreeCAxTCU3R/usr/Mod/Path/PathScripts/PathJobGui.py", line 177, in openTaskPanel
    self.taskPanel.setupUi(activate)
  File "/tmp/.mount_FreeCAxTCU3R/usr/Mod/Path/PathScripts/PathJobGui.py", line 1542, in setupUi
    self.form.modelMoveDown.clicked.connect(
<class 'AttributeError'>: 'PySide2.QtWidgets.QTableWidgetItem' object has no attribute 'clicked'
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-10-01:
Is this somehow related with #9467?

cam72cam, 2023-11-11:
I believe it's the same type of race condition as #9467.

luzpaz, 2025-05-29:
Going to close this. If someone can reproduce, please request for it to be re-opened