FreeCAD/FreeCAD #9471: [Problem] TechDraw_CenterLineGroup crashes with certain input

doxygen-spammer, 2023-05-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu Core 20 (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33040 (Git) Snap 691
Build type: Release
Branch: master
Hash: 1e830a8800156097ccb3ef6124dced30d4002e76
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Manipulator 1.4.9
[/code]
```

### Subproject(s) affected?

Techdraw

### Problem description

TechDraw_CenterLineGroup (Insert cosmetic center line between two lines) crashes in some regions of my shape.

I wasn’t able to produce a minimal test file / minimal steps to reproduce, so I am attaching my original file.

I tried to narrow down the problem by removing as much stuff as possible from the document.
But I didn’t find a point where the problem disappears.
The problem is still present when the shape is imported from STEP.

Part_CheckGeometry shows no errors on my shape.

[schrank_bauteile_zeichnungen_this_crashes_.zip](https://github.com/FreeCAD/FreeCAD/files/11384461/schrank_bauteile_zeichnungen_this_crashes_.zip)

### Steps to Reproduce
* Open the document
* Open the drawing page and go to the bottom of the drawing. It shows a row of 3 round and 1 rectangular features.
* Select the upper and lower line of the rectangular feature.
* Try to insert a cosmetic center line using TechDraw_CenterLineGroup.

### Anything else?

I use FreeCAD from snap, because It doesn’t work otherwise.
This is the backtrace created by FreeCAD automatically.
I get the same with `snap run --gdbserver freecad`.

```
Program received signal SIGSEGV, Segmentation fault.
#0  /lib/x86_64-linux-gnu/libc.so.6(+0x43090) [0x7f8e341c9090]
#1  0x7f8db8616549 in TechDraw::BaseGeom::findEndPoints() from /snap/freecad/691/usr/lib/TechDraw.so+0x29
#2  0x7f8db86166f5 in TechDraw::BaseGeom::getStartPoint() from /snap/freecad/691/usr/lib/TechDraw.so+0x25
#3  0x7f8db862eab6 in TechDraw::CenterLine::CenterLine(std::shared_ptr<TechDraw::BaseGeom>, int, double, double, double, double) from /snap/freecad/691/usr/lib/TechDraw.so+0xf6
#4  0x7f8db8632f4e in TechDraw::CenterLine::CenterLine(Base::Vector3<double>, Base::Vector3<double>, int, double, double, double, double) from /snap/freecad/691/usr/lib/TechDraw.so+0x9e
#5  0x7f8db86331d0 in TechDraw::CenterLine::CenterLineBuilder(TechDraw::DrawViewPart*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, int, bool) from /snap/freecad/691/usr/lib/TechDraw.so+0x240
#6  0x7f8d967902e4 in TechDrawGui::TaskCenterLine::createCenterLine() from /snap/freecad/691/usr/lib/TechDrawGui.so+0x184
#7  0x7f8d967917dd in TechDrawGui::TaskCenterLine::TaskCenterLine(TechDraw::DrawViewPart*, TechDraw::DrawPage*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, bool) from /snap/freecad/691/usr/lib/TechDrawGui.so+0x2bd
#8  0x7f8d96791f32 in TechDrawGui::TaskDlgCenterLine::TaskDlgCenterLine(TechDraw::DrawViewPart*, TechDraw::DrawPage*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > >, bool) from /snap/freecad/691/usr/lib/TechDrawGui.so+0x1a2
#9  0x7f8d966f4b53 in exec2LineCenterLine(Gui::Command*) from /snap/freecad/691/usr/lib/TechDrawGui.so+0x333
#10  0x7f8d966f6516 in CmdTechDrawCenterLineGroup::activated(int) from /snap/freecad/691/usr/lib/TechDrawGui.so+0x276
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-05-04:
I can not reproduce this.  Maybe I am not following directions well enough?

[image]

[code]
OS: Ubuntu Core 20 (X-Cinnamon/cinnamon)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33040 (Git) Snap 691
Build type: Release
Branch: master
Hash: 1e830a8800156097ccb3ef6124dced30d4002e76
Python 3.8.10, Qt 5.15.7, Coin 4.0.0, Vtk 7.1.1, OCC 7.6.3
Locale: English/Canada (en_CA)
Installed mods: 
  * A2plus 0.4.60k
[/code]

xtemp09, 2023-05-05:
I downloaded the ZIP file, unpacked, opened the first file `schrank_bauteile_zeichnungen_this_crashes_4.FCStd`. Zoomed in, selected the top and bottom line of the rectangle at the bottom of the thing. I pressed _Add Centerline between 2 Lines_ and observed the segmentation fault.

<details><summary>Here is the stacktrace</summary>

```cpp
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: invalid address (fault address: 0x30)
  * frame #0: 0x00007fffb785c23c TechDraw.so`TechDraw::BaseGeom::findEndPoints() + 58
    frame #1: 0x00007fffb785c3b5 TechDraw.so`TechDraw::BaseGeom::getStartPoint() + 45
    frame #2: 0x00007fffb786f8f8 TechDraw.so`TechDraw::CenterLine::CenterLine(std::shared_ptr<TechDraw::BaseGeom>, int, double, double, double, double) + 338
    frame #3: 0x00007fffb7872528 TechDraw.so`TechDraw::CenterLine::CenterLine(Base::Vector3<double>, Base::Vector3<double>, int, double, double, double, double) + 152
    frame #4: 0x00007fffb787291e TechDraw.so`TechDraw::CenterLine::CenterLineBuilder(TechDraw::DrawViewPart*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>>>, int, bool) + 936
    frame #5: 0x00007fffb20add51 TechDrawGui.so`TechDrawGui::TaskCenterLine::createCenterLine() + 97
    frame #6: 0x00007fffb20ae7e8 TechDrawGui.so`TechDrawGui::TaskCenterLine::TaskCenterLine(TechDraw::DrawViewPart*, TechDraw::DrawPage*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>>>, bool) + 486
    frame #7: 0x00007fffb20ae9b4 TechDrawGui.so`TechDrawGui::TaskDlgCenterLine::TaskDlgCenterLine(TechDraw::DrawViewPart*, TechDraw::DrawPage*, std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>>>, bool) + 122
    frame #8: 0x00007fffb2038cf7 TechDrawGui.so`exec2LineCenterLine(Gui::Command*) + 353
    frame #9: 0x00007fffb203925f TechDrawGui.so`CmdTechDraw2LineCenterLine::activated(int) + 199
    frame #10: 0x00007ffff64a35e8 libFreeCADGui.so`Gui::Command::_invoke(int, bool) + 440
    frame #11: 0x00007ffff64a3bb4 libFreeCADGui.so`Gui::Command::invoke(int, Gui::Command::TriggerSource) + 198
    frame #12: 0x00007ffff648fb73 libFreeCADGui.so`Gui::Action::onActivated() + 27
    frame #13: 0x00007ffff6499330 libFreeCADGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::Action::*)()>::call(void (Gui::Action::*)(), Gui::Action*, void**) + 32
    frame #14: 0x00007ffff6499370 libFreeCADGui.so`QtPrivate::QSlotObject<void (Gui::Action::*)(), QtPrivate::List<>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) + 62
    frame #15: 0x00007ffff26f1793 libQt5Core.so.5`___lldb_unnamed_symbol11821 + 1315
    frame #16: 0x00007ffff3365be6 libQt5Widgets.so.5`QAction::triggered(bool) + 70
    frame #17: 0x00007ffff33688fc libQt5Widgets.so.5`QAction::activate(QAction::ActionEvent) + 172
    frame #18: 0x00007ffff34f3142 libQt5Widgets.so.5`___lldb_unnamed_symbol16279 + 450
    frame #19: 0x00007ffff34fad39 libQt5Widgets.so.5`___lldb_unnamed_symbol16303 + 201
    frame #20: 0x00007ffff33af4ee libQt5Widgets.so.5`QWidget::event(QEvent*) + 526
    frame #21: 0x00007ffff336c713 libQt5Widgets.so.5`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 131
    frame #22: 0x00007ffff3374364 libQt5Widgets.so.5`QApplication::notify(QObject*, QEvent*) + 2692
    frame #23: 0x00007ffff646d21a libFreeCADGui.so`Gui::GUIApplication::notify(QObject*, QEvent*) + 180
    frame #24: 0x00007ffff26b9e3a libQt5Core.so.5`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 314
    frame #25: 0x00007ffff3372e47 libQt5Widgets.so.5`QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) + 439
    frame #26: 0x00007ffff33c9843 libQt5Widgets.so.5`___lldb_unnamed_symbol14622 + 3555
    frame #27: 0x00007ffff33cbfd5 libQt5Widgets.so.5`___lldb_unnamed_symbol14633 + 517
    frame #28: 0x00007ffff336c713 libQt5Widgets.so.5`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 131
    frame #29: 0x00007ffff646d21a libFreeCADGui.so`Gui::GUIApplication::notify(QObject*, QEvent*) + 180
    frame #30: 0x00007ffff26b9e3a libQt5Core.so.5`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 314
    frame #31: 0x00007ffff2b41307 libQt5Gui.so.5`QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) + 1687
    frame #32: 0x00007ffff2b16a2c libQt5Gui.so.5`QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 172
    frame #33: 0x00007fffeb9f8d6e libQt5XcbQpa.so.5`___lldb_unnamed_symbol4098 + 30
    frame #34: 0x00007ffff1b1bd3b libglib-2.0.so.0`g_main_context_dispatch + 619
    frame #35: 0x00007ffff1b706c8 libglib-2.0.so.0`___lldb_unnamed_symbol2709 + 488
    frame #36: 0x00007ffff1b193e3 libglib-2.0.so.0`g_main_context_iteration + 51
    frame #37: 0x00007ffff27130b8 libQt5Core.so.5`QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 104
    frame #38: 0x00007ffff26b875b libQt5Core.so.5`QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 299
    frame #39: 0x00007ffff26c0cf4 libQt5Core.so.5`QCoreApplication::exec() + 148
[... cut ...] ) at libc-start.c:392:3
    frame #44: 0x000055555556ad45 FreeCAD`_start + 37
```

</details>

<details><summary>Version info</summary>

```ruby
OS: Ubuntu 22.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33041 (Git)
Build type: Debug
Branch: master
Hash: b5a548e3fd931745667dd482aa39305f4db49ecf
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

<details><summary>Screenshot</summary>

[image]

</details>

doxygen-spammer, 2023-05-05:
> stop reason = signal SIGSEGV: invalid address (fault address: 0x30)
> [...] TechDraw.so`TechDraw::BaseGeom::getStartPoint() + 45

I have seen that once.
It is caused by code like this:

```
const std::vector<vertex *> &points = all_points();
return points[leftmost_point_index()];
```

if leftmost_point_index() fails and returns -1 by convention, then the 
indexing operation will read from the vector element -1.
The element -1 probably stores the length of the vector, while the shape 
happens to have 0x30 vertices.

WandererFan, 2023-05-05:
OK, managed to reproduce by wiping out my config and starting over.  Not a fix, but if you go to the advanced tab in the techdraw preferences and check "Allow crazy edges" it should work without a crash.

commit 970fd867 (2023-05-07): [TD]fix #9471 - prevent impossible centerline
closed 2023-05-07 by WandererFan