FreeCAD/FreeCAD #9473: [Issue] Adding object to another object group does not work if the object is hidden

leoheck, 2023-05-03:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 23.04 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33040 (Git) AppImage
Build type: Release
Branch: master
Hash: 1e830a8800156097ccb3ef6124dced30d4002e76
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.0
Locale: English/United States (en_US)
Installed mods: 
  * Defeaturing 1.2.0
  * Curves 0.6.8
  * fasteners 0.4.56
  * lattice2 1.0.0
  * freecad.gears 1.0.0
  * Assembly4 0.50.2-dev
  * kicadStepUpMod 10.18.8
  * ProDarkThemePreferencePack 1.0.0
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Issue description

Since drag and drop is not working on Ubuntu 23.04, a simple alternative would be updating the Group property of the Object. However, if the object that is going to be move is hidden, Freecad does not move it to the new group. 

This was tested by moving a hidden Body, to a Part. 

### Anything else?

This issue is related to this one https://github.com/FreeCAD/FreeCAD/issues/9392

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-05-04:
what process did you follow to move the body into the part, can you include a file?

leoheck, 2023-05-04:
I put it wrong in the description above (now it is fixed there. I am trying to move one *Part* inside another part. The way I am doing is as follows:

1. Select the Target Part.
2. Go to the properties of that part
3. Locate the Group property
4. Click the elipsis in the value column to edit the Group property.
5. A popup window will appear, but ignore it since it is small and over complicated. Instead, select by clicking on an object in the Objects tree. If this object is hidden (grey), when you click Ok in the popup window, it won't appear in inside of the Target Part.

leoheck, 2023-05-04:
Hey @adrianinsaval it is not the hidden part. Today I cannot move the Part to a Folder.
This is weird I thought I could do this yesterday. 

https://user-images.githubusercontent.com/1277920/236284932-afb9c42b-e660-4503-9cf2-e7f37862546e.mp4

I am still using the same Freecad.

leoheck, 2023-05-05:
Actually I cannot use this method to move Parts/Folders somehow.

Roy-043, 2023-07-16:
Can't confirm:
```
OS: Windows 8 build 9600
Word size of FreeCAD: 64-bit
Version: 0.21.0.33492 (Git)
Build type: Release
Branch: master
Hash: 6e235a4809e14674aa2cd23e68630b68998a53d4
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```

Roy-043, 2023-07-20:
Closing this as not reproducible.

closed 2023-07-20 by Roy-043