FreeCAD/FreeCAD #9476: [Issue] Sketcher constraint filter : choosing "Associated constraints" ruins the contraint list

0penBrain, 2023-05-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?p=678955#p678955

### Version

0.20 (Release)

### Full version info

```shell
OS: Ubuntu 22.04.2 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33021 (Git)
Build type: Debug
Branch: master
Hash: c77b2f33c29b9f0f24ddadae4d43900fa720758b
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Issue description

To reproduce :
* New doc, new sketch
* Draw a rectangle to get some constraints displayed
* Go to constraints filter and choose "Associated constraints"
* Go again and revert to "Selected constraints" => :exclamation: The constraint list is empty and there is nothing to do to get it back but closing and reopening the sketch.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2023-05-10:
> The constraint list is empty and there is nothing to do to get it back but closing and reopening the sketch.

You can unset "Associated constraints" so that this and "Selected constraints" are unchecked.

An alternative way to reproduce the problems is:
* unset "Selected constraints" and set it again

wwmayer, 2024-03-25:
The bug is still present in a modified form. In v0.20 and v0.21 the option "Selected constraints" is checked while in v0.22 it's unchecked by default.
1. New doc, new sketch
2. Draw a rectangle to get some constraints displayed
3. Go to constraints filter and choose "Select constraints"
4. Close the sketch and reopen it
5. Go to constraints filter and choose "Associated constraints"
6. Go again and revert to "Selected constraints"
7. The constraint list remains empty

0penBrain, 2024-04-27:
> The question is what exactly "Selected constraints" is supposed to do. If it means that only selected constraints are listed then the bug is that it shows all constraints when opening a sketch. At this point the list must be empty because no constraint can be selected.

That's exactly this. However I think it would also be sensible that "Selected constraints" is unchecked by default (user would probably wonder why constraint list is empty).
@PaddleStroke

PaddleStroke, 2024-05-01:
'Selected constraints' is supposed to show only selected constraint.

I have done some test and found the issue to be : 
1 - 'Selected constraints' seems to be saving its state. If you check it, close freecad and open freecad again, it is still checked.
2 - On opening a sketch, the checked state of 'Selected constraints' is ignored.

After further testing, it appears that ALL the checkboxes of the filter are saving their state. So if you uncheck 'Geometric', close freecad and open again. Open a sketch with a rectangle, all the geometric are still gone.
So it seems that 1) is part of this feature and that 2) is the issue in this case.

However I would say that saving the state of the filter is a bad idea/feature. It feels unlikely that you will need the same filter you used in previous document...

So there are two possible solutions : 
1 - remove the saving of the filter.
2 - fix the fact that if checked 'Selected constraints' is ignored on a new sketch edition.

I vote for 1. Do you guys agree?

maxwxyz, 2024-05-01:
yes, 1 definitely

0penBrain, 2024-05-01:
> So there are two possible solutions : 1 - remove the saving of the filter. 2 - fix the fact that if checked 'Selected constraints' is ignored on a new sketch edition.
> 
> I vote for 1. Do you guys agree?

1 as long as it means that filter setting is lost between 2 FC sessions. However I think filter setting shall be preserved along the same session. Just closing/reopening a sketch should not lose the filters. Am I correct?

PaddleStroke, 2024-05-02:
in my 1 proposal I had in mind that closing/reopening the sketch would close the filter. I am not sure how to achieve the behavior you describe. Also I think it could be confusing as a behavior.

Then what about keeping the filter saved as it currently is (and fix 2 then), but every time you open a sketch the filter are disabled, ie the checkbox on the left is unchecked. So you don't loose the fitler state, but they are disabled every time you open a sketch.

Thoughts?

maxwxyz, 2024-05-02:
I'm still in favor of the first proposal. Exiting the sketch removes filters, same as it removes selection.

0penBrain, 2024-05-02:
> Then what about keeping the filter saved as it currently is (and fix 2 then), but every time you open a sketch the filter are disabled, ie the checkbox on the left is unchecked. So you don't loose the fitler state, but they are disabled every time you open a sketch.

LGTM. Though I'm not even sure you need to fix 2, as enabling the filter may solve it. Not bad however to have a look at it so everything is clean. :wink:

0penBrain, 2025-01-05:
Confirmed issue still exists.

matthiasdanner, 2025-06-19:
I created a PR for proposal 2 of @PaddleStroke. however this has some implications on existing users (please consider the heads up in the Pull request)

PR #22072 (2025-07-17): Sketcher: Fix Constraint Filter Handling

## Issues
Fixes #9476 and #9373

It fixes the problem, that "Special Filters" of constraints are not applied correctly.
To reproduce the bug see https://github.com/FreeCAD/FreeCAD/issues/9476#issuecomment-2017384091

The problem was when initializing the filters all special filters were initialized and the first special filter (selected constraints) disabled the second filter (associated constraints). Then the second filter was initialized (which was disabled) and reset the specialFilterMode variable to none. However the selected constraints checkbox s [... cut]
closed 2025-07-17 by chennes