FreeCAD/FreeCAD #9477: [Feature Request] Displacement magnitude activated by default in the default results pipeline in FEM

FEA-eng, 2023-05-04:
### Is there an existing request for this?

- [X] I have searched the existing requests

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=77149

### Subproject(s) affected?

FEM

### Idea description

With the new improvements introduced by @donovaly, the results pipeline is automatically created after running an analysis which is great. However, what is shown is a confusing legend with no color map displayed on the model. Thus, I would suggest making the following small change - setting Field in that automatically created results pipeline settings to Displacement (Vector: Magnitude) or Displacement Magnitude (that's what that sole legend shows) instead of None by default. This way the user would see the results right after solving an analysis which is much less confusing than what is currently displayed.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

FEA-eng, 2023-05-17:
Would it be sufficient to add `obj.ViewObject.Field = 'Displacement Magnitude'` after `obj.ViewObject.DisplayMode = "Surface"` in ObjectsFem.py ? I can't build FreeCAD from source so I don't know if it's enough.

luzpaz, 2023-05-17:
@FEA-eng have you tested it ? You can edit the FEM python code realtime and test

luzpaz, 2023-05-30:
@FEA-eng if you're on Linux and the code you want to test is in python: 
1. Download the dev AppImage
2. Unpack it (https://wiki.freecad.org/AppImage#Unpacking_AppImages)
3. Find the specific python file
4. Test the AppImage, see https://wiki.freecad.org/AppImage#Modifying_AppImages

FEA-eng, 2023-05-30:
@luzpaz Unfortunately, I only have Windows. I guess that there's no such procedure for that OS, right?

luzpaz, 2023-05-30:
Maybe @donovaly knows?

FEA-eng, 2023-09-06:
@luzpaz I prepared a virtual Ubuntu machine and tested this small change based on your instruction. It works as expected so I made a PR: https://github.com/FreeCAD/FreeCAD/pull/10569

luzpaz, 2023-09-06:
Nice work! JFYI, https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue

FEA-eng, 2023-09-06:
> Nice work! JFYI, https://docs.github.com/en/issues/tracking-your-work-with-issues/linking-a-pull-request-to-an-issue

Thanks. I guess that they should add "solves" to this list. I changed it to "resolves" ;-)

Now I hope that someone will merge this. It shows some failed checks but I guess that it's not a fault of the code change. I added just a single line of code, after all. And tested that it works.

maxwxyz, 2023-10-25:
Is it possible to hide the automatically created Result Pipelines? When multiple results are expected (e.g. Eigenmodes, multiple timesteps) 10+ Pipeline results are created and activated by default. I now have to manually select each pipeline in the tree and hide it. 
Also, when I delete a pipeline before hiding it, the pipeline is gone but visible in the 3D view and cannot be hidden again.