FreeCAD/FreeCAD #9479: [Bug] Workbench selector index issue

Roy-043, 2023-05-04:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://forum.freecad.org/viewtopic.php?t=77280

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.21.0.32985 (Git)
Build type: Release
Branch: master
Hash: 5cd377f931312ede305e4c764ba2028b20eaacfd
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Dutch/Netherlands (nl_NL)
Installed mods:
```
```

### Subproject(s) affected?

Core

### Issue description

1. In the Preferences Editor go to Workbenches -> Available Workbenches.
2. Disable the first WB in the list (Arch WB in my case).
3. Press OK, but do not restart FreeCAD.
4. Select the 4th workbench from the selector.
5. Result: Correct WB is activated.
6. Select the 5th workbench from the selector.
7. Result: the 6th WB is activated.
8. Again select the 4th workbench from the selector.
9. Result: the 5th WB is activated.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-05-05:
@PaddleStroke could you take a look please?

PaddleStroke, 2023-05-09:
Yes I'm looking into it today

PaddleStroke, 2023-05-09:
So what happens is that the disabled workbench does not disappear from the workbench selector Combobox on-the-fly. However the list of workbench is actually refreshed when you change of workbench.

It appears that `WorkbenchGroup::addTo` is called : 
- On app start for QToolBar and for QMenu.
- On every workbench change for QMenu only.

So when addTo is called for QMenu it calls `refreshWorkbenchList` and recreates the view/workbenches menu.

So actually the view/workbenches menu becomes correct (without the disabled wb) after this. And then it creates the index issue for the comboxbox (as the list is updated but not the combobox items).

What I'm not sure to understand is why addTo is called for the QMenu on every wb change. I seemed to understand that only the 'not-already-created' menus / toolbars are addTo.

Anyway I think the solution to this (and it will let us get rid of the need to reboot) is to call refreshWorkbenchList and refresh the qmenu and qtoolbar on saveSettings.

PaddleStroke, 2023-05-09:
@wwmayer

I don't see the point of resizing the action group then loop to set the actions one by one.
Why not creating a new QList of QAction* and setting the new list ?

```
void WorkbenchGroup::setWorkbenchData(int index, const QString& wb)
{
    QList<QAction*> workbenches = groupAction()->actions();
    QString name = Application::Instance->workbenchMenuText(wb);
    QPixmap px = Application::Instance->workbenchIcon(wb);
    QString tip = Application::Instance->workbenchToolTip(wb);

    workbenches[index]->setObjectName(wb);
    workbenches[index]->setIcon(px);
    workbenches[index]->setText(name);
    workbenches[index]->setToolTip(tip);
    workbenches[index]->setStatusTip(tr("Select the '%1' workbench").arg(name));
    workbenches[index]->setVisible(true);
    if (index < 9) {
        workbenches[index]->setShortcut(QKeySequence(QString::fromUtf8("W,%1").arg(index+1)));
    }
}

void WorkbenchGroup::refreshWorkbenchList()
{
    QStringList enabled_wbs_list = DlgSettingsWorkbenchesImp::getEnabledWorkbenches();

    // Resize the action group.
    QList<QAction*> workbenches = groupAction()->actions();
    int numActions = workbenches.size();
    int extend = enabled_wbs_list.size() - numActions;
    if (extend > 0) {
        for (int i=0; i<extend; i++) {
            QAction* action = groupAction()->addAction(QLatin1String(""));
            action->setCheckable(true);
            action->setData(QVariant(numActions++)); // set the index
        }
    }

    // Show all enabled wb
    int index = 0;
    for (const auto& wbName : enabled_wbs_list) {
        setWorkbenchData(index++, wbName);
    }
}
```

Besides this seems to generate an issue, where we can only add more actions and not remove. Here for instance we're disabling one wb, so removing one. So the qmenu is actually not correct, the last element is duplicate : 
[image]

closed 2023-06-19 by sliptonic