FreeCAD/FreeCAD #9481: [Issue] The about info reports MacOS 10.16 on MacOS 12 etc.

Jolbas, 2023-05-05:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Forums discussion

https://github.com/FreeCAD/FreeCAD/issues/5672#issuecomment-1493439120

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.0.33071 (Git)
Build type: Release
Branch: master
Hash: 703ba9b3cf6f673e6870aa8b7e1badbdeb6c9cd2
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.0
Locale: C/Default (C)
Installed mods: 
  * kugghjul.py
  * dxf-library
  * fcgear 1.0.0
  * __pycache__
  * sheetmetal 0.2.61
  * tida.py
  * Curves 0.6.8
  * Assembly4 0.12.7
[/code]
```

### Subproject(s) affected?

None

### Issue description

On prebuilt versions of FreeCAD the about dialog reports MacOS 10.16 even though the OS is newer. When I build it myself it reports MacOS 12.6 which is correct

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Jolbas, 2023-05-05:
My locally built version is not affected
```
OS: macOS 12.6
Word size of FreeCAD: 64-bit
Version: 0.21.0.33080 (Git)
Build type: Release
Branch: master
Hash: d30946b4e5533e5354f235e0f5acc17960d812d9
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.6, OCC 7.7.0
Locale: English/United States (en_US)
Installed mods: 
  * kugghjul.py
  * dxf-library
  * fcgear 1.0.0
  * __pycache__
  * sheetmetal 0.2.61
  * tida.py
  * Curves 0.6.8
  * Assembly4 0.12.7
```

JohnOCFII, 2023-05-05:
> ```shell
> OS: macOS 10.16
> ```

>```
> OS: macOS 12.6
>```

I'm now thinking that "About" information applies to the **Build** environment, and not the current running environment.  I'd guess that means that the build environment is running on a much older macOS.

luzpaz, 2023-05-08:
Trying to dig and find where the issue is. So far I've found:  

* https://github.com/looooo/freecad-feedstock/blob/6122d9f386edf78ea2c2dde6ad578f0626821e78/.azure-pipelines/azure-pipelines-osx.yml#L6-L16
* https://github.com/looooo/freecad-feedstock/blob/843a322a4e7060fde4656ac0aa16569cf17faf3c/recipe/meta.yaml#LL92C13-L92C13

adrianinsaval, 2023-05-11:
> Edit: we should probably transfer this issue to FreeCAD-bundle ? (CC @looooo)

I don't know, perhaps we are using the wrong API to get the OS version, it should not be reporting what it was built on but what it's running on.

Jolbas, 2023-05-12:
Splash screen uses [QSysInfo::prettyProductName()](https://doc.qt.io/qt-5/qsysinfo.html#prettyProductName) which should give the info from current running OS.
https://github.com/FreeCAD/FreeCAD/blob/0a755e2b3d46a9fa852c0dfcd53f4be7d6439a6c/src/Gui/Splashscreen.cpp#L748

@luzpaz  Do you mean this code is changed somehow in the packaging?

Could it be a bug in QT?
There seems to be the same error in all available [releases](https://github.com/FreeCAD/FreeCAD-Bundle/releases). The oldest being 0.19

adrianinsaval, 2023-05-12:
so essentially this will get solved once we move to qt6

luzpaz, 2023-05-17:
We should probably add a temp ifdef hack until we do migrate to Qt6 in that case

closed 2023-05-18 by chennes