FreeCAD/FreeCAD #9500: [Problem] Edit->Transform Freezes FC

jimzim111, 2023-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 22.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33105 (Git)
Build type: Release
Branch: master
Hash: b48f6d68c1c7ca6c5fd8446d509b5dba5c54eb0e
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Under some conditions using Edit -> Transform causes FC to freeze and repeatedly print error messages in the console.  (in this case it freezes at least several minutes, even though it's working with very simple shapes).

To reproduce:
- Open attached file
- Select tree view Stock007
- Ctrl-click tree view Model-Top
- Edit menu -> Alignment
- FreeCAD freezes and must be killed

While frozen, FC continuously prints messages like "QImage::pixel: coordinate (184,-1073739052) out of range" in the console.

### Anything else?

Attached FC file:
[Alignment Freeze.zip](https://github.com/FreeCAD/FreeCAD/files/11413923/Alignment.Freeze.zip)

Forum post https://forum.freecad.org/viewtopic.php?t=78157

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Jolbas, 2023-05-07:
I have an idea what it could be. I'll have a look.

leoheck, 2023-05-08:
I confirm this on Ubuntu 23.04. I also never hear about this Alignment tool

```
[code]
OS: Ubuntu 23.04 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33110 (Git) AppImage
Build type: Release
Branch: master
Hash: 18e40941ceb36c52701a7c74ea26043ef3385701
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Defeaturing 1.2.0
  * Curves 0.6.8
  * fasteners 0.4.56
  * lattice2 1.0.0
  * freecad.gears 1.0.0
  * sheetmetal 0.2.63
  * Assembly4 0.50.2
  * kicadStepUpMod 10.18.8
  * ProDarkThemePreferencePack 1.0.0
  * Help 1.0.3
[/code]

```

adrianinsaval, 2023-05-10:
just for clarity, Edit-Transform and Edit->alignment are different things

leoheck, 2023-05-10:
What does this `Edit->Alignment` does?

commit 6f8e3f37 (2023-05-08): Gui: fix NaviCube in split views (fixes #9500)
* initialize base and hilite color with more sensible default values
* handle case if no font name is set
* apply user preferences to NaviCube