FreeCAD/FreeCAD #9507: BMI multi dimension lines | parameter "Ext Overshoot" din't work

christoph-mz, 2023-05-07:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Germany (de_DE)
Installed mods: 
  * 3D_Printing_Tools
  * BIM 2021.12.0
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

Draft

### Problem description

The parameter "Ext Overshoot" din't work, while creating dimension lines with option "continue". The parameter "Ext Overshoot" exist and can be changed but the line isn't show. Creating dimension lines without option "continue" works fine. 

### Anything else?

[image]
[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-05-07:
What's 'BMI' stand for ?

christoph-mz, 2023-05-08:
Sorry,  I meant „BIM“.

 

 

 

 

Regards

 

Christoph

yorikvanhavre, 2023-05-08:
I can't confirm this.. for me:

- Creating a single dimension, then setting the "Ext Overshoot" value to something above zero works normally
- Creating a multiple dimension with the "continue" option, then setting the "Ext Overshoot" value to something above zero works normally too

What I notice now that is annoying is that one is not able to set a default overshoot value via the Draft Style button. That should be fixed.

christoph-mz, 2023-05-08:
Hello,

 

on the propery view the values can by change, but the Ext Lines under the dimension line are still missing.

christoph-mz, 2023-05-11:
> But is the dimension line exactly on the measured points? That would explain it.
>
> Extension lines, for ex. 7cm, is a maximum value. If the distance between the dim line and the measured points is smaller than 7cm, the extension lines will have that smaller value. If the dimension line is exactly on the measured points, there is no extension line at all.
>
> Otherwise, can you attach a file that shows the problem?

Hi,

 

I hope this draw show you what I mean.

luzpaz, 2023-05-14:
@christoph-mz we don't see the file or image you attached. Please log in and drag-drop

christoph-mz, 2023-05-14:
I have attached the file. 
[test_01.zip](https://github.com/FreeCAD/FreeCAD/files/11471818/test_01.zip)

christoph-mz, 2023-05-17:
>  <https://user-images.githubusercontent.com/1136856/238319686-158e7b29-f477-459f-9b9a-60a63907225a.png>
> The cause of the problem is what I explained above. The dimension line is exactly on the measured points. As a result, there is no space to put the extension lines.
> Try changing the Y value of the Dimline property of object Dimension001 from 100 to 150. You will see the extension lines appearing.
> I don;'t think this is a bug, I would say this is the intended behaviour. AFAIK other CAD apps like AutoCAD work the same way

Hi,

thanks for your description. Now I understand the problem. But please let me know, how do I input dimension lines with option “Continue” correctly. 

If I input a single dimension line, I set point 1 and 2 and then the position of my line. But if I activate the option “Continue” I can’t input the line correctly. I do something wrong. :( I have tried many time with different order but the results are always wrong.

yorikvanhavre, 2023-05-17:
You need to place the dimension line **above** the measured points, not exactly on it. It does not matter if you are in continue mode or not. Fr ex: point 1 and 2 are on Y=0, point 3 (dim line has Y=200 (anything, but not 0)

christoph-mz, 2023-05-19:
> I just created this with in dimension continue mode... It seems to work normally?
>  <https://user-images.githubusercontent.com/1136856/239539532-865d7b70-7e1b-42a9-b98e-0ffd0ae65006.png>
>
> What I did:
>
> *	pressed the dimension button
> *	turned the continue mode on
> *	picked point 1
> *	picked point 2
> *	picked point 3 well above the line between point 1 and point 2
> *	picked point 4 right to point 2
> *	picked point 5 right to point 4
> *	pressed escape

picked point 3 well above --> works

picked point 3 well below --> not work

 

I need the line below my drawing.

Roy-043, 2023-05-21:
I think that @christoph-mz and @yorikvanhavre are using different commands. There are 3 dimension buttons available in the BIM WB. AFAICT BIM_DimensionAligned works OK, but there are issues with BIM_DimensionHorizontal and BIM_DimensionVertical. The latter 2 do not work properly. There are index issues.

Error report:
```
16:00:15  <class 'IndexError'>
16:00:15  Traceback (most recent call last):
16:00:15    File "D:\BKG_Tmp\FreeCAD_0.21_Last_Conda\Mod\Draft\draftguitools\gui_dimensions.py", line 424, in action
16:00:15      _p = DraftVecUtils.project(self.point.sub(self.node[0]),
16:00:15  IndexError: list index out of range
```

Version:
```
OS: Windows 8.1 Version 6.3 (Build 9600)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33175 (Git)
Build type: Release
Branch: master
Hash: c4a6b3a017469f7bd2c20dce44ac1264155a8488
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C) [ OS: Dutch/Netherlands (nl_NL) ]
Installed mods:
  * BIM 2021.12.0
```

PR #9632 (2023-05-24): Draft: Fixes continuing dimensions - fixes #9507
Fixes BIM horizontal/vertical dimensions used in continue mode - fixes #9507

- [x]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR
commit f68c045f (2023-05-24): Draft: Fixes continuing dimensions - fixes #9507 (#9632)
closed 2023-05-24 by Roy-043