FreeCAD/FreeCAD #9512: Deleting a Folder/Part does not delete all objects

leoheck, 2023-05-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Ubuntu 23.04 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33110 (Git) AppImage
Build type: Release
Branch: master
Hash: 18e40941ceb36c52701a7c74ea26043ef3385701
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Defeaturing 1.2.0
  * Curves 0.6.8
  * fasteners 0.4.56
  * lattice2 1.0.0
  * freecad.gears 1.0.0
  * sheetmetal 0.2.63
  * Assembly4 0.50.2
  * kicadStepUpMod 10.18.8
  * ProDarkThemePreferencePack 1.0.0
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Issue description

Deleting a Group/Part does not delete all objects recursively when replying to delete its contents as well in the following popup when deleting objects in the object tree.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-05-11:
example file?

leoheck, 2023-05-12:
Ah, I was testing here to create this file. 

Then I realized now that the problem is with the `Part>Common`.

I have a `Group` with a `Common` inside. 
Then deleting this Group does not delete the Parts inside the Common.

So, the issue is that `Part>Common` does not delete its contents nor offer me to do that.

When deleting nested things, Freecad could always offer to delete recursively. There is not a single time I want to preserve the contents of something that I did not move out before the deletion. 

The file is attached here: [delete_example.zip](https://github.com/FreeCAD/FreeCAD/files/11458365/delete_example.zip)

Steps:
1. Select all the objects on the root level. 
2. Hit Del
3. And the parts inside all the `Common` will last.

adrianinsaval, 2023-06-06:
> For instance, what happens when you delete a Part that has a Common inside? As far as I recall, the objects remain in the Object tree.

I tested and it removes everything if everything is inside the Part

luzpaz, 2023-11-26:
@leoheck can you still reproduce on development version ?

luzpaz, 2023-12-02:
@leoheck ping

leoheck, 2023-12-09:
Hi @luzpaz it still does not delete recursively.
Check this video.

https://github.com/FreeCAD/FreeCAD/assets/1277920/c4669a08-2604-495a-a7cd-76d03d4a3d00

I forgot to expand the groups, but here is a picture of where those 2 objects were.

> Before the deletion
[image]

> After the deletion
[image]

luzpaz, 2025-01-05:
Confirmed on  
```shell
OS: Manjaro Linux (KDE/plasma/xcb)
Architecture: x86_64
Version: 1.1.0dev.39750 (Git) Conda AppImage
Build type: Release
Branch: main
Hash: 7b548bee7f781837640df96831a3e7a7cfa4cf9b
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
```

### Before 
[image]

### After delete
[image]

maxwxyz, 2026-08-17:
Is this still an issue? I think a recent PR might have solved this.