FreeCAD/FreeCAD #9513: [Issue] Cannot make meshes anymore

leoheck, 2023-05-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Ubuntu 23.04 (ubuntu:GNOME/ubuntu-xorg)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33110 (Git) AppImage
Build type: Release
Branch: master
Hash: 18e40941ceb36c52701a7c74ea26043ef3385701
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: English/United States (en_US)
Installed mods: 
  * Defeaturing 1.2.0
  * Curves 0.6.8
  * fasteners 0.4.56
  * lattice2 1.0.0
  * freecad.gears 1.0.0
  * sheetmetal 0.2.63
  * Assembly4 0.50.2
  * kicadStepUpMod 10.18.8
  * ProDarkThemePreferencePack 1.0.0
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

None

### Issue description

Select Mesh WB
Select the Part to be converted to a mesh
Click Create Mesh from Shape
And then it throws this error.

The Mesh options can be edited, but only Standard and Gmsh tabs are clickable.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-05-08:
Looks like netgen is not properly configured/installed in that AppImage...

luzpaz, 2023-05-08:
Transfer this ticket to the FreeCAD-Bundle repo?

leoheck, 2023-05-08:
@luzpaz I created a new issue there on Freecad Bundle. Should I close this one? You can close it too if you have permission, please.

closed 2023-05-08 by adrianinsaval