FreeCAD/FreeCAD #9519: freecad 20 coupe

666krisna666, 2023-05-08:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 7 Version 6.1 (Build 7601: SP 1)
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: French/France (fr_FR)
Installed mods: 
  * 3D_Printing_Tools
  * fasteners 0.4.56
  * freecad.gears 1.0.0
[/code]
```

### Subproject(s) affected?

None

### Problem description

quand je fais une coupe sur un solide avec un filetage realise par percage freecad plante

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

vkhodygo, 2023-05-08:
@666krisna666 
Could you please edit your issue to be in English only? I'm afraid not all of us are that proficient in French.

WandererFan, 2023-05-08:
Tres difficile de faire des reparations sans une fichier qui demontre la probleme.
Tu peu l'attacher ici par le "drag and drop".

Very difficult to fix this without a sample file showing the problem.  You can attach it here by "drag and drop".

closed 2023-05-08 by WandererFan