FreeCAD/FreeCAD #9520: Cannot create mesh

tonealone, 2023-05-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.21.0.33110 (Git)
Build type: Release
Branch: master
Hash: 18e40941ceb36c52701a7c74ea26043ef3385701
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: C/Default (C)
[/code]
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

here's the output from console:
```
21:13:44  Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/Applications/FreeCAD.app/Contents/Resources/lib/python3.11/site-packages/shiboken2/files.dir/shibokensupport/feature.py", line 139, in _import
    return original_import(name, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
<class 'ImportError'>: dlopen(/Applications/FreeCAD.app/Contents/Resources/lib/MeshPart.so, 0x0002): Symbol not found: __ZNK17XCAFDoc_ColorTool8GetColorERK9TDF_LabelR14Quantity_Color
  Referenced from: <388A345A-BA59-33E8-99F7-BD253D5BB832> /Applications/FreeCAD.app/Contents/Resources/lib/libnglib4smesh.dylib
  Expected in:     <24B29773-5A02-3D45-883F-DCD16A0F3606> /Applications/FreeCAD.app/Contents/Resources/lib/libTKXCAF.7.7.1.dylib
21:13:46  Traceback (most recent call last):
  File "<string>", line 5, in <module>
<class 'NameError'>: name 'MeshPart' is not defined
21:13:46  name 'MeshPart' is not defined
```

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-05-09:
Duplicate of #9513 which was then superseded by https://github.com/FreeCAD/FreeCAD-Bundle/issues/173

This is most likely a packaging problem rather than a FreeCAD problem

luzpaz, 2023-05-15:
Closing, will add https://github.com/FreeCAD/FreeCAD-Bundle/issues/173 to v0.21 release list

closed 2023-05-15 by luzpaz