FreeCAD/FreeCAD #9523: `CMake Error at cMake/FreeCAD_Helpers/SetupEigen.cmake:14` encountered while attempting to build FreeCAD 0.20.2 from source tarball

capsulecorplab, 2023-05-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
N/A; can't build FreeCAD
```

### Subproject(s) affected?

None

### Issue description

Encountered following CMake error while attempting to build FreeCAD from source tarball

```
CMake Error at cMake/FreeCAD_Helpers/SetupEigen.cmake:14 (if):
  if given arguments:

    "VERSION_LESS" "3.3.1"

  Unknown arguments specified
Call Stack (most recent call first):
  CMakeLists.txt:66 (SetupEigen)

```

### Anything else?

Error occurred while attempting to run `cmake .. -DVTK_DIR=/home/kasm-default-profile/install_files/VTK-9.2.6/build` in FreeCAD build directory, `/home/kasm-default-profile/install_files/FreeCAD-0.20.2/freecad-build` (see https://github.com/tetrabiodistributed/tetra-workspace-image/blob/c71158efc224878186f2b1e320643b6ac7fbb438/playbook.yaml#L587)

Full build error can be found at https://github.com/tetrabiodistributed/tetra-workspace-image/actions/runs/4923028963/jobs/8794397233#step:5:1130

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

capsulecorplab, 2023-05-13:
> Do you have the package for eigen3 installed?

Ah, I did not! I'll close this issue as adding a line to  `apt install libeigen3-dev` as per https://github.com/tetrabiodistributed/tetra-workspace-image/pull/22/commits/3174b21917d658fbc2223f579cf5cc8b1219e004 seems to have absolved the aforementioned error. Now onto debugging another cmake error in [cMake/FreeCAD_Helpers/SetupQt.cmake](https://github.com/tetrabiodistributed/tetra-workspace-image/actions/runs/4940050163/jobs/8838027596#step:5:1150) :upside_down_face:

luzpaz, 2023-05-14:
```
CMake Error at cMake/FreeCAD_Helpers/SetupEigen.cmake:14 (if):
  if given arguments:

    "VERSION_LESS" "3.3.1"

  Unknown arguments specified
Call Stack (most recent call first):
  CMakeLists.txt:66 (SetupEigen)
```

Can we make CMake give a more explicit warning/error ?

commit cd4ac3c8 (2023-05-17): fixes #9523: CMake Error at cMake/FreeCAD_Helpers/SetupEigen.cmake
closed 2023-05-17 by wwmayer