FreeCAD/FreeCAD #9525: [Issue] FreeCAD freezed after opening a simple sketch

YosypC, 2023-05-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Europe (en_150)
Installed mods: 
  * BOLTSFC
  * fasteners
  * fcgear
  * Help 1.0.3
  * lattice2 1.0.0
  * WebTools
```

### Subproject(s) affected?

None

### Issue description

After a crash, whenever I open the sketch FC just freezes. Hard stop required.

### Anything else?

[MotherboardPlate.zip](https://github.com/FreeCAD/FreeCAD/files/11433341/MotherboardPlate.zip)
Realthunder's [related issue](https://github.com/realthunder/FreeCAD/issues/800).

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

YosypC, 2023-05-10:
> I opened the file, FreeCAD neither crashed nor froze. When loading the file it reported:

That report is totally fine since I didn't provide the linked files.

I tried 0.20 from a fresh install onto a W11 machine, and it worked fine, just like you. I brought my original installation folder to that machine, it's totally fine. I deleted the appdata folder on my original machine, the problem still occurs. I installed the latest weekly build, still occurs.
Is there any other folder I can delete, or logs I can provide?

I tried opening the program in safe mode but it immediately crashes.

xtemp09, 2023-05-10:
@YosypC, I'm not arguing, you just reported crash/freeze. I opened the file, there was no crash/freeze. How can one reproduce the problem?

xtemp09, 2023-05-10:
@YosypC, how can one reproduce the freeze step-by-step?

YosypC, 2023-05-10:
@xtemp09 I've been able to reproduce it on my original machine multiple times consistently, I'll do the same on my Surface later:

1. Create a new document. Saving it now shouldn't be any different, I did it at the end
2. Create a sketch on the XZ axis
3. Use the polyline tool to form a triangular perimeter starting from the origin with the cathetuses along the origin axes
4. Make the cathetuses equal
5. Trace a construction line (or make it later as such) starting from the origin to the middle of hypotenuse, then constraining it perpendicular to the hypotenuse
6. Apply a Distance Constrain (NOT horizontal or vertical) from the hypotenuse to the origin. You should have come up with this exact geometry:
[image]
7. Save and close the document: rebooting FC is not needed. Without closing the document, I can edit the sketch freely.
8. Open the document once again, edit the sketch: it should not freeze, however after panning the view it should finally break.

YosypC, 2023-05-10:
I cannot replicate the issue on my Surface with 0.20 .
@slizgi can you verify on your machine?

xtemp09, 2023-05-10:
I tried to reproduce it on Windows and Ubuntu with the latest git version. Then I tried to replicate it with 0.20.2 on Windows. Nothing freezes.

<details><summary>0.20.2</summary>

```ruby
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

</details>

<details><summary>0.21</summary>

```ruby
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.33110 (Git)
Build type: Release
Branch: master
Hash: 18e40941ceb36c52701a7c74ea26043ef3385701
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

</details>

<details><summary>0.21</summary>

```ruby
OS: Ubuntu 22.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33117 (Git)
Build type: Debug
Branch: master
Hash: 6765afd12b7392133bcbf9672e3c89d334b5af7b
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

YosypC, 2023-05-10:
I've experimented some more with my file and I've found that the issue does not originate from a ConstrainDistance. I initially thought it was the Perpendicular constraint following a deletion from my Surface and a re-application on my desktop, saving and reopening, the file did not freeze after a simple pan, but a resizing was needed.

1 - I then proceeded to delete all constraints, resize was needed.
2 - Then delete the separating, construction line, same behavior.
3 - I then tried deleting the hypotenuse, working fine.
4 - Came back to step file n.2, this time I could edit the sketch.
5 - Came back to file n.1, this time also froze.
6 - I then moved the single n.1 file in another folder and I had no problem, obviously with all the linking error.
7 - I also moved the original file with all constraints into an empty folder, froze.
8 - Moved another file inside (alone), and, to my surprise it doesn't freeze without resizing the sketch, although it's apparently the same file as the previous one: all constraints are present, without ConstrainDistance.

I've provided the selection of files: you can work on the "Cleared" ones, as the problem occurs constantly and I've gotten rid of the unnecessary geometry.
[Problem.zip](https://github.com/FreeCAD/FreeCAD/files/11445471/Problem.zip)
Probably more of use for who has access to a debug session instead of a common mortal like me. I am very tired, tomorrow I'll probably continue troubleshooting the cause of the freeze by gradually getting rid of constraints and/or geometry.

xtemp09, 2023-05-11:
@YosypC, I've downloaded and opened the files, still, nothing freezes or crashes. Have you updated Windows 10 recently?

Perhaps, you should run FreeCAD with _gdb_ or _lldb_ to, at least, show the stack trace. Running FreeCAD with `--write-log` option also might help.

adrianinsaval, 2023-05-11:
when only one system freezes randomly even after resetting all preferences the graphics card is the first suspect IMO, make sure your drivers are up to date or try reinstalling the driver. You could maybe try if you can reproduce the issue on the same machine on 
a live linux environment.

YosypC, 2023-05-11:
@xtemp09 Not a major version, the usual small Windows patches, but I've had this problem for weeks in RT's branch
I can't find anything useful in the logs, I also noticed it did not freeze by changing the solver inside the sketch. I could only make it work with the Resizing file (the one I could pan the sketch but not move any geometry) and not in the other.
Now I can't make it freeze with the Resizing file, whatever I do. I did not save anything.

(read below as it may be a total waste of time)
[FreeCAD.log](https://github.com/FreeCAD/FreeCAD/files/11455364/FreeCAD.log)
[FreeCAD Resizing With Solver Changed.log](https://github.com/FreeCAD/FreeCAD/files/11455366/FreeCAD.Resizing.With.Solver.Changed.log)

@adrianinsaval I assumed FreeCAD ran on the CPU without the OpenGL VBO but then I realized there's a software rendering option, and after installing the most recent drivers and failing the sketch test, I ticked the software OpenGL and everything works fine. I get consistent results with both (positive with software rendering, freezes without).
I'll make further tests with months old drivers, perhaps I'll contact AMD for a bug report.

RX 5700 XT with driver version 23.4.3

luzpaz, 2023-05-30:
Any progress on this ?

YosypC, 2023-05-30:
@luzpaz I can confirm I haven't had any crashes nor freezes in the little time of use I've had this month, but I haven't tested old drivers yet without the software rendering. You can close the thread if you deem so.

luzpaz, 2023-05-30:
Thanks! I'll close this ticket for now..but if you have time to test the old drivers and can reproduce that would be amazing if you could report back or if you encounter this bug at all.🙏

closed 2023-05-30 by luzpaz