FreeCAD/FreeCAD #9526: [Issue] "Cannot use selected object. Selected object must belong to the active body" but sketch is already inside

YosypC, 2023-05-09:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Europe (en_150)
Installed mods: 
  * BOLTSFC
  * fasteners
  * fcgear
  * Help 1.0.3
  * lattice2 1.0.0
  * WebTools
[/code]
```

### Subproject(s) affected?

None

### Issue description

[image]

### Anything else?

[Connectors.zip](https://github.com/FreeCAD/FreeCAD/files/11433625/Connectors.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

wwmayer, 2023-05-10:
It would be more helpful if you gave step by step instructions of what you did so that the issue can be reproduced. Just posting an image of the result without further information doesn't help a lot.

YosypC, 2023-05-10:
> It would be more helpful if you gave step by step instructions of what you did so that the issue can be reproduced. Just posting an image of the result without further information doesn't help a lot.

Apologies, I missed the instructions!

I just tried to pad the sketch. **But** I now realized that I zipped a working backup instead of the screenshot file. I am afraid to tell you it's now lost in time since everything works now with that I have at the moment. You can reopen the issue if you all find something new.

closed 2023-05-10 by YosypC