FreeCAD/FreeCAD #9532: [Problem] Changing a Binder's Offset value moves it flat on Origin plane

YosypC, 2023-05-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.33110 (Git)
Build type: Release
Branch: master
Hash: 18e40941ceb36c52701a7c74ea26043ef3385701
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/Europe (en_150)
```

### Subproject(s) affected?

None

### Problem description

https://github.com/FreeCAD/FreeCAD/assets/50027752/7a2c7af6-095d-46ac-8260-267f85541da2

### Anything else?

[MotherboardPlate.zip](https://github.com/FreeCAD/FreeCAD/files/11439845/MotherboardPlate.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

Marodut, 2023-05-21:
Hi everyone, I had a peek @YosypC 's file. 
The Subshape binder "jumps" position when changing the "Offset" property, with no regard to the actual value that's been typed.
[image]

@YosypC Can you replicate this issue in a dedicated .FCStd file, with just the bare minimum necessary to have the bug occur ?

@YosypC The archive that you have uploaded (.zip) has missing dependencies , that makes it tedious to check.
Would you upload a revised version? 

Thank you.

YosypC, 2023-06-02:
@Marodut @luzpaz I apologize for the wait, I had to take some time off.

Replicating the bug has been easier than expected: I just applied the Binder to a neutral draft face, just like my project's.
[Bug.zip](https://github.com/FreeCAD/FreeCAD/files/11637768/Bug.zip)

That binder is affected, while the other isn't.

YosypC, 2023-08-30:
Bug still present in my file in new .22dev, but I couldn't replicate it in a dedicated one.
I can't easily get rid of dependencies without breaking everything, so here's a [zip](https://github.com/FreeCAD/FreeCAD/files/12478632/Active.zip) with the whole project.

I've also experimented a bit the values, check them out:
[image]
[image]
[image]

```
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33992 (Git)
Build type: Release
Branch: master
Hash: 90bafb3a069468455d8c65d030ef8750b510c39b
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Italian/Italy (it_IT)
```

closed 2025-01-22 by github-actions[bot]