FreeCAD/FreeCAD #9539: [Problem] workbenches are repeated many times

dominikh, 2023-05-10:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: NixOS 23.05 (Stoat)
Word size of FreeCAD: 64-bit
Version: 0.21.0.Unknown
Build type: Release
Python 3.10.10, Qt 5.15.8, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.56
  * ProDarkThemePreferencePack 1.0.0
[/code]
```

### Subproject(s) affected?

None

### Problem description

Any place that lists workbenches, such as the dropdown in the toolbar, `View -> Workbench`, or the list of workbenches in the preferences, lists all of the workbenches repeated several times.

### Anything else?

`git bisect` points to 4407387ab43f31dab4957fc9b64d588f859ed0dc.

I have used the same profile (as in, the same `~/.config/FreeCAD`) with version 0.20.2 and several revisions between 0.20.2 and master. Removing `~/.config/FreeCAD` makes the problem disappear.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-05-10:
Thanks for the report: @PaddleStroke, any idea?

dominikh, 2023-05-10:
Looking at `~/.config/FreeCAD/user.cfg`, I see this

```
          <FCText Name="Ordered">ArchWorkbench,DraftWorkbench,FastenersWorkbench,FemWorkbench,InspectionWorkbench,Me
shWorkbench,OpenSCADWorkbench,PartDesignWorkbench,PartWorkbench,PathWorkbench,PointsWorkbench,ReverseEngineeringWork
bench,RobotWorkbench,SketcherWorkbench,SpreadsheetWorkbench,StartWorkbench,SurfaceWorkbench,TechDrawWorkbench,WebWor
kbench,ArchWorkbench,DraftWorkbench,FastenersWorkbench,FemWorkbench,
...
```

That is, the workbenches are repeated in the configuration file. I cannot tell if the configuration file causes the issue, or if the issue caused the configuration file. Since I've used this profile while bisecting FreeCAD for other problems it's quite possible that one of the tested commits caused the broken configuration to be written.

I've attached my configuration just in case it is helpful.
[FreeCAD.zip](https://github.com/FreeCAD/FreeCAD/files/11446522/FreeCAD.zip)

PaddleStroke, 2023-05-11:
I think it's caused by this back and forth between version that something unexpected happened in the user preference.
To avoid this in the future, we can prevent duplicates from the 'Ordered' list of workbenches.

PaddleStroke, 2023-05-11:
@dominikh can you please check if with this PR you can reproduce the bug you had ?

https://github.com/FreeCAD/FreeCAD/pull/9541

It won't fix your current user config unless you make some changes to the enabled list of workbench.

dominikh, 2023-05-11:
Saving the configuration with your PR applied writes a fixed configuration :+1:

closed 2023-05-12 by chennes