FreeCAD/FreeCAD #9540: [Problem] Part Design datum points not highlighting when selected

the-wasters-of-time, 2023-05-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods: 
  * Curves 0.6.8
  * Help 1.0.3
[/code]
```

### Subproject(s) affected?

PartDesign

### Problem description

Highlight colours are not displayed when pre-selecting or selecting data points. They are displayed when pre-selecting or selecting data lines and data planes.

Forum discussion here - https://forum.freecad.org/viewtopic.php?t=78265

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

AjinkyaDahale, 2023-05-25:
I see version number is 0.20.2. Has anyone tested with 0.21?

luzpaz, 2023-06-11:
@the-wasters-of-time can you test on dev version please ? (https://github.com/FreeCAD/FreeCAD-Bundle/releases/tag/weekly-builds)

the-wasters-of-time, 2023-07-15:
Hi Lupaz,

I have tested the weekly build, and datum points are still not highlighting
here, neither selection nor preselection. As before, datum lines and planes
do highlight. See attached simple file to test at your end.

Mal Smith.

The Wasters of Time - https://soundcloud.com/thewastersoftime

luzpaz, 2023-09-05:
@the-wasters-of-time can you paste the [About](https://wiki.freecad.org/About) info on the v0.21.1 you tested on ?

luzpaz, 2023-11-14:
@the-wasters-of-time ping

the-wasters-of-time, 2023-11-19:
Hi Lupaz,

Apologies for missing this email. See below the About info:

[code]
OS: Manjaro Linux (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.1.33694 (Git)
Build type: Release
Branch: makepkg
Hash: f6708547a9bb3f71a4aaade12109f511a72c207c
Python 3.11.5, Qt 5.15.11, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/Australia (en_AU)
Installed mods:
  * Manipulator 1.5.0
  * Render 2023.2.6
  * sheetmetal 0.2.63
  * Help 1.0.3
  * POV-Ray-Rendering
  * InventorLoader 1.4.0
  * CurvedShapes 1.0.4
  * A2plus 0.4.60n
  * Ship 2022.4.11
  * Curves 0.6.13
[/code]

Regards,

Mal Smith

The Wasters of Time - https://soundcloud.com/thewastersoftime

alwinsDen, 2024-02-15:
@luzpaz Can you assign this to me?

PR #12439 (2024-02-26): PD: Make datum points highlightable/selectable
Give the user some feedback by changing the color of the datum point marker as the user selects or hovers over it.

Closes #9540
commit 74dc66f6 (2024-02-26): PD: Make datum points highlightable/selectable (#12439)

* PD: Make datum points highlightable/selectable

Give the user some feedback by changing the color of the datum point
marker as the user selects or hovers over it.

Closes #9540
closed 2024-02-26 by adrianinsaval