FreeCAD/FreeCAD #9543: [Problem] 6dof / spacemouse center of rotation is unintuitive

dominikh, 2023-05-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
-
```

### Subproject(s) affected?

Core

### Problem description

6dof input devices for CAD, like the 3Dconnexion Space Mouse, conceptually treat the input device as the object you're manipulating. That is, movements of the input device should map to movements of the object in the CAD software. In particular, any rotations should be centered on an object's center.

In FreeCAD, however, rotations using 6dof devices seem to rotate the view around the center of the viewport. This leads to very unintuitive controls if the object isn't also at the center of the viewport.

### Anything else?

This was previously discussed at https://forum.freecad.org/viewtopic.php?style=3&t=50690.

In terms of existing FreeCAD functionality, rotations using the SpaceMouse should behave similar to as if the _Rotation mode_ was set to _Object center_, whereas right now it behaves like _Window center_.

One difference between _Object center_ and the desired SpaceMouse navigation is that with _Object center_, the object is chosen by hovering over it with the mouse, while for the SpaceMouse, it'd make more sense to rotate around the object that's closest to the center of the screen, _if_ the whole part or assembly isn't visible. This allows "flying" towards an object, keeping the center of rotation on what we're aiming for. https://www.youtube.com/watch?v=4bmv3LtxMdU has some segments displaying what intuitive motion with a SpaceMouse should look like.

I am probably glossing over some minor details that further improve SpaceMouse navigation, but just basic rotation around object centers would already greatly improve over what we have now.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

andreymrt, 2023-06-22:
One thing I found helpful in SolidWorks when using SpaceMouse is that SolidWorks adds a small ephemeral marker in model space at the point of rotation while the SpaceMouse is being operated.
That rotation marker also seems to be fixed in model space during translations, and resets shortly after letting go of the SpaceMouse

Cysign, 2023-08-18:
Just bought a Spacenavigator, which I thought it was broken at first.
When I opened FreeCad there was a slightly drift through the Spacenavigator. So it's completly useless. Also the behavior described above still is the same.

So I just ooened blender, where the Spacenavigator works properly. So the FreeCad implementation of the 3dconnexion devices is still broken (checked this several years ago and that' s what made me NOT using FreeCad... later I sold the Spaceexplorer I used to test). But now, several years later the same bad behavior still exists :(

What's the problem with it? Can't we just have a look how Blender implements this inputdevices and improove it in FreeCad?

dominikh, 2023-08-18:
> What's the problem with it? Can't we just have a look how Blender implements this inputdevices and improove it in FreeCad?

My guess would be that the problem is the empty intersection of people with such a device, people who use FreeCAD, and people who have the time and willingness to fix the code.

Cysign, 2023-09-10:
In general the Space Navigator does work in FreeCad. But Axis are swapped. And it's too fast.
Where can I find the code that is responsible for the Space Navigator implementation?
And is there a tutorial for compiling FreeCad? I'm not a programmer...but probably we can make it work :)

luzpaz, 2023-09-17:
https://github.com/FreeCAD/FreeCAD/tree/master/src/Gui/3Dconnexion

numo68, 2023-12-21:
@Cysign 
> But Axis are swapped. And it's too fast.

The sensitivity, axis flipping and other things can be tweaked through the Tools / Customize / Spaceball Motion menu, are stored as BaseApp.Spaceball.Motion group in user.cfg and processed in src/Gui/GuiApplicationNativeEventAware.cpp

> And is there a tutorial for compiling FreeCad?

Which platform are you on?

In any case, this is IMO not what the original issue posted by @dominikh is about, and should have an own issue if something is not right.

@dominikh
> In FreeCAD, however, rotations using 6dof devices seem to rotate the view around the center of the viewport. This leads to very unintuitive controls if the object isn't also at the center of the viewport.

Is there any thread in the forum that arrived at the conclusion how this should work, or are there multiple opinions? I also find the current way not very intuitive, but I don't know any other 3D tool to be able to compare, nor am I a FreeCAD power user.

dominikh, 2023-12-21:
@numo68 I've come up with my opinion based on how the device functions on Windows with software like Inventor and SketchUp, using the official drivers. The "you're manipulating the object" interpretation is based on various marketing and training material. You can also see the expected behavior in various YouTube videos demonstrating the device.

I am not aware of any discussions on FreeCAD's behavior.

luzpaz, 2023-12-22:
JFYI, there should be a bunch of forum threads on this topic.

glabifrons, 2024-06-08:
> So probably I'm going to sell the Spacemouse, again... I don't like hardware that isn't compatible with all of my tools (in this case Blender and FreeCad).

If you haven't sold it yet, check out #12644 as it is exactly what you're talking about. Ticking the "invert YZ" checkbox swaps them to a usable state that should match other software.

FEA-eng, 2024-06-29:
With #12929, this should no longer be a problem, at least on Windows,

maxwxyz, 2024-11-03:
@dominikh is this still an issue with the latest versions?

Ely31, 2024-11-03:
yes, it is

luzpaz, 2025-05-29:
@maxwxyz does the 3dconnexion dev know about this ?

maxwxyz, 2025-05-30:
@pskowronskiTDx FYI

kevenwyld, 2025-06-05:
So I wanted to share my experience because I didn't see a clear example of the problem in this thread.  Or at least of the problem as I see it.

In FreeCAD the center of rotation does NOT move, but the part does.  That seems to be the difference compared to other platforms/software.

Here is a video of FreeCAD on Linux where I move only RX (and a tiny bit TY, but it doesn't impact this test, it's just hard not to do).  Rotating the part away from me.  I then I move the part towards me with TZ, and make the same RX/TY rotation again.  You can see in the video that the part moves away from the center of rotation when I translate it in Z.  You can see me reset the center by middle-mouse click to put the center back in the center of the part.

https://github.com/user-attachments/assets/ab5e7403-91b7-4cec-b0c4-a7a09bb6c00b

Now here's a video of the same thing in Fusion360 on windows.  Unfortunately windows lacks a real time display of the active axes of the device, but I'm making the same movements as I did on Linux in FC.  In this video you can see that when moving the part in Z the center of rotation stays with the part, or rather the part does not move away from the center when translating.

https://github.com/user-attachments/assets/52c372f2-bcf5-43d9-87f5-caef2516f95a

This phenomenon occurs in FreeCAD in every translation direction.   I believe this is the root of most of the frustration when people compare FreeCAD's 3d mouse behavior to that of other applications and other platforms.

AFAICT there is no setting to adjust translating or not translating the center of rotation when you translate the part.  On windows this is handled by the application specific settings in the 3dconnexion driver settings under "rotation center" which can be either "auto" or "use selected item".  But neither allows moving the object away from the center as far as I can tell.

I hope this helps move this discussion forward.

kevenwyld, 2025-10-08:
Tested in:

```
OS: Arch Linux (sway/sway/wayland)
Architecture: x86_64
Version: 1.1.0dev.14555 (Git shallow) AppImage
Build date: 2025/10/07 22:49:55
Build type: Release
Branch: main
Hash: 68f083a6d169ec523fa9d46b7cc40addb1e4ffbe
Python 3.11.13, Qt 6.8.3, Coin 4.0.3, Vtk 9.3.1, boost 1_86, Eigen3 3.4.0, PySide 6.8.3
shiboken 6.8.3, xerces-c 3.3.0, IfcOpenShell 0.8.2, OCC 7.8.1
Locale: English/United States (en_US)
Navigation Style/Orbit Style/Rotation Mode: CAD/Trackball/Drag at cursor
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/109.605/1
Installed mods: 
  * fasteners 0.5.42
  * Assembly3 0.12.2
  * lattice2 1.0.0
```

As before, Everything I reported in my previous https://github.com/FreeCAD/FreeCAD/issues/9543#issuecomment-2942507112 is still the case in the latest weekly build as of this writing.