FreeCAD/FreeCAD #9544: [Problem] Zooming in 3D view causes substantial decrease in frame rate

dominikh, 2023-05-11:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: NixOS 23.05 (Stoat)
Word size of FreeCAD: 64-bit
Version: 0.21.0.Unknown
Build type: Release
Python 3.10.11, Qt 5.15.9, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.2
Locale: English/United States (en_US)
Installed mods: 
  * fasteners 0.4.56
  * ProDarkThemePreferencePack 1.0.0
[/code]
```

### Subproject(s) affected?

Core

### Problem description

When zooming in and out, the frame drops from 60 fps (my display's refresh rate) to roughly 30 fps, as determined by the frame counter overlay of FreeCAD. This is independent of the complexity of the scene.

This isn't really noticeable when zooming with a mouse wheel, but it is highly noticeable when using a Space Mouse, where most movement will contain some amount of zoom, and where the unstable frame rate leads to a choppy experience.

### Anything else?

My hypothesis is that the slowdown is caused by updating the dimensions in the bottom right corner of the main window. To test this, I've patched FreeCAD to remove said updates, and this solved the issue.

```
diff --git i/src/Gui/MainWindow.cpp w/src/Gui/MainWindow.cpp
index 1cb9a32475..86c17ef3e7 100644
--- i/src/Gui/MainWindow.cpp
+++ w/src/Gui/MainWindow.cpp
@@ -2165,7 +2165,7 @@ void MainWindow::setPaneText(int i, QString text)
         showStatus(MainWindow::Pane, text);
     }
     else if (i==2) {
-        d->sizeLabel->setText(text);
+        // d->sizeLabel->setText(text);
     }
 }
```

I know that both UI layout and text rendering can be expensive, but missing the 16.6 ms target is still surprising to me.

A similar issue occurs when moving the mouse over objects, which updates the "preselected" information in the bottom left of the main window. This, however, is less of an issue with the Space Mouse, albeit still problematic when using the Space Mouse and the normal mouse simultaneously, e.g. when making edits.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-05-11:
Most likely the culprit is copying QString. Perhaps, standard `const QString &` is better.

xtemp09, 2023-10-16:
I cannot reproduce this slowdown on Windows 10.

<details><summary>Windows 10 build 19044</summary>

```ruby
OS: Windows 10 build 19044
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34594 (Git)
Build type: Release
Branch: master
Hash: e9187412d9b3b372ee4c1d7e2fc7a0bf08a2e647
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```
</details>

I cannot reproduce the slowdown on Ubuntu 22.04 with this line commented or without.

<details><summary>Ubuntu 22.04.3 LTS</summary>

```ruby
OS: Ubuntu 22.04.3 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.34667 (Git)
Build type: Debug
Branch: main
Hash: af5b8714a8371162addb97dd25c29250fcecf7ad
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: Russian/Russia (ru_RU)
```

</details>

The problem can be caused by the Space Mouse or its code in FreeCAD.

xtemp09, 2023-10-18:
@dominikh, if you replace

```cpp
void MainWindow::setPaneText(int i, QString text)
```

with

```cpp
void MainWindow::setPaneText(int i, const QString &text)
```

do you notice the slowdown? Of course, one needs to change the declaration of the function too in https://github.com/FreeCAD/FreeCAD/blob/3b0720c2b2e8233701a74ac72d211a19e825e909/src/Gui/MainWindow.h#L220

dominikh, 2023-10-18:
> The problem can be caused by the Space Mouse or its code in FreeCAD.

Note that the Space Mouse only makes it easier to notice the problem. I can reproduce it with just a normal mouse wheel, without a Space Mouse even connected to the system. Zooming by scrolling in/out rapidly is limited to 30 fps while the sizeLabel gets updated. When zoomed in so far that the label reads "0.00 mm x 0.00 mm", further zooming does happen at 60 fps. 

> void MainWindow::setPaneText(int i, const QString &text)

The proposed change made no difference.

xtemp09, 2023-10-19:
@dominikh, I pushed the commit xtemp09/FreeCAD@5acec63ee639b55e874067832725ff10cd721e67 into the branch [gui-fix4](https://github.com/xtemp09/FreeCAD/tree/gui-fix4).

Could you compile and check if it fixes the problem? To git clone it, one has to do:

```ruby
git clone -b gui-fix4 --filter=blob:none https://github.com/xtemp09/FreeCAD.git
```

luzpaz, 2024-12-15:
@OficineRobotica is this ticket still valid ? Looking at your screencasts don't show this issue AFAICT

Ely31, 2025-03-24:
This is still an issue

OS: KDE neon 6.3 (KDE/plasmax11/xcb)
Architecture: x86_64
Version: 1.1.0dev.40845 (Git)
Build date: 2025/03/18 14:59:22
Build type: Unknown
Branch: main
Hash: c1ada6bc4070df7804ccea69773b0cfa0d8226fd
Python 3.12.3, Qt 5.15.16, Coin 4.0.2, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Stylesheet/Theme/QtStyle: OpenDark.qss/OpenDark/Qt default
Logical DPI/Physical DPI/Pixel Ratio: 96/141.585/1
Installed mods: 
  * IconThemes
  * fasteners 0.5.37
  * sheetmetal 0.7.22
  * freecad.gears 1.3.0
  * OpenTheme 2024.9.1

Ely31, 2025-08-05:
Keep it open, built a few days ago and still the same ~20fps when zooming.

xtemp09, 2025-08-05:
@Ely31, could you provide _About_ info?

xtemp09, 2025-08-05:
@Ely31, do you zoom with the mouse wheel or a Space Mouse? Does navigation style affect the decrease?

----

> Build type: Unknown

Is it a release or debug build?

bombela, 2025-08-05:
> It doesn't matter if you use a space mouse or a regular mouse wheel. The
> issue is much more prominent with a space mouse however because it's
> practically impossible to not input some zoom movement while using it.
> Here's a video with the fps counter in freecad enabled, panning and
> orbiting with mouse are smooth, zooming with mouse wheel is terrible, and
> using the spacemouse at the end is the same lag as the mouse wheel.
>
>
> https://github.com/user-attachments/assets/cd59b018-6637-4ffe-8c4c-391502eedc00

Same issue here. Though mainline lags so much compared to
linkstage3-stable, the zoom issue is not the biggest of my problem.

I am using the scroll wheel on the mouse. It drops in fps when zooming.

ls3 is butter smooth at 120fps for me.

What about the mouse polling rate? or the monitor refresh rate? Maybe some
bad interaction?

My monitor is 120hz, 4k, 10bit. Mouse is 1kHz polling

ma00aem, 2025-09-14:
Hi i have also noticed this. But it seams to be better with perspective than Isometric?

FEA-eng, 2025-11-14:
Can someone confirm if this issue is still relevant?