FreeCAD/FreeCAD #9550: [Problem] FEM_ClippingPlaneAdd icon missing from GUI

chennes, 2023-05-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.33158 +1 (Git)
Build type: Release
Branch: minorTranslationFixes20230511
Hash: 80b29d2dbfed242e7320d03a4c17464340b8d9a5
Python 3.8.6+, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.5.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

As reported first here: https://github.com/FreeCAD/FreeCAD-translations/issues/197#issuecomment-1537280732

The icon is present in the source tree, but does not appear in the GUI. There is no error in the log file.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-05-24:
bumpingfor visibility

luzpaz, 2023-06-10:
bump to get this in before release

PR #9561 (2023-06-12): Fix FEM_ClippingPlaneAdd.svg (#9550)
SVG was incorrectly edited in a previous commit.   
Fixes #9550

---

- [X]  Your Pull Request meets the requirements outlined in section 5 of [CONTRIBUTING.md](https://github.com/FreeCAD/FreeCAD/blob/master/CONTRIBUTING.md) for a Valid PR
commit 9ce14bd9 (2023-06-12): Fix FEM_ClippingPlaneAdd.svg (#9550)

SVG was incorrectly edited in a previous commit.  
Fixes #9550
closed 2023-06-12 by chennes