FreeCAD/FreeCAD #9553: [Problem] Open obj file - no possibility of aborting the import operation

kaktusus, 2023-05-12:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
OS: Debian GNU/Linux 12 (bookworm) (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33110 (Git) AppImage
Build type: Release
Branch: master
Hash: 18e40941ceb36c52701a7c74ea26043ef3385701
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: Polish/Poland (pl_PL)
Installed mods: 
  * Curves 0.5.0
  * sheetmetal 0.2.56
  * Reinforcement
```

### Subproject(s) affected?

None

### Problem description

I was asked to report this problem here.
I previously wrote about it here: https://github.com/FreeCAD/FreeCAD-translations/issues/217 problems with lack of translations, by the way.

While performing this operation, it is not possible to cancel and close the window :shoc

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-05-12:
Does this apply to the Mesh importer or the Arch importer?

kaktusus, 2023-05-12:
I have not yet selected any of the available options
and at this point I cannot cancel the operation.

closed 2025-01-22 by github-actions[bot]