FreeCAD/FreeCAD #9556: Background color is broken [Problem]

sdifru, 2023-05-13:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Arch Linux
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: makepkg
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.11.3, Qt 5.15.9, Coin 4.0.1, Vtk 9.2.6, OCC 7.6.3
Locale: English/United States (en_US)
[/code]
```

### Subproject(s) affected?

None

### Problem description

I installed freecad and opened it. The background color looks like this:
[image]
The background color should be smooth, transitioning from dark blue to white.

I changed the background color and the problem persist:
[image]

My os is arch linux with swaywm as the window manager.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-05-13:
This problem comes from OpenGL. What OpenGL are you using?
Try to set on software OpenGL rendering. Edit → Preferences → Display turn on _Use software OpenGL_.

---

I prefer to run on Linux with the following command:

```ruby
LIBGL_ALWAYS_SOFTWARE=true GALLIUM_DRIVER=llvmpipe ./FreeCAD
```

This command runs FreeCAD with software rendering via Mesa. Of course, you need to install Mesa first.

sdifru, 2023-05-13:
I've set it to use opengl and the problem still exist:
[image]

I have mesa:
```
pacman -Q | grep -i mesa
libva-mesa-driver 23.0.3-1
mesa 23.0.3-1
```
I've run `LIBGL_ALWAYS_SOFTWARE=true GALLIUM_DRIVER=llvmpipe FreeCAD` and the problem still exist.

I noticed if I rotate view (hold the middle mouse button), it is a little bit laggy.

I installed the freecad from pacman, not by downloading appimage.

sdifru, 2023-05-13:
When I run either `FreeCAD` or `LIBGL_ALWAYS_SOFTWARE=true GALLIUM_DRIVER=llvmpipe FreeCAD` in a command line, I get this message: `Coin warning in glxglue_isdirect(): Couldn't get current GLX context.`

xtemp09, 2023-05-13:
I've just install Arch Linux and sway on a virtual machine. There are no lines shown in your screenshots.

<details><summary>Screenshot</summary>

[image]

</details>

FlachyJoe, 2023-05-13:
> the problem still exist.
> I noticed if I rotate view (hold the middle mouse button), it is a little bit laggy.

You run in 16bit display with no acceleration. What's your hardware ?

sdifru, 2023-05-13:
> I've just install Arch Linux and sway on a virtual machine. There are no lines shown in your screenshots.

I don't know why mine has.

> You run in 16bit display with no acceleration. What's your hardware ?

CPU: AMD Athlon 3000G (4) @ 3.500GHz
GPU: AMD ATI Radeon Vega Series / Radeon Vega Mobile Series

I used to use linux mint and fedora and freecad ran fine.

sdifru, 2023-05-14:
I downloaded the appimage file and ran it in the command line. It produced this error:
```
> ./freecad
FreeCAD 0.20.2, Libs: 0.20.2R29603 (Git)
© Juergen Riegel, Werner Mayer, Yorik van Havre and others 2001-2022
FreeCAD is free and open-source software licensed under the terms of LGPL2+ license.
FreeCAD wouldn't be possible without FreeCAD community.
  #####                 ####  ###   ####
  #                    #      # #   #   #
  #     ##  #### ####  #     #   #  #   #
  ####  # # #  # #  #  #     #####  #   #
  #     #   #### ####  #    #     # #   #
  #     #   #    #     #    #     # #   #  ##  ##  ##
  #     #   #### ####   ### #     # ####   ##  ##  ##

/tmp/.mount_freeca5FAcWD/AppRun: line 43:  8903 Aborted                 (core dumped) ${MAIN} "$@"
```

xtemp09, 2023-05-14:
I've just downloaded and run an appimage of FreeCAD on the same virtual machine with Arch Linux and sway. It works fine. I would have copied and pasted the about info, but clipboard does not work. =)

FlachyJoe, 2023-05-14:
@sdifru your display driver is misconfigured or need an update. You can check the configuration of your working Mint and apply it to Arch.
Arch documentation about ATI hardware is here: https://wiki.archlinux.org/title/ATI

sdifru, 2023-05-14:
>  would have copied and pasted the about info, but clipboard does not work.

@xtemp09 Try to install `wl-clipboard`. How did you install the arch with sway? Did you use `archinstall`?

> your display driver is misconfigured or need an update...Arch documentation about ATI hardware is here: https://wiki.archlinux.org/title/ATI

@FlachyJoe I've already had `mesa` and `amdgpu` installed:
```
$ pacman -Q | grep -i mesa
libva-mesa-driver 23.0.3-1
mesa 23.0.3-1
$ pacman -Q | grep -i amdgpu
xf86-video-amdgpu 23.0.0-1
```

I don't have `nomodeset` or `vga=` as a kernel paremeter as described [here](https://wiki.archlinux.org/title/ATI#Loading):
```
cat /proc/cmdline
BOOT_IMAGE=/vmlinuz-linux-lts root=UUID=xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx rw rootfstype=ext4 loglevel=3 quiet
```

FlachyJoe, 2023-05-14:
@sdifru 
```
cat /var/log/Xorg.0.log
cat ~/.local/share/xorg/Xorg.0.log 
```
?

sdifru, 2023-05-15:
```
$ cat /var/log/Xorg.0.log
[    10.119] (--) Log file renamed from "/var/log/Xorg.pid-467.log" to "/var/log/Xorg.0.log"
[    10.121]
X.Org X Server 1.21.1.8
X Protocol Version 11, Revision 0
[    10.121] Current Operating System: Linux archlinux 6.1.27-1-lts #1 SMP PREEMPT_DYNAMIC Mon, 01 May 2023 18:18:27 +0000 x86_64
[    10.121] Kernel command line: BOOT_IMAGE=/vmlinuz-linux-lts root=UUID=xxxxxxxx-xxxx-xxxx-xxxx-33963327xxxx rw rootfstype=ext4 loglevel=3 quiet
[    10.121]
[    10.121] Current version of pixman: 0.42.2
[    10.121]    Before reporting problems, check http://wiki.x.org
        to make sure that you have the latest version.
[    10.121] Markers: (--) probed, (**) from config file, (==) default setting,
        (++) from command line, (!!) notice, (II) informational,
        (WW) warning, (EE) error, (NI) not implemented, (??) unknown.
[    10.121] (==) Log file: "/var/log/Xorg.0.log", Time: Fri May  5 09:53:29 2023
[    10.123] (==) Using config directory: "/etc/X11/xorg.conf.d"
[    10.123] (==) Using system config directory "/usr/share/X11/xorg.conf.d"
[    10.125] (==) No Layout section.  Using the first Screen section.
[    10.125] (==) No screen section available. Using defaults.
[    10.125] (**) |-->Screen "Default Screen Section" (0)
[    10.125] (**) |   |-->Monitor "<default monitor>"
[    10.125] (==) No monitor specified for screen "Default Screen Section".
        Using a default monitor configuration.
[    10.125] (==) Automatically adding devices
[    10.125] (==) Automatically enabling devices
[    10.125] (==) Automatically adding GPU devices
[    10.125] (==) Automatically binding GPU devices
[    10.125] (==) Max clients allowed: 256, resource mask: 0x1fffff
[    10.126] (WW) The directory "/usr/share/fonts/misc" does not exist.
[    10.126]    Entry deleted from font path.
[    10.126] (WW) The directory "/usr/share/fonts/TTF" does not exist.
[    10.126]    Entry deleted from font path.
[    10.127] (WW) `fonts.dir' not found (or not valid) in "/usr/share/fonts/OTF".
[    10.127]    Entry deleted from font path.
[    10.127]    (Run 'mkfontdir' on "/usr/share/fonts/OTF").
[    10.127] (WW) The directory "/usr/share/fonts/Type1" does not exist.
[    10.127]    Entry deleted from font path.
[    10.127] (WW) The directory "/usr/share/fonts/100dpi" does not exist.
[    10.127]    Entry deleted from font path.
[    10.127] (WW) The directory "/usr/share/fonts/75dpi" does not exist.
[    10.127]    Entry deleted from font path.
[    10.127] (==) FontPath set to:

[    10.127] (==) ModulePath set to "/usr/lib/xorg/modules"
[    10.127] (II) The server relies on udev to provide the list of input devices.
        If no devices become available, reconfigure udev or disable AutoAddDevices.
[    10.127] (II) Module ABI versions:
[    10.127]    X.Org ANSI C Emulation: 0.4
[    10.127]    X.Org Video Driver: 25.2
[    10.127]    X.Org XInput driver : 24.4
[    10.127]    X.Org Server Extension : 10.0
[    10.128] (++) using VT number 1

[    10.128] (II) systemd-logind: logind integration requires -keeptty and -keeptty was not provided, disabling logind integration
[    10.129] (II) xfree86: Adding drm device (/dev/dri/card0)
[    10.129] (II) Platform probe for /sys/devices/pci0000:00/0000:00:08.1/0000:06:00.0/drm/card0
[    10.152] (--) PCI:*(6@0:0:0) 1002:15d8:1002:15d8 rev 204, Mem @ 0xe0000000/268435456, 0xf0000000/2097152, 0xfcc00000/524288, I/O @ 0x0000e000/256, BIOS @ 0x????????/131072
[    10.152] (WW) Open ACPI failed (/var/run/acpid.socket) (No such file or directory)
[    10.152] (II) LoadModule: "glx"
[    10.153] (II) Loading /usr/lib/xorg/modules/extensions/libglx.so
[    10.161] (II) Module glx: vendor="X.Org Foundation"
[    10.161]    compiled for 1.21.1.8, module version = 1.0.0
[    10.161]    ABI class: X.Org Server Extension, version 10.0
[    10.161] (II) Applying OutputClass "AMDgpu" to /dev/dri/card0
[    10.161]    loading driver: amdgpu
[    10.161] (==) Matched amdgpu as autoconfigured driver 0
[    10.161] (==) Matched ati as autoconfigured driver 1
[    10.161] (==) Matched modesetting as autoconfigured driver 2
[    10.161] (==) Matched fbdev as autoconfigured driver 3
[... 389 lines cut ...]
[    11.037] (II) config/udev: Adding input device HD-Audio Generic Front Headphone (/dev/input/event15)
[    11.037] (II) No input driver specified, ignoring this device.
[    11.037] (II) This device may have been added with another device file.
[    11.038] (II) config/udev: Adding input device PC Speaker (/dev/input/event3)
[    11.038] (II) No input driver specified, ignoring this device.
[    11.038] (II) This device may have been added with another device file.
[    19.722] (II) event1  - Power Button: device removed
[    19.751] (II) event2  - Video Bus: device removed
[    19.795] (II) event0  - Power Button: device removed
[    19.811] (II) event4  - Logitech USB Optical Mouse: device removed
[    19.855] (II) event5  - Logitech USB Keyboard: device removed
[    19.875] (II) event6  - Logitech USB Keyboard Consumer Control: device removed
[    19.915] (II) event7  - Logitech USB Keyboard System Control: device removed
[    19.931] (II) AIGLX: Suspending AIGLX clients for VT switch
[   656.801] (II) UnloadModule: "libinput"
[   656.801] (II) UnloadModule: "libinput"
[   656.801] (II) UnloadModule: "libinput"
[   656.801] (II) UnloadModule: "libinput"
[   656.801] (II) UnloadModule: "libinput"
[   656.801] (II) UnloadModule: "libinput"
[   656.802] (II) UnloadModule: "libinput"
[   656.803] (II) Server terminated successfully (0). Closing log file.
```

`~/.local/share/xorg/Xorg.0.log` doesn't exist.

sdifru, 2023-05-18:
When I looked into the `.zshrc` file, I found these lines:
```
export QT_QPA_PLATFORM=wayland↴
export QT_WAYLAND_DISABLE_WINDOWDECORATION="1"
```

After deleting it, freecad worked.

closed 2023-05-18 by sdifru