FreeCAD/FreeCAD #9579: [Problem] Сrash when creating regular polygons in Sketcher

fsv141, 2023-05-15:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: releases/FreeCAD-0-20
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
[/code]
```

### Subproject(s) affected?

None

### Problem description

Sketcher_CompCreateRegularPolygon --> Square / N-Gon / 5-gon / 6-gon / 7-gon / 8-gon / --> CRASH

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-05-16:
Я не имею ни малейшего понятия как воспроизвести это падение.

fsv141, 2023-05-16:
Да, к сожалению я сегодня тоже не могу его воспроизвести, хотя вчера чего только не делал. Естественно, FreeCAD перегружал бесчисленное число раз. Спишем на глюки ОС, Python или ещё чего-то.

closed 2023-05-16 by fsv141