FreeCAD/FreeCAD #9588: [Problem] Find maintainer(s) for PPA and to update 3rd-party dependencies in Debian/Ubuntu package ecosystem

luzpaz, 2023-05-16:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
n/a
```

### Subproject(s) affected?

Other (specify in description)

### Problem description

This is a major issue as most Ubuntu users will initially install FreeCAD for the Ubuntu store and it will break. So they'll be made aware of the PPA which is also broken. Then they'll find out about AppImages, flatpak, and snap. So lets save them the time by trying to nip this issue in the bud by finding maintainers that can:  
1. manage the PPA (development and stable)
2. update the 3rd party dependencies in the Debian/Ubuntu ecosystem 

Forum thread: https://forum.freecad.org/viewtopic.php?t=69055

### Anything else?

This issues was created as a placeholder in the v0.21 Roadmap to raise awareness.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

abdullahtahiriyo, 2023-05-18:
@luzpaz 

I quite miss the PPA framework. It was really convenient to have a reasonable up-to-date version. Good initiative. Thanks.

yorikvanhavre, 2023-06-19:
We're discussing making this a FPA job with grant. I think it can be removed from 0.21 target now as it's more a long-term issue

adrianinsaval, 2024-01-26:
I'm now maintaining the ubuntu PPA at least until january 2025. freecad-daily ppa is now working for ubuntu 20.04 and 22.04, I will get 23.10 working soon and 24.04 once it's released, they include occt 7.6.3 and will later work on shipping occt 7.8

closed 2024-01-26 by adrianinsaval