FreeCAD/FreeCAD #9598: [Problem] Some behavior of chain dimension tools in TechDraw

xtemp09, 2023-05-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu 22.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33195 (Git)
Build type: Debug
Branch: master
Hash: 8eeb095fccb130ff22b0491680cec7f0b5df9b9c
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)

OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.33110 (Git)
Build type: Release
Branch: master
Hash: 18e40941ceb36c52701a7c74ea26043ef3385701
Python 3.10.10, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Russian/Russia (ru_RU)
```

### Subproject(s) affected?

Techdraw

### Problem description

1. Insert something, for example, a Cube in Part
2. Insert a default page in TechDraw
3. Insert a projection group of the cube
4. Select a point
5. Deselect the point (nothing is selected)
6. Click _Create horizontal chain dimensions_

It causes segmentation fault and crash on Ubuntu. On Windows FreeCAD reports:

```
<Exception> Access violation
```

but continues to work.

---

Reproducible with:

- Create horizontal chain dimensions
- Create vertical chain dimensions
- Create oblique chain dimensions

### Anything else?

<details><summary>Stacktrace on Linux</summary>

```c++
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: invalid address (fault address: 0x10)
  * frame #0: 0x00007ffff5ca9dda libFreeCADGui.so`std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>>>::size(this=size=0) const at stl_vector.h:988:40
    frame #1: 0x00007ffff5e3262e libFreeCADGui.so`std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>>>>::vector(this=size=0, __x=size=0) at stl_vector.h:598:61
    frame #2: 0x00007fffafe9c6ae TechDrawGui.so`execCreateHorizChainDimension(cmd=0x000060900014cb80) at CommandExtensionDims.cpp:1198:72
    frame #3: 0x00007fffafe9e41d TechDrawGui.so`CmdTechDrawExtensionCreateChainDimensionGroup::activated(this=0x000060900014cb80, iMsg=0) at CommandExtensionDims.cpp:1439:38
    frame #4: 0x00007ffff5e6eba7 libFreeCADGui.so`Gui::Command::_invoke(this=0x000060900014cb80, id=0, disablelog=true) at Command.cpp:445:26
    frame #5: 0x00007ffff5e6e8df libFreeCADGui.so`Gui::Command::invoke(this=0x000060900014cb80, i=0, trigger=TriggerAction) at Command.cpp:410:12
    frame #6: 0x00007ffff5e59a59 libFreeCADGui.so`Gui::ActionGroup::onActivated(this=0x000060500010e140) at Action.cpp:573:22
    frame #7: 0x00007ffff5e673e6 libFreeCADGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::Action::*)()>::call(f=79 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00, o=0x000060500010e140, arg=0x00007fffffffc150) at qobjectdefs_impl.h:152:20
    frame #8: 0x00007ffff5e669ab libFreeCADGui.so`void QtPrivate::FunctionPointer<void (Gui::Action::*)()>::call<QtPrivate::List<>, void>(f=79 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00, o=0x000060500010e140, arg=0x00007fffffffc150) at qobjectdefs_impl.h:185:95
    frame #9: 0x00007ffff5e65bad libFreeCADGui.so`QtPrivate::QSlotObject<void (Gui::Action::*)(), QtPrivate::List<>, void>::impl(which=1, this_=0x00006020001a67a0, r=0x000060500010e140, a=0x00007fffffffc150, ret=0x0000000000000000) at qobjectdefs_impl.h:418:49
    frame #10: 0x00007ffff14f1793 libQt5Core.so.5`___lldb_unnamed_symbol11821 + 1315
    frame #11: 0x00007ffff2165be6 libQt5Widgets.so.5`QAction::triggered(bool) + 70
    frame #12: 0x00007ffff21688fc libQt5Widgets.so.5`QAction::activate(QAction::ActionEvent) + 172
    frame #13: 0x00007ffff226408a libQt5Widgets.so.5`___lldb_unnamed_symbol15146 + 394
    frame #14: 0x00007ffff22641e7 libQt5Widgets.so.5`QAbstractButton::mouseReleaseEvent(QMouseEvent*) + 247
    frame #15: 0x00007ffff2360d9e libQt5Widgets.so.5`QToolButton::mouseReleaseEvent(QMouseEvent*) + 14
    frame #16: 0x00007ffff21af4ee libQt5Widgets.so.5`QWidget::event(QEvent*) + 526
    frame #17: 0x00007ffff216c713 libQt5Widgets.so.5`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 131
    frame #18: 0x00007ffff2174364 libQt5Widgets.so.5`QApplication::notify(QObject*, QEvent*) + 2692
    frame #19: 0x00007ffff5e0c4bc libFreeCADGui.so`Gui::GUIApplication::notify(this=0x00007fffffffd300, receiver=0x000060300019e660, event=0x00007fffffffc920) at GuiApplication.cpp:85:40
    frame #20: 0x00007ffff14b9e3a libQt5Core.so.5`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 314
    frame #21: 0x00007ffff2172e47 libQt5Widgets.so.5`QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) + 439
    frame #22: 0x00007ffff21c8d40 libQt5Widgets.so.5`___lldb_unnamed_symbol14622 + 736
    frame #23: 0x00007ffff21cbfd5 libQt5Widgets.so.5`___lldb_unnamed_symbol14633 + 517
    frame #24: 0x00007ffff216c713 libQt5Widgets.so.5`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 131
    frame #25: 0x00007ffff5e0c4bc libFreeCADGui.so`Gui::GUIApplication::notify(this=0x00007fffffffd300, receiver=0x0000606000068880, event=0x00007fffffffce20) at GuiApplication.cpp:85:40
    frame #26: 0x00007ffff14b9e3a libQt5Core.so.5`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 314
    frame #27: 0x00007ffff1941307 libQt5Gui.so.5`QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) + 1687
    frame #28: 0x00007ffff1916a2c libQt5Gui.so.5`QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 172
    frame #29: 0x00007fffe9df8d6e libQt5XcbQpa.so.5`___lldb_unnamed_symbol4098 + 30
    frame #30: 0x00007fffef11bd3b libglib-2.0.so.0`g_main_context_dispatch + 619
    frame #31: 0x00007fffef1706c8 libglib-2.0.so.0`___lldb_unnamed_symbol2709 + 488
    frame #32: 0x00007fffef1193e3 libglib-2.0.so.0`g_main_context_iteration + 51
    frame #33: 0x00007ffff15130b8 libQt5Core.so.5`QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 104
    frame #34: 0x00007ffff14b875b libQt5Core.so.5`QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 299
    frame #35: 0x00007ffff14c0cf4 libQt5Core.so.5`QCoreApplication::exec() + 148
    frame #36: 0x00007ffff5c9e974 libFreeCADGui.so`Gui::Application::runApplication() at Application.cpp:2369:21
    frame #37: 0x000055555557edfe FreeCAD`main(argc=1, argv=0x00007fffffffdfb8) at MainGui.cpp:270:45
    frame #38: 0x00007ffff0a29d90 libc.so.6`__libc_start_call_main(main=(FreeCAD`main at MainGui.cpp:99:1), argc=1, argv=0x00007fffffffdfb8) at libc_start_call_main.h:58:16
    frame #39: 0x00007ffff0a29e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:99:1), argc=1, argv=0x00007fffffffdfb8, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdfa8) at libc-start.c:392:3
    frame #40: 0x000055555557dfe5 FreeCAD`_start + 37
```

</details>

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-05-19:
fixed by 6a7b04a441

closed 2023-05-19 by WandererFan

xtemp09, 2023-05-23:
@WandererFan, yes you fixed the crash. Now, when a user does not select anything FreeCAD shows QMessageBox. I think it's quite extensive, a message in the status bar or a notification would suffice.

What do you think?

---

And note:

If a user does not select anything, FreeCAD shows the message box. However, if a user selects a vertex, an edge or a face, FreeCAD just deselects them. I think it can be confusing for a new user. Could you add some notification in the Report view/status bar or, if you consider message box as a good way to inform user, add a message box?

WandererFan, 2023-05-24:
Showing a message box is how all the error conditions are handled.  Maybe @edi271 will consider redo-ing the error handling.

xtemp09, 2023-05-24:
Alright, what do you think about selection? I think a typical user selects an edge to insert a length dimension. However, to add chain dimensions one has to select vertices instead of edges. FreeCAD does not do anything if a user selects edges and presses chain dimensions tool button. I think FreeCAD should add chain dimensions in this case or tell the user to select vertices.

WandererFan, 2023-05-25:
I don't use this feature, but it makes sense to me that a user should be able to chain dimensions based on edges or vertexes.

xtemp09, 2023-06-16:
@WandererFan, I encountered the following sentences:

```
Can not make 2d diameter dimension from selection
```

```
Front is waiting for face finding to finish
```

I think they should be corrected.

There are too many entries of "_Can not_" in FreeCAD source code. Merriam-Webster recommends _cannot_ over _can not_. [^1]

[^1]:  https://www.merriam-webster.com/words-at-play/cannot-vs-can-not-is-there-a-difference