FreeCAD/FreeCAD #9600: [Problem] Sketcher grid auto-sizing misbehaves under certain camera conditions

0penBrain, 2023-05-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu 22.04.2 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33218 +1 (Git)
Build type: Debug
Branch: myBranch
Hash: 719339657d4b6e04535d2f777ed6e1b595bf5538
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Sketcher

### Problem description

* New document, new sketch on XY plane
* Enable grid, eventually grid auto-spacing too (shouldn't make a difference)
* Enter this in Python to setup the camera orientation

`Gui.ActiveDocument.ActiveView.setCamera('#Inventor V2.1 ascii\n\n\nOrthographicCamera {\n  viewportMapping ADJUST_CAMERA\n  position 1324.2247 -914.65112 1955.8688\n  orientation 0.67654729 0.52537006 0.51601374  0.88435292\n  nearDistance -243.53113\n  farDistance 2977.4526\n  aspectRatio 1\n  focalDistance 30.806904\n  height 1013.2249\n\n}\n')`

At this step, the grid is already partly missing on left and top of screen, and will quickly totally go out of the viewport after 3 or 4 levels of zoom-in centered at the sketch origin.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

0penBrain, 2023-05-17:
@PaddleStroke interested in tackling this ?

PaddleStroke, 2023-05-19:
These days I am fairly busy but when I have more time I can have a look if it's still pending.

maxwxyz, 2024-11-02:
Can confirm, even when the camera is slightly rotated, the grid almost disappear:
[image]

@Rexbas FYI

```
OS: Windows 11 build 26100
Word size of FreeCAD: 64-bit
Version: 1.1.0dev.39057 (Git)
Build type: Release
Branch: main
Hash: 3be681ef62bf82d32eb1f1935d99e83c48e48da3
Python 3.11.9, Qt 5.15.13, Coin 4.0.3, Vtk 9.2.6, OCC 7.7.2
Locale: German/Germany (de_DE)
Stylesheet/Theme/QtStyle: FreeCAD Dark.qss/FreeCAD Dark/Fusion
Installed mods: 
  * CfdOF 1.27.13
  * Curves 0.6.51
  * dodo 1.0.1
  * fasteners 0.5.29
  * fasteners.backup1721936159.7542229 0.5.24 (Disabled)
  * freecad.gears 1.3.0
  * Manipulator 1.5.7
  * OpenTheme 2024.9.1
  * OpticsWorkbench 1.0.25
  * Rocket 3.3.0
  * sheetmetal 0.5.3
  * Silk 0.1.5
  * Silk.backup1727166062.7227929 0.1.5 (Disabled)

```

Rexbas, 2024-11-02:
Reverting #12349 helps, but still is not perfect. That PR removes the bounding box from the grid. The problem with this is that coin3d auto clipping does not know how big the grid is when there is no bounding box. The reason the bounding box was removed is to fix the Fit all function. However, I think we should look for another way to do Fit all and bring back the bounding box to the grid. I have already looked into this some time ago but I don't remember the details (https://github.com/FreeCAD/FreeCAD/pull/12306#issuecomment-1979603251).

A similar PR like #12349 was made for Draft so it could have the same problem (#6594).

maxwxyz, 2024-11-02:
Could https://github.com/FreeCAD/FreeCAD/pull/17312 solve this too? @FlachyJoe FYI

Rexbas, 2024-11-02:
#17312 seems to help but also not perfect: 
[image]

leo-smi, 2026-10-23:
still happening

[image]

```
OS: KDE Flatpak runtime (KDE/plasmawayland/wayland)
Architecture: x86_64
Version: 1.1.4.45039 (Git)
Build date: 2026/09/28 13:46:34
Build type: Release
Branch: (HEAD detached at 1.1.4)
Hash: 4fd3bf320d9566a27e60069fc8387448aaa3a094
Python 3.13.15, Qt 6.11.2, Coin 4.0.10, Vtk 9.4.2, boost 1_86, Eigen3 3.4.1, PySide 6.11.2
shiboken 6.11.2, SMESH 7.7.1.0, xerces-c 3.3.0, IfcOpenShell 0.8.4, OCC 7.8.1
Locale: Portuguese/Brazil (pt_BR)
Navigation Style/Orbit Style/Rotation Mode: OpenSCAD/Rounded Arcball/Window center
Stylesheet/Theme/QtStyle: FreeCAD.qss/FreeCAD Dark/
Logical DPI/Physical DPI/Pixel Ratio: 96/144.85/1
Installed mods: 
  * NeoRibbon 0.3.6

```

```
OS: Linux Mint 22.3 x86_64 
Host: Aspire A514-54 V1.34 
Kernel: 7.0.0-38-generic 
Uptime: 9 hours, 40 mins 
Packages: 3999 (dpkg), 49 (flatpak) 
Shell: bash 5.2.21 
Resolution: 1920x1080 
DE: Plasma 5.27.12 
WM: kwin 
WM Theme: Breeze 
Theme: [Plasma], Breeze [GTK2/3] 
Icons: [Plasma], Slot-Dark-Icons [GTK2/3] 
Terminal: konsole 
CPU: 11th Gen Intel i5-1135G7 (8) @ 4.200GHz 
GPU: Intel TigerLake-LP GT2 [Iris Xe Graphics] 
Memory: 9531MiB / 19774MiB 

```