FreeCAD/FreeCAD #9602: [Problem] OpenSCAD CSG file not imported correctly

tcurdt, 2023-05-17:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * ExtremeProDark 2.6.4
  * ExtremeProLight 1.4.3
[/code]
```

### Subproject(s) affected?

OpenSCAD

### Problem description

I have export a CSG in OpenSCAD and imported it FreeCAD. It looks very different:

**In OpenSCAD:**
[image]

**In FreeCAD:**
[image]

So far I have not manage to break this down into a simpler testcase. That's why I have attached the original CSG file.

[linear_motor_plate.csg.zip](https://github.com/FreeCAD/FreeCAD/files/11503166/linear_motor_plate.csg.zip)

### Anything else?

I am wondering if this another story similar to https://github.com/FreeCAD/FreeCAD/issues/6271
Searching on the forum did not yield anything for me.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

chennes, 2023-05-18:
Do you have a minute to try it with a recent developer build of FreeCAD? We've made quite a few bugfixes to the OpenSCAD importer in the past year.

tcurdt, 2023-05-18:
Sorry, I didn't think of that right away. 🙄
I didn't realize my version was so old.

I can confirm with:

```
OS: macOS 12.6
Word size of FreeCAD: 64-bit
Version: 0.21.0.31483 (Git)
Build type: Release
Branch: master
Hash: 588620fd9fd138892b1cd2466f5c87d441493174
Python 3.11.0, Qt 5.15.6, Coin 4.0.0, Vtk 9.2.2, OCC 7.6.3
Locale: English/United States (en_US) [ OS: C/Default (C) ]
Installed mods: 
  * ExtremeProDark 2.6.4
  * ExtremeProLight 1.4.3
```

It works just fine now!

[image]

Thanks for the quick response!

closed 2023-05-18 by tcurdt