FreeCAD/FreeCAD #9618: [Problem] Spreadsheet: Replace Spaces with Underscore in Alias

Manuelzwei, 2023-05-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.21.0.32110 (Git)
Build type: Release
Branch: master
Hash: a000dcf60af18f489fff48d29d7f2f4b2092483f
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: German/Luxembourg (de_LU)
Installed mods: 
  * kicadStepUpMod 10.17.1
  * offline-documentation 1.0.0-alpha
[/code]
```

### Subproject(s) affected?

None

### Problem description

Please, can you automatically replace Spaces with Underscores in the Spreadsheet Alias-Names?

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-05-22:
please make a proper description of the problem you are having...

xtemp09, 2024-12-17:
Is there a program with similar interface? I have never encountered a program replacing spaces with underscores when user is typing.

I, actually, find it contradicting any interface I have seen. If a user enters an invalid name, the program must highlight it, but never replace the name with something it deems fit.

luzpaz, 2024-12-17:
Maybe we modify the ticket instead to: 
UI should report to user invalid alias syntax (via highlight or a recommended modification)

xtemp09, 2024-12-19:
I would like to bring this topic up again. If I enter an alias, make an expression with it, then introduce spaces to the alias, the expressions keep working, though I will be unable to reuse the alias containing spaces in another formula.

mwganson's PR #18567 corrects `bool PropertySheet::isValidAlias(const std::string& candidate)`. One can see that spaces are allowed, as per [page](https://wiki.freecad.org/Spreadsheet_SetAlias).

When entering an alias in the property dialog, it highlights the text in red to show that the alias is invalid, though, it is not the case when entering alias in the main form of spreadsheet (spaces, again, are allowed, the alias is just unreusable).

So, is the issue now that _FreeCAD_ does not offer hint to correct the alias? To my knowledge, _FreeCAD_ never offers hint anywhere upon entering something invalid, only pointing out that the value is incorrect, like when typing an expression.

NXTler, 2025-05-26:
Perhaps this can be fixed the same way as this #19767