FreeCAD/FreeCAD #9619: [Feature Request] Manipulation with cosmetic objects in TechDraw

xtemp09, 2023-05-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
text
```

### Subproject(s) affected?

Techdraw

### Problem description

My proposal is simple:

- Deletion of cosmetic object with the delete key on keyboard (and/or via an option in the context menu)
- Changing properties of cosmetic object with double click (and/or via an option in the context menu)

### Anything else?

### Forum thread

https://forum.freecad.org/viewtopic.php?style=10&t=78463

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

WandererFan, 2023-09-15:
Deletion of cosmetic elements via the Del key implemented by 01256623ff

luzpaz, 2023-09-17:
I've updated OP with commit and some markdown tweaks

FEA-eng, 2024-06-29:
> * [ ]  Changing properties of cosmetic object with double click (and/or via an option in the context menu)

With #11188, double-clicking on any edge (including cosmetic lines) brings the line decoration tool where you can set the appearance (style, thickness, color, visibility) of the line. So maybe this issue can be closed as completed.

luzpaz, 2024-07-27:
@xtemp09 :point_up_2:

closed 2024-07-27 by xtemp09