FreeCAD/FreeCAD #9620: Visualization after importing step file

LuigiWR, 2023-05-20:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
[code]
OS: Windows 10 Version 2009
Word size of FreeCAD: 64-bit
Version: 0.20.2.29177 +426 (Git)
Build type: Release
Branch: (HEAD detached from 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.8.10, Qt 5.15.2, Coin 4.0.1, Vtk 8.2.0, OCC 7.6.3
Locale: Italian/Italy (it_IT)
Installed mods: 
  * A2plus
  * ExplodedAssembly
  * fasteners
  * Manipulator
[/code]
```

### Subproject(s) affected?

None

### Problem description

I generated a step file of a PCB using ALTIUM DESIGNER electronic cad. I can display it correctly also with other step file visualizers such as E Drawing, IDA Step, Fusion360, Shapr3D. Only with FreeCad the visualization is wrong as you can see from the images.  In the zip file there is the original step file.

[image]
[image]
[image]
[image]
[image]
[image]
[BOXALRIT.zip](https://github.com/FreeCAD/FreeCAD/files/11522996/BOXALRIT.zip)

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-05-21:
Here is the screenshot from the git version:

[image]

LuigiWR, 2023-05-27:
OK, I'm very happy to see that you can display correctly the file... The problem is that on 3 different PC (one of witch is with WIN7), the visualization of the file is the same, and it's wrong. So, the question is: what I have to do to obtain the correct visualization?

luzpaz, 2023-05-27:
Wait for next version release or use the development releases.

luzpaz, 2023-06-10:
Closing as 'wont fix' since it's fixed in dev and probably won't be backported.

closed 2023-06-10 by luzpaz