FreeCAD/FreeCAD #9621: `CMake Error at cMake/FreeCAD_Helpers/SetupQt.cmake:18` encountered while attempting to build FreeCAD 0.20.2 from source tarball

capsulecorplab, 2023-05-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
N/A; can't build FreeCAD
```

### Subproject(s) affected?

None

### Problem description

Encountered following CMake error while attempting to build FreeCAD from source tarball

```
CMake Error at cMake/FreeCAD_Helpers/SetupQt.cmake:18 (find_package):
  By not providing "FindQt5Svg.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "Qt5Svg", but
  CMake did not find one.

  Could not find a package configuration file provided by "Qt5Svg" with any
  of the following names:

    Qt5SvgConfig.cmake
    qt5svg-config.cmake

  Add the installation prefix of "Qt5Svg" to CMAKE_PREFIX_PATH or set
  "Qt5Svg_DIR" to a directory containing one of the above files.  If "Qt5Svg"
  provides a separate development package or SDK, be sure it has been
  installed.
Call Stack (most recent call first):
  CMakeLists.txt:69 (include)
```

### Anything else?

Error occurred while attempting to run cmake .. -DVTK_DIR=/home/kasm-default-profile/install_files/VTK-9.2.6/build in FreeCAD build directory, /home/kasm-default-profile/install_files/FreeCAD-0.20.2/freecad-build (see https://github.com/tetrabiodistributed/tetra-workspace-image/blob/94ae499fa3e2b50e2483572ffadeeb648ebd7a6b/playbook.yaml#L599)

Full build error can be found at https://github.com/tetrabiodistributed/tetra-workspace-image/actions/runs/4970598989/jobs/8894568437#step:5:1141

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-05-22:
Are you sure you have all dependencies installed? If I understand correctly this is on ubuntu, check that you have all the dependencies as listed in the wiki: https://wiki.freecad.org/Compile_on_Linux (expand the debian and ubunut section), note that the list might be a little outdated

capsulecorplab, 2023-05-22:
> Are you sure you have all dependencies installed? If I understand correctly this is on ubuntu, check that you have all the dependencies as listed in the wiki: https://wiki.freecad.org/Compile_on_Linux (expand the debian and ubunut section), note that the list might be a little outdated

Thanks! I found that wiki page shortly after opening this issue and attempted to rerun the build with the suggested apt command for installing python 3 and Qt5 dependencies (see https://github.com/tetrabiodistributed/tetra-workspace-image/pull/22/commits/fcea5bb8cdf10dc329e685acd6aa0ef0f6a7e784#diff-d231edbab049480bb9012864178dcd0b9ccef7997ec46e14fc16ffe762b56f6cR579), but now I'm encountering a `No package matching 'python3-pyside2uic' is available` error (see https://github.com/tetrabiodistributed/tetra-workspace-image/actions/runs/5041899641/jobs/9042008555#step:5:1145). Will close this issue for now and open a new issue for the latter

closed 2023-05-22 by capsulecorplab

adrianinsaval, 2023-05-22:
Please make a post on the forum instead, this is not a FreeCAD problem but an Ubuntu problem