FreeCAD/FreeCAD #9622: Build failure on StringHasher test

berniev, 2023-05-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
current dev
```

### Subproject(s) affected?

None

### Problem description

[2273/4432] Building CXX object tests/CMakeFiles/Tests_run.dir/src/App/StringHasher.cpp.o
FAILED: tests/CMakeFiles/Tests_run.dir/src/App/StringHasher.cpp.o 
/usr/local/bin/ccache /usr/local/opt/ccache/bin/ccache /usr/local/opt/llvm/bin/clang++ -DBOOST_ATOMIC_DYN_LINK -DBOOST_ATOMIC_NO_LIB -DBOOST_DATE_TIME_DYN_LINK -DBOOST_DATE_TIME_NO_LIB -DBOOST_FILESYSTEM_DYN_LINK -DBOOST_FILESYSTEM_NO_LIB -DBOOST_PP_VARIADICS=1 -DBOOST_PROGRAM_OPTIONS_DYN_LINK -DBOOST_PROGRAM_OPTIONS_NO_LIB -DBOOST_REGEX_DYN_LINK -DBOOST_REGEX_NO_LIB -DBOOST_SYSTEM_DYN_LINK -DBOOST_SYSTEM_NO_LIB -DBOOST_THREAD_DYN_LINK -DBOOST_THREAD_NO_LIB -DFMT_SHARED -DGL_SILENCE_DEPRECATION -DHAVE_CONFIG_H -DPYCXX_6_2_COMPATIBILITY -DQT_CORE_LIB -DQT_XML_LIB -D_OCC64 -I/Users/bernie/CLionProjects/FreeCAD/cmake-build-freecadrelease17/tests/Tests_run_autogen/include -I/Users/bernie/CLionProjects/FreeCAD/cmake-build-freecadrelease17 -I/Users/bernie/CLionProjects/FreeCAD/cmake-build-freecadrelease17/src -I/Users/bernie/CLionProjects/FreeCAD/src -I/usr/local/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/include/python3.11 -isystem /Users/bernie/CLionProjects/FreeCAD/tests/lib/googletest/include -isystem /Users/bernie/CLionProjects/FreeCAD/tests/lib/googletest -isystem /usr/local/include -isystem /usr/local/lib/QtCore.framework/Headers -iframework /usr/local/lib -isystem /usr/local/share/qt/mkspecs/macx-clang -isystem /usr/local/lib/QtXml.framework/Headers -Wall -Wextra -Wpedantic -Wno-write-strings  -Wno-undefined-var-template -g -DFC_DEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk -mmacosx-version-min=12.6 -fcolor-diagnostics -I/usr/local/Cellar/open-mpi/4.1.4_2/include -fPIC -std=gnu++17 -MD -MT tests/CMakeFiles/Tests_run.dir/src/App/StringHasher.cpp.o -MF tests/CMakeFiles/Tests_run.dir/src/App/StringHasher.cpp.o.d -o tests/CMakeFiles/Tests_run.dir/src/App/StringHasher.cpp.o -c /Users/bernie/CLionProjects/FreeCAD/tests/src/App/StringHasher.cpp
In file included from /Users/bernie/CLionProjects/FreeCAD/tests/src/App/StringHasher.cpp:6:
In file included from /Users/bernie/CLionProjects/FreeCAD/src/App/StringHasher.h:35:
In file included from /Users/bernie/CLionProjects/FreeCAD/src/CXX/Objects.hxx:37:
In file included from /Users/bernie/CLionProjects/FreeCAD/src/CXX/WrapPython.h:58:
In file included from /usr/local/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/include/python3.11/Python.h:44:
/usr/local/opt/python@3.11/Frameworks/Python.framework/Versions/3.11/include/python3.11/object.h:239:23: error: expected member name or ';' after declaration specifiers
    PyType_Slot *slots; /* terminated by slot==0. */
    ~~~~~~~~~~~       ^

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

berniev, 2023-05-22:
Changing StringHasher.h:
```
#include <bitset>
#include <memory>

#include <QByteArray>
#include <QVector>

#include <Base/Bitmask.h>
#include <Base/Handle.h>
#include <Base/Persistence.h>
#include <CXX/Objects.hxx>
#include <utility>

#include <Base/PyObjectBase.h>
```

to:
```
#include <Base/Bitmask.h>
#include <Base/Handle.h>
#include <Base/Persistence.h>
#include <Base/PyObjectBase.h>
#include <CXX/Objects.hxx>
#include <QByteArray>
#include <QVector>
#include <bitset>
#include <memory>
#include <utility>
```

Now tests build. Will test full build in a minute.

I guess this means there is some dependency on order, implying incomplete includes somewhere?

adrianinsaval, 2023-05-23:
@chennes

berniev, 2023-05-24:
With above mods  full build is ok too.

berniev, 2023-05-25:
After experimenting with various ordering, it seems `Base/PyObjectBase.h` must precede `CXX/Objects.hxx`.
I also tried using quotes where they are supposed to be, with no effect.

BTW, the ordering in earlier post was generated by clang-format.

luzpaz, 2023-08-13:
Any traction on this ?

berniev, 2023-08-13:
It seems in the last weeks this has ceased to be a problem

closed 2023-08-13 by berniev