FreeCAD/FreeCAD #9626: [Problem][Crash] TechDraw : FC crashes if deleting a balloon while in edit

0penBrain, 2023-05-21:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Ubuntu 22.04.2 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33264 (Git)
Build type: Debug
Branch: master
Hash: 52b41fee06ca59542ddd3e948bf7d5d53703e9e5
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
```

### Subproject(s) affected?

Techdraw

### Problem description

To reproduce:
* Create a ballon annotation in a TechDraw page
* Double-click it (3D View or tree) to edit it (panel opens in the Task panel)
* Select again the balloon by clicking it (3D View or tree) and delete it
* Cancel the opened task/edit panel --> :warning: FC crashes

I also found it to crash if OKing the dialog instead of canceling it but you need to do it twice

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

xtemp09, 2023-05-21:
I tried to reproduce the same with dimension tools. The task remains opened and cannot be closed. FreeCAD responses:

```
Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: View provider detached
ControlSingleton::showDialog: Can't show  TechDrawGui::TaskDlgProjGroup  since there is already an active task dialog
Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: View provider detached
Unhandled Base::Exception caught in GUIApplication::notify.
The error message is: View provider detached
<App> AutoTransaction.cpp(136): Transaction locked, ignore new transaction 'Edit Cube'
```

Even cube's properties (a cube from _Part_) cannot be edited, i.e. FreeCAD becomes unresponsive because of the active task. Even if the project is closed the task remains active. Even if you create another project. I tried to run PartDesign in the new project and got segmentation fault.

<details><summary>Backtrace from lldb</summary>

```c++
* thread #1, name = 'FreeCAD', stop reason = signal SIGSEGV: invalid address (fault address: 0x148)
  * frame #0: 0x00007ffff6236333 libFreeCADGui.so`Gui::Document::getDocument(this=0x00007fffae77a520) const at Document.cpp:1031:15
    frame #1: 0x00007ffff6241865 libFreeCADGui.so`Gui::Document::abortCommand(this=0x00007fffae77a520) at Document.cpp:2275:5
    frame #2: 0x00007fffb0072dba TechDrawGui.so`TechDrawGui::TaskDimension::reject(this=0x00006050001229b0) at TaskDimension.cpp:157:10
    frame #3: 0x00007fffb007399d TechDrawGui.so`TechDrawGui::TaskDlgDimension::reject(this=0x00006080000b8a00) at TaskDimension.cpp:443:13
    frame #4: 0x00007ffff67e6b2f libFreeCADGui.so`Gui::TaskView::TaskView::reject(this=0x000060b00001c1a0) at TaskView.cpp:704:34
    frame #5: 0x00007ffff67eb33b libFreeCADGui.so`QtPrivate::FunctorCall<QtPrivate::IndexesList<>, QtPrivate::List<>, void, void (Gui::TaskView::TaskView::*)()>::call(f=a0 6a 7e f6 ff 7f 00 00 00 00 00 00 00 00 00 00, o=0x000060b00001c1a0, arg=0x00007fffffffa810) at qobjectdefs_impl.h:152:13
    frame #6: 0x00007ffff67eb2ad libFreeCADGui.so`void QtPrivate::FunctionPointer<void (Gui::TaskView::TaskView::*)()>::call<QtPrivate::List<>, void>(f=a0 6a 7e f6 ff 7f 00 00 00 00 00 00 00 00 00 00, o=0x000060b00001c1a0, arg=0x00007fffffffa810) at qobjectdefs_impl.h:185:13
    frame #7: 0x00007ffff67eb1f2 libFreeCADGui.so`QtPrivate::QSlotObject<void (Gui::TaskView::TaskView::*)(), QtPrivate::List<>, void>::impl(which=1, this_=0x00006020001e7d80, r=0x000060b00001c1a0, a=0x00007fffffffa810, ret=0x0000000000000000) at qobjectdefs_impl.h:418:17
    frame #8: 0x00007ffff1cf1793 libQt5Core.so.5`___lldb_unnamed_symbol11821 + 1315
    frame #9: 0x00007ffff2b0e698 libQt5Widgets.so.5`___lldb_unnamed_symbol16400 + 216
    frame #10: 0x00007ffff1cf17c8 libQt5Core.so.5`___lldb_unnamed_symbol11821 + 1368
    frame #11: 0x00007ffff2a62116 libQt5Widgets.so.5`QAbstractButton::clicked(bool) + 70
    frame #12: 0x00007ffff2a6239e libQt5Widgets.so.5`___lldb_unnamed_symbol15140 + 62
    frame #13: 0x00007ffff2a63fc4 libQt5Widgets.so.5`___lldb_unnamed_symbol15146 + 196
    frame #14: 0x00007ffff2a641e7 libQt5Widgets.so.5`QAbstractButton::mouseReleaseEvent(QMouseEvent*) + 247
    frame #15: 0x00007ffff29af4ee libQt5Widgets.so.5`QWidget::event(QEvent*) + 526
    frame #16: 0x00007ffff296c713 libQt5Widgets.so.5`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 131
    frame #17: 0x00007ffff2974364 libQt5Widgets.so.5`QApplication::notify(QObject*, QEvent*) + 2692
    frame #18: 0x00007ffff62f536d libFreeCADGui.so`Gui::GUIApplication::notify(this=0x00007fffffffd298, receiver=0x00006030001d5b90, event=0x00007fffffffb240) at GuiApplication.cpp:85:34
    frame #19: 0x00007ffff1cb9e3a libQt5Core.so.5`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 314
    frame #20: 0x00007ffff2972e47 libQt5Widgets.so.5`QApplicationPrivate::sendMouseEvent(QWidget*, QMouseEvent*, QWidget*, QWidget*, QWidget**, QPointer<QWidget>&, bool, bool) + 439
    frame #21: 0x00007ffff29c8d40 libQt5Widgets.so.5`___lldb_unnamed_symbol14622 + 736
    frame #22: 0x00007ffff29cbfd5 libQt5Widgets.so.5`___lldb_unnamed_symbol14633 + 517
    frame #23: 0x00007ffff296c713 libQt5Widgets.so.5`QApplicationPrivate::notify_helper(QObject*, QEvent*) + 131
    frame #24: 0x00007ffff62f536d libFreeCADGui.so`Gui::GUIApplication::notify(this=0x00007fffffffd298, receiver=0x0000606000068a60, event=0x00007fffffffb840) at GuiApplication.cpp:85:34
    frame #25: 0x00007ffff1cb9e3a libQt5Core.so.5`QCoreApplication::notifyInternal2(QObject*, QEvent*) + 314
    frame #26: 0x00007ffff2141307 libQt5Gui.so.5`QGuiApplicationPrivate::processMouseEvent(QWindowSystemInterfacePrivate::MouseEvent*) + 1687
    frame #27: 0x00007ffff2116a2c libQt5Gui.so.5`QWindowSystemInterface::sendWindowSystemEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 172
    frame #28: 0x00007fffe9ef8d6e libQt5XcbQpa.so.5`___lldb_unnamed_symbol4098 + 30
    frame #29: 0x00007ffff0b1bd3b libglib-2.0.so.0`g_main_context_dispatch + 619
    frame #30: 0x00007ffff0b706c8 libglib-2.0.so.0`___lldb_unnamed_symbol2709 + 488
    frame #31: 0x00007ffff0b193e3 libglib-2.0.so.0`g_main_context_iteration + 51
    frame #32: 0x00007ffff1d130b8 libQt5Core.so.5`QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) + 104
    frame #33: 0x00007ffff1cb875b libQt5Core.so.5`QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) + 299
[... cut ...]
    frame #38: 0x00007ffff1229e40 libc.so.6`__libc_start_main_impl(main=(FreeCAD`main at MainGui.cpp:99), argc=1, argv=0x00007fffffffdfb8, init=<unavailable>, fini=<unavailable>, rtld_fini=<unavailable>, stack_end=0x00007fffffffdfa8) at libc-start.c:392:3
    frame #39: 0x000055555557bc15 FreeCAD`_start + 37
```

</details>

<details><summary>Version info</summary>

```ruby
OS: Ubuntu 22.04.2 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33255 (Git)
Build type: Debug
Branch: master
Hash: e7425e8634a150b79dfe5733e9b41fc39040b433
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 9.1.0, OCC 7.5.1
Locale: English/United States (en_US)
```

</details>

WandererFan, 2023-05-22:
I can fix the crash on delete of the balloon during the dialog, but there seems to be another problem behind the scenes.  abortCommand is not terminating the active command (?):

guiDocument->hasPendingCommand;  (true)
Gui::Command::abortCommand();
guiDocument->hasPendingCommand;  (still true)

start a new command, get
20:00:01  <App> AutoTransaction.cpp(136): Transaction locked, ignore new transaction 'Create Balloon'
balloon is created, but action is not added to undo list

edit balloon, get
20:00:05  <App> AutoTransaction.cpp(136): Transaction locked, ignore new transaction 'Edit Balloon'

transaction locked state seems to persist forever.

WandererFan, 2023-05-22:
test file:

[failOnDeleteDuringTask_startPoint.FCStd.txt](https://github.com/FreeCAD/FreeCAD/files/11537810/failOnDeleteDuringTask_startPoint.FCStd.txt)

double click the balloon to start the dialog, delete balloon in tree, close dialog.

WandererFan, 2023-05-24:
https://freecad.github.io/DevelopersHandbook/technical/ObjectDeletion.html

0penBrain, 2023-05-25:
I think one issue is that currently `getInEdit` only supports editing VPs in `3DInventor` views : https://github.com/FreeCAD/FreeCAD/blob/a182c0140926e3682d1c305fceac632eee09d408/src/Gui/Document.cpp#L515

Maybe at one time we have to upgrade this mechanism so it can be used with any kind of `MDIView`. ;)

benj5378, 2023-06-15:
@WandererFan Can we close this issue?

This was a trace from before you PR.
```
Thread 1 "FreeCAD" received signal SIGSEGV, Segmentation fault.
0x00007fff48929bcb in TechDrawGui::TaskBalloon::reject (this=0x555559aea540) at /home/bensay/Development/FreeCAD/src/Mod/TechDraw/Gui/TaskBalloon.cpp:122
122	    m_parent->updateView(true);
(gdb) bt full
#0  0x00007fff48929bcb in TechDrawGui::TaskBalloon::reject()
    (this=0x555559aea540)
    at /home/bensay/Development/FreeCAD/src/Mod/TechDraw/Gui/TaskBalloon.cpp:122
        doc = 0x555557c92cb0
#1  0x00007fff4892a591 in TechDrawGui::TaskDlgBalloon::reject()
     (this=0x555558e59e00)
    at /home/bensay/Development/FreeCAD/src/Mod/TechDraw/Gui/TaskBalloon.cpp:237
#2  0x00007ffff6ed9538 in Gui::TaskView::TaskView::reject()
    (this=0x5555562529d0)
    at /home/bensay/Development/FreeCAD/src/Gui/TaskView/TaskView.cpp:704
        success = false
```

WandererFan, 2023-06-15:
We could close it, as the original complaint is resolved.  But if we do, we should make a new issue to address getInEdit being limited to 3d viewproviders.

0penBrain, 2023-06-16:
> We could close it, as the original complaint is resolved. But if we do, we should make a new issue to address getInEdit being limited to 3d viewproviders.

@wwmayer may you please comment on this ? https://github.com/FreeCAD/FreeCAD/issues/9626#issuecomment-1563234348
Do you think it's worth and doable extanding `getInEdit` so it can deal with more types of MDIViews ? Thx

luzpaz, 2023-08-16:
@wwmayer :point_up:

luzpaz, 2023-11-26:
bumping for visibility

WandererFan, 2023-11-26:
From the narrow focus of TechDraw, this ticket can be closed.  

AFAIK, TechDraw, Spreadsheet and DependencyGraph are the only areas that use a different type of MDIView (ie not a View3DInventor).

luzpaz, 2023-11-26:
Closing per WandererFan

closed 2023-11-26 by luzpaz