FreeCAD/FreeCAD #9635: [Problem]  Backup file numbering is confusing

sliptonic, 2023-05-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
N/A
```

### Subproject(s) affected?

Core

### Problem description

If the user sets the preference setting for more than one backup file to be created, the numbers assigned to backup files are confusing.  
For example, if the user sets the value to 3, the first backup is .FCStd1.  Then .FCStd2, etc.  When the available slots are filled, older files need to be renamed/deleted/overwritten.  You can end up with strange effects like this:
#1 is the newest
#3 is the next newest
#2 is the oldest.

[image]

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-05-22:
I suggest moving to the .FCBak naming scheme with dates as default instead

sliptonic, 2023-05-22:
Make .FCBak the default would be a good move.  .FCStdn has been around a long time.  It's still worth fixing the numbering order for now, declare it deprecated and get rid of it maybe in 1.0

AjinkyaDahale, 2023-05-22:
Currently the FCBak files can have arbitrary date/time formats as option. That would have to be restricted if we want to use the scheme.

I believe talking in terms of time rather than date would be better since there can be multiple versions within the same day.

Lastly, are there any suggestions of what to do with older backups? Do we let them be or try to migrate them? If the latter, this could be an extra cost to check until we support the previous version, so I'm of the opinion we might want to ignore them but ensure we don't change backup system for a long time after this.

AjinkyaDahale, 2023-05-22:
> Make .FCBak the default would be a good move. .FCStdn has been around a long time. It's still worth fixing the numbering order for now, declare it deprecated and get rid of it maybe in 1.0

So how's this for the gameplan?
1. Starting 0.22 until 1.0, check for existing `.FCStdn` files in the working directory and report add an info (maybe dialog box or new notification system) that this is deprecated and offer to migrate it based on file edit time (note that this may not be possible if the edit times in filesystems were somehow changed).
2. From 1.0, stop checking for `.FCStdn` entirely. Unfortunately someone will end up opening a very old file and it's backups will be awkward to recover. This will of course be a case whatever X.X version we choose to deprecate this.

One question here: what to do in case someone had been using `.FCBak` with a custom timestamp?

sliptonic, 2023-05-22:
versiontest.20230522-091452.FCBak

I'm not a big fan of using a period in the filename.  
Also time/date string reflects when the file was previously saved, not when the backup was created, if that makes sense.
So it appears to be renaming the original file with the date/time.FCBak and then saving the new copy under the real filename.

adrianinsaval, 2023-05-22:
> Also time/date string reflects when the file was previously saved, not when the backup was created, if that makes sense.
Isn't that better? It's the actual date when this state was saved

> So it appears to be renaming the original file with the date/time.FCBak and then saving the new copy under the real filename.

That's expected behaviour and AFAIK .FCStd# behaves the same

sliptonic, 2023-05-23:
Yes. The FCBak method is better.  Better data and more intuitive. 

FCStd# might be doing the same thing but the result is a mess

AjinkyaDahale, 2023-05-23:
> > Also time/date string reflects when the file was previously saved, not when the backup was created, if that makes sense.
> 
> Isn't that better? It's the actual date when this state was saved

If I'm understanding this distinction correctly it does sound like a clever detail.

yorikvanhavre, 2023-05-23:
Also having one unified extension would allow better tricks with file explorers, such as having a special icon, associating them with FreeCAD, or even hiding them automatically, which are hard to achieve with the current system.

AjinkyaDahale, 2023-05-23:
> Regarding custom timestamp, does it matter? The logic should just check if the maximum number of backups is present and delete the oldest one if it is. Then write the new one with whatever template the user has.

Depends on how "oldest" is calculated. If the backups were copied from one place to another the files may have different timestamps than when they were actually edited. Also, if the user decides to use a different timestamp style, how do we tell which files are backups?

sliptonic, 2023-05-23:
> Depends on how "oldest" is calculated.

Probably best not to overthink it.  Just use mtime from the filesystem

Manuelzwei, 2023-05-23:
How is time represented? 
I think I remember around 1900 it was 0. Then every second counts up. 
This number of the current date and time stamp is the same on all operating systems, and is then converted based on the locale and displayed accordingly.
It is certainly advantageous to display the spelling with 'FILE.FCBac_2023_05_21-23_45_59', 'FILE.FCBac_2023_05_21-23_46_00'
This is certainly easier to read than "FILE_46788765432334452" and "FILE_46788765432334453" 

Yes, I wrote the timestamps 21.07.1970, but many years ago (...1982) I wrote "2023_05_21-23_45_59". 
Users [should] learn that this is the [only] correct way to enter the time and date.

AjinkyaDahale, 2023-05-26:
@Manuelzwei I agree on your point that the two methods you showed are correct, and that the date-time format is clearer than the "seconds since X" format. IIRC the Unix epoch starts at 1 Jan 1970, 00:00. I don't know of any counting starting from 1900, but that's still a different question.

AjinkyaDahale, 2023-05-26:
Regarding using a dedicated folder for backups, I'm skeptical that it would provide any additional benefit. Regardless, it seems like a separate issue if you're interested in discussing it further.

adrianinsaval, 2023-05-26:
Subfolders for backups has it's own issue: #6697

Btw, I think I didn't mention this before but wmayer once explained that this odd behaviour of the .FCStd# backups was intended as it is supposed to be more reliable than renaming all backups every time a new one is made.

AjinkyaDahale, 2023-05-26:
@adrianinsaval thanks for pointing out the issue requesting backup folders.

Using FCBak with timestamp gives the best of both worlds, at the cost of looking at larger numbers to compare.

Manuelzwei, 2023-05-28:
> @Manuelzwei I agree on your point that the two methods you showed are correct, and that the date-time format is clearer than the "seconds since X" format. IIRC the Unix epoch starts at 1 Jan 1970, 00:00. I don't know of any counting starting from 1900, but that's still a different question.

Sorry: 1970 and not 1900... Thank you, you are right

Manuelzwei, 2023-05-28:
> 

> Btw, I think I didn't mention this before but wmayer once explained that this odd behaviour of the .FCStd# backups was intended as it is supposed to be more reliable than renaming all backups every time a new one is made.

Yes, renumbering is not good. However, if the backup files have a unique number like 2023_12_21-13_42_58, they don't need to be renumbered. The timestamp with highest number is the newest and so on

adrianinsaval, 2023-05-29:
Yep, that's how the FCBak scheme works and why I suggested just moving to use that as default.

adrianinsaval, 2023-05-29:
I thought about this again and honestly I don't see any value in keeping the .FCStd# scheme around even now, the only thing it has going for it is that it's easier to rename it to .FCStd but this isn't a big advantage either because as trade off you have to change the actual name of the file so it doesn't conflict with the existing .FCStd file. And of course there's a simple way to overcome this disadvantage with the .FCBak scheme, we can change it to .FCStd.old (which communicates even better that this is a backup as some file managers recognize this extension as so)

adrianinsaval, 2023-05-29:
for backwards compatibility we can make the code count the .FCStd# and .FCBak files as existing backups and remove them as new backups are created and the queue is filled.

AjinkyaDahale, 2023-05-30:
> for backwards compatibility we can make the code count the .FCStd# and .FCBak files as existing backups and remove them as new backups are created and the queue is filled.

I think this might be a little unnecessary complication. The worst that happens is that there are more backups than we set up, which is better than less.

AjinkyaDahale, 2023-05-30:
> I thought about this again and honestly I don't see any value in keeping the .FCStd# scheme around even now,...

Agreed.

> And of course there's a simple way to overcome this disadvantage with the .FCBak scheme, we can change it to .FCStd.old (which communicates even better that this is a backup as some file managers recognize this extension as so)

I don't like this alternative since 1) the extension in this case is `.old` which can be whole world of confusion; and 2) this becomes yet another extension for backups in addition to existing `.FCStd#` and `.FCBak`.

I picture that in the future these backups should be recoverable _within_ FreeCAD, possibly with full version control to compare new vs old models. That would begin by automating the necessary renaming, thus removing any efforts in doing so manually.

yorikvanhavre, 2023-05-31:
I would start by simply changing the default to use .FCBak and leave the .FCStd# system in place for now. In a couple of months, when most people will nave switched to the new one, we can retire the old system, if needed.

adrianinsaval, 2023-06-04:
> I think this might be a little unnecessary complication. The worst that happens is that there are more backups than we set up, which is better than less. If a user is interested in having a uniform backup system, we could offer a macro that opens the xml within the FCStd# zip file, extracts last modified time, and renames the files. This macro would be separate from the BackupPolicy stuff.

It's not too much complication, I tried and it already works with mixed FCStd# and FCBak, we would add just one more extension to detect.

> I don't like this alternative since 1) the extension in this case is `.old` which can be whole world of confusion

I don't think it's any more confusing than the current extension, I'd argue it's less confusing.

> and 2) this becomes yet another extension for backups in addition to existing `.FCStd#` and `.FCBak`.

I don't see this as a problem, it's just one new extension and the other two will disappear

closed 2023-07-23 by chennes

ifohancroft, 2023-07-28:
Sorry to comment on a closed issue, but is there a way to use the new extension without a date/time?

AjinkyaDahale, 2023-07-28:
@ifohancroft how many backups have you set?

There should be at most 4-5 backups. If there are more than the value set, this should be fixed.

As to how to remove time stamp, we are moving to discourage this practice. Please see the related issue for the discussion.

ifohancroft, 2023-07-28:
> @ifohancroft how many backups have you set?
> 
> There should be at most 4-5 backups. If there are more than the value set, this should be fixed.
> 
> As to how to remove time stamp, we are moving to discourage this practice. Please see the related issue for the discussion.

Thank you!

P.S. Sorry, what is the related issue as I couldn't find it?

AjinkyaDahale, 2023-07-28:
My mistake I thought I was typing in the PR. This is the very issue I was talking about!

PR #9668 (2023-07-23): [Core] Fix ordering confusion in backups
Fixes #9635.