FreeCAD/FreeCAD #9636: [Feature Request]

grd, 2023-05-22:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
Smooth edges would be very welcome. Right now there exist only hard edges in the view. My idea would be to introduce two buttons (I hope for four but I think that the other two would be impossible). The two buttons are shaded and shaded with edges. The two other buttons would be wire frame and hidden edges. The position of the buttons I don't know yet.
```

### Subproject(s) affected?

Core

### Problem description

"the 3d view is cluttered with smooth edges that are often not relevant"

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

maxwxyz, 2023-05-23:
Similar to: https://github.com/FreeCAD/FreeCAD/issues/5583
Is there a possibility to display the parts in a shaded view for all rounded edges and only display the hard edges with a line style? This would result in a shaded view combined with edges but without smooth edges.
[image]
Especially for complex cast parts like housings or when working with surfaces it is helpful to identify hard vs smooth edges. It allows to quickly see hard edges for e.g. a spotface or helps to see the continuity of a surface. Also, in complex parts, too many edges are more disturbing than helping but often you need to select or use the sharp edges for new features or as references.

In CATIA and NX, you can use a specific composite rendering style to shade the surfaces of an object with its edges, except the smooth ones. A smooth edge is an edge that does not break the continuity of the geometry, contrary to a sharp edge.
Different render styles are shown [here](https://www.3ds.com/fileadmin/3dxml/dircopy_official/PlrUserMap/plr-t-Player-RenderingStyles.htm)

When using surfacing or creating complex cast parts in Part Design which are set up with multiple positive and negative bodies (boolean) which result in a complex geometry this view / render style is very handy as it hides all (unnecessary) smooth edges and displays only the sharp / hard edges.
This is a not so complex casting model (housing) and even there the too many lines around all fillets are obstructing. Usually all negative parts for a e.g. a core are combined bodies which are then filleted and afterwards removed from the positive bodies. 
[image]

adrianinsaval, 2023-05-23:
ahh I forgot that request existed, it's essentially a duplicate, should we close the other one and keep this one or the other way around? I guess the only difference is that the other wants to have the smooth edges displayed but thinner/lighter than hard edges, that could be implemented as a preference for the display style.

grd, 2023-05-23:
Adrian, you are completely right, why didn't you show us that before? Yet you were talking about how hard it is to make this happen. But to me there is still one issue open and that is hidden lines, like this:

[image]

And the issue of Max is still relevant.

adrianinsaval, 2023-05-23:
> why didn't you show us that before?

I spoke of this... https://forum.freecad.org/viewtopic.php?p=682650#p682650

> Yet you were talking about how hard it is to make this happen.

That is not what I said, I said we don't know how to efficiently detect the smooth edges right now, that it was likely possible but not using HLR directly. I said the UI was already there and what is missing is the actual functionality to show or not smooth edges. From the get go I've been trying to get people focused on the actual missing functionality and how to implement it.

grd, 2023-05-23:
Never mind, we both said things.

maxwxyz, 2025-06-04:
As mentioned on Discord, it would be great if the smooth edges would be still selectable - so maybe drawn pretty thin and only in light color / with transparency overlayed.

pieterhijma, 2026-02-21:
@pierreporte, thanks for pinging me on this. I believe seam lines are a necessity in BReps, so we can't go without it. However, I do agree that we don't need to have them in the 3D view. I assume you are referring to this? Otherwise, please let me know how I can address this in any way from the perspective of OCC (other than reporting seam issues).

ElementW, 2026-09-13:
This is a purely drawing thing, where CAD suites supporting this mode draw objects as smooth without edges, then, depending on what exact style we're talking about, either:
* draw the lines with a shader checking for adjacent surfaces: if either side of the line has the same depth and normal vector (within some tolerance), skip drawing the line at that point
  * OCCT's implementation details like ring seams are reduced to zero influence by this
* use an edge detection shader in screen-space to add the lines back
  * adds visual contrast everywhere but creates lines that do not correspond to *wires* at the horizon of curved surfaces

TechDraw's solution does not apply here: its requirements are vector output, but 3D view has pixel output and requires high processing speed. Doing line removal on the CPU is non-interactive levels of slow on any moderately complex part, while GPU rendering tricks are fast and readily available here.

That thing needs to happen in our Coin render paths. Paging @tritao for if screen-space post-processing is planned at some point once the Vulkan rework is stable.

[^wires]: Unless they're without an attached surface, so splines and whatnot still display.

tritao, 2026-09-13:
Planned yes, see https://github.com/FreeCAD/FPA-grant-proposals/issues/107, depending if it's funded or not.

Once we have the infrastructure in place, should be fairly doable to prototype this.