FreeCAD/FreeCAD #9639: [Problem] macOS: BackSpace Key (⌫, ERASE TO THE LEFT) is not recognised to set as shortcut

dirkolbrich, 2023-05-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * ProDarkThemePreferencePack 1.0.0
  * BIM 2021.12.0

---

OS: macOS 13.3
Word size of FreeCAD: 64-bit
Version: 0.21.0.31483 (Git)
Build type: Release
Branch: master
Hash: 588620fd9fd138892b1cd2466f5c87d441493174
Python 3.11.0, Qt 5.15.6, Coin 4.0.0, Vtk 9.2.2, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * ProDarkThemePreferencePack 1.0.0
  * BIM 2021.12.0
```

### Subproject(s) affected?

Core

### Problem description

This is a regression bug for v0.21, as the intended behaviour is possible in v0.20.

See the bug description in the German forum section [[Löschtaste beim Mac](https://forum.freecad.org/viewtopic.php?t=78467)](https://forum.freecad.org/viewtopic.php?t=78467).

Description:
The default setting for `Std_Delete` is the ⌦, ERASE TO THE RIGHT key. To change the behaviour to a more macOS like setting, one can change the assigned key to the `BackSpace`(⌫, ERASE TO THE LEFT) key.

workflow for v0.20: 
menubar "Tools"
-> selection "Customize"
-> a new window "Customize" opens
-> tab "Keyboard"
-> category "Edit"
-> command "&Delete"
-> on the right side within the textfield "Press new shortcut:" press the "BackSpace" (⌫,ERASE TO THE LEFT) key
-> assign the new key with the button "Assign"

[image]

for v.021 the workflow is nearly the same, the "Customize" window does look a bit different.
[BUG] - with assignment of the "BackSpace" key, the key is not recognized or saved, the list on the left side of the window only shows a question mark.

[image]

### Anything else?

I suspect commit https://github.com/FreeCAD/FreeCAD/commit/66358a8e38fe73153b42d8bcec80aa7df6a7028e at [line 409](https://github.com/FreeCAD/FreeCAD/blame/aa3b2f39a6bbd473b3ec9022ed40d8f071ccbaee/src/Gui/Widgets.cpp#L409) as reason for the regression, but this need investigation.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

luzpaz, 2023-06-01:
@0penBrain is this something you could fix ?

0penBrain, 2023-06-03:
> @0penBrain is this something you could fix ?

I don't think so, there seems to be long standing issues in Qt+MacOS. See https://bugreports.qt.io/browse/QTBUG-70940 and related bug there.
I perfectly works here so I'm totally unable to reproduce.

dirkolbrich, 2023-06-03:
What does "perfectly works here" mean? Which system and configuration are you running?

The mentioned [QTBUG](https://bugreports.qt.io/browse/QTBUG-70940) is older than FC v0.20, but the intended behaviour works correctly in v0.20 as shown above, so this QTBUG is not related to this issue.

berniev, 2023-06-04:
Issue confirmed per OP (most recent weekly build).
```
OS: macOS 12.6.6
Word size of FreeCAD: 64-bit
Version: 0.21.0.33345 (Git)
Build type: Release
Branch: master
Hash: a3c4f4bb01f5201daf666852444a8f9c5adbda8d
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: C/Default (C)
```

And the UI/UX is not intuitive.

Unfortunately my own build is down.

0penBrain, 2023-06-09:
> What does "perfectly works here" mean? Which system and configuration are you running?

"Perfectly works" means ... hmmm perfectly works. I can assign and use Backspace key as a shortcut.

> The mentioned [QTBUG](https://bugreports.qt.io/browse/QTBUG-70940) is older than FC v0.20.2, but the intended behaviour works correctly in v0.20.2 as shown above, so this QTBUG is not related to this issue.

It may have been (voluntarily or inadvertently) solved in Qt 5.15.4 and then reintroduced in Qt 5.15.6.
We need someone that can compile on MacOS.

dirkolbrich, 2023-06-10:
Thanks, but you just answered with a classic "it works on my machine"™. Can you please share the FC version and the type of system you are using. From your answer I assume you are not using a macOS (Intel or M1) system.

0penBrain, 2023-06-10:
> Thanks, but you just answered with a classic "it works on my machine"™. Can you please share the FC version and the type of system you are using. From your answer I assume you are not using a macOS (Intel or M1) system.

Sure. Build with which I tested is gone but here the current one.

[code]
OS: Ubuntu 22.04.2 LTS (KDE/plasma)
Word size of FreeCAD: 64-bit
Version: 0.21.0.33383 +4 (Git)
Build type: Debug
Branch: MyBranch
Hash: 99879f6edee28710f490e92535a12d2667f56ed1
Python 3.10.6, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.5.2
Locale: English/United States (en_US)
[/code]

rtm7777, 2023-07-31:
It works on Ubuntu, but on Mac I can't change it

luzpaz, 2023-08-10:
@chennes can I add this to the 1.0 milestone ?

luzpaz, 2023-08-17:
Anyone want to tackle this issue ?

dirkolbrich, 2023-08-18:
I did some more investigation with a more recent dev version.

```
OS: macOS 13.4.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.33848 (Git)
Build type: Release
Branch: master
Hash: 312975edba4e5af0ca1348bfdcf3a71567c52c1b
Python 3.10.12, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.5, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * ProDarkThemePreferencePack 1.0.0
  * BIM 2021.12.0
```

The logic for the GUI window seems to work as intended, as the symbol "BackSpace" (⌫,ERASE TO THE LEFT) is recognised in the window itself and correctly displayed. But after closing and reopening the Shortcut window, the ⌫ symbol is not recognised any more.

1. open "Tools" -> "Customize", keyboard shortcut for Std_Delete is set to ⌫ (ERASE TO THE LEFT) but not recognized in the UI
2. typing new shortcut field "New shortcut" recognises ⌫ (ERASE TO THE LEFT) as the symbol is shown in the UI
3. exiting the field "New shortcut", the ⌫ (ERASE TO THE LEFT) symbol added to the "Current shortcut" field -> the UI logic of this window seems to work as intended
4. Closing and reopening the window "Tools" -> "Customize" shows that the symbol ⌫ is not saved/recognizes internally

[image]

dirkolbrich, 2023-08-18:
Oh, that sounds serious. Haven't tested other shortcuts before. But can confirm this issue.
1. open "Tools" -> "Customize"
2. directly first entry for Std_Export type into field "New shortcut" the combination Ctrl+L
3. after living this field under the field "Current shortcut" the combination Ctrl+L briefly appears but than changes to "none" -> this new shortcut is not recognised or instantly deleted
4. after closing and reopening the "Customize" window the shortcut for Std_Export is empty

luzpaz, 2023-09-09:
CC @PLChris (Can you test this?)

sk7fast, 2023-09-17:
Same issue.

Lol but for now I just put it on "x".
like blender, lol.

luzpaz, 2023-11-15:
@JohnOCFII can you reproduce on latest macOS dev build ?

rtm7777, 2023-11-29:
[image]
I give it a try with this version and the issue still present

rtm7777, 2024-06-10:
Just mentioning that in the recent build (37668) still not working

theosib, 2024-07-17:
Still a problem in dev 38231.

luzpaz, 2024-08-18:
@dirkolbrich can you possible test the fix in #15895 ?

closed 2024-08-18 by chennes

rtm7777, 2024-09-06:
I'm not sure it's related to this issue, but we still have a place where shortcut it's working
I'm using 38622 build on Mac M2
After I changed the shortcut everything is fine
But if I add some filler/chamfer to some edges and later decide to remove some of them but via `delete` button it's not respecting new shortcut
[image]
Can be deleted only with default one

luzpaz, 2024-09-06:
@rtm7777 can you add your full [About](https://wiki.freecad.org/About) info pls?

rtm7777, 2024-09-06:
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38622 (Git)
Build type: Release
Branch: main
Hash: ed97f3c4591303ee3b54906768f8f4eba5211265
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * freecad.gears 1.2.0
  * QuickMeasure 2022.10.28
  * fasteners 0.5.27
  * Curves 0.6.47

luzpaz, 2024-09-06:
@johnocfii this weekend if you have an extra moment, could you test this usecase https://github.com/FreeCAD/FreeCAD/issues/9639#issuecomment-2334647003 ?

JohnOCFII, 2024-09-07:
> @JohnOCFII this weekend if you have an extra moment, could you test this usecase [#9639 (comment)](https://github.com/FreeCAD/FreeCAD/issues/9639#issuecomment-2334647003) ?

Tested this just now.  I was a bit unclear on the note from @rtm7777 -- I can use the `Delete` key to remove items from the Tree listing, but if the issue is not being able to use the `Delete` key for the `Cancel` button on this panel -- @rtm7777 is correct. I don't think there are _any_ keybindings associated with those two buttons. Nothing I pressed (Delete, Function delete, etc.) caused the Cancel function to fire.

[image]

```
OS: macOS 14.6.1
Word size of FreeCAD: 64-bit
Version: 1.0.0RC1.38643 (Git)
Build type: Release
Branch: main
Hash: fe40748c10503eed367840239db3e0f81873e359
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
Installed mods: 
  * FeedsAndSpeeds 0.5.0
  * OpenTheme 2024.9.1
  * fasteners 0.5.28
  * Curves 0.6.47
```

rtm7777, 2024-09-07:
I'll try to make it more clear, when you edit some fillet you have a list of selected edges there, and if you highlight any of them and press delete (new shortcut) button nothing is happening. But if I use default one it's working.

hyarion, 2024-09-20:
@rtm7777 I have the same issue.

The reason why this is the case is that the view sets up the shortcut manually and hard codes it to `QKeySequence::Delete`.
Maybe we should create a new issue for this?

PR #15895 (2024-08-18): Use portable text to set shortcut
This should resolve #9639.

At least on macOS, it seems that setting nontrivial shortcuts only works when the shortcut is converted to `PortableText`, rather than `NativeText`.

Before, it was impossible to map, eg, the backspace key or CMD-T to an action. Now this appears to work.

Happy to make any changes to this PR if needed :)