FreeCAD/FreeCAD #9641: [Problem] Recompute resets placement angle

FEA-eng, 2023-05-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33263 (Git)
Build type: Release
Branch: master
Hash: 864f99bf984c237ef38d9451fd6f3a5ec34e6eb4
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: Polish/Poland (pl_PL)
```

### Subproject(s) affected?

Core

### Problem description

Recompute caused e.g. by inserting a new Part container, resets the placement angle of a part (the angle was defined using a formula), causing the gear to be incorrectly oriented.

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-05-23:
please include full reproduction steps and example file here on the issue, our contribution process says:

> Links to the forum or other resources are permitted but the issue SHOULD be complete and accurate and SHOULD NOT require the reader to visit the forum or any other platform to understand what is being described.

FEA-eng, 2023-05-24:
> please include full reproduction steps and example file here on the issue, our contribution process says:

Done. But in such a case it would be nice to add support for the .FCStd file format here, if possible.

Roy-043, 2023-10-05:
This is a recompute issue. The smaller gear has a rotation angle of 13,85°, but is initially displayed as if the angle is 0°. Recomputing the model solves this. The model is then correctly displayed, but the gears no longer mesh. This is simply because the rotation angle of 13,85° is incorrect, it should be 0°.

Closing this.

closed 2023-10-05 by Roy-043