FreeCAD/FreeCAD #9642: [Problem] GPU is not being used in FreeCAD weekly conda Windows releases.

ftdt10, 2023-05-23:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
[code]
OS: Windows 10 build 19045
Word size of FreeCAD: 64-bit
Version: 0.21.0.33263 (Git)
Build type: Release
Branch: master
Hash: 864f99bf984c237ef38d9451fd6f3a5ec34e6eb4
Python 3.10.11, Qt 5.15.8, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: English/United States (en_US)
Installed mods: 
  * Assembly3 0.12.0
  * Curves 0.6.9
  * kicadStepUpMod 10.19.1
[/code]
```

### Subproject(s) affected?

None

### Problem description

FreeCAD weekly builds conda Windows version doesn't use GPU in Windows.
In Task Manager, I confirmed that GPU is not being used, only the CPU is being used.
So, FPS is very slow in FreeCAD weekly release. For reference, it's normal on Ubuntu.  

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

ftdt10, 2023-05-23:
The black screen problem happens with old HD5500 iGPU.
I hope it works for the iGPU also.

adrianinsaval, 2023-05-23:
Then not using the GPU is expected, I will close this issue since the black screen problem is something separate so it should go to it's own issue (although I don't think there's anything we can do about it since it's a driver problem)

closed 2023-05-23 by adrianinsaval