FreeCAD/FreeCAD #9651: [Problem] Shape.transformGeometry(matrix) fails on OCC 7.7.x

Jolbas, 2023-05-24:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.21 (Development)

### Full version info

```shell
OS: macOS 12.6.3
Word size of FreeCAD: 64-bit
Version: 0.21.0.33263 (Git)
Build type: Release
Branch: master
Hash: 864f99bf984c237ef38d9451fd6f3a5ec34e6eb4
Python 3.11.3, Qt 5.15.8, Coin 4.0.0, Vtk 9.2.2, OCC 7.7.1
Locale: C/Default (C)
Installed mods: 
  * kugghjul.py
  * dxf-library
  * fcgear 1.0.0
  * __pycache__
  * sheetmetal 0.2.61
  * tida.py
  * Curves 0.6.8
  * Assembly4 0.12.7
```

### Subproject(s) affected?

None

### Problem description

Running this python code:

```
obj = App.ActiveDocument.addObject("Part::Box","Box")
obj.recompute()
obj.Shape.transformGeometry(App.Matrix())
```
Gives this error in python console but nothing in report view:
`<class 'Part.OCCError'>: gp_GTrsf::Trsf() - non-orthogonal GTrsf`

The error originates from https://github.com/Open-Cascade-SAS/OCCT/blob/5d8b1a4076398556095c71c093ca517cde77d4d8/src/gp/gp_GTrsf.hxx#L446 in OCCT, which is called from https://github.com/Open-Cascade-SAS/OCCT/blob/5d8b1a4076398556095c71c093ca517cde77d4d8/src/BRepTools/BRepTools_GTrsfModification.cxx#L314 
It seems like an OCCT problem but I'm not sure how to proceed. From other posts in forum here it seems to only affect OCCT 7.7

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

adrianinsaval, 2023-05-24:
where in our code are we calling this? perhaps there's new parameters that we need to pass or we need to move to a different occt function.
@FreeCAD/maintainers shall we target this for 0.21? I vote yes, even if we stick to 7.6 for the release it would be good to have it compatible with 7.7 IMO

Jolbas, 2023-05-24:
We hav it in our code here: https://github.com/FreeCAD/FreeCAD/blob/60d99ca57012bbdca93101feafe1b45372d81f01/src/Mod/Part/App/TopoShape.cpp#LL3070C9-L3070C9

Jolbas, 2023-05-24:
Forum links
https://forum.freecad.org/viewtopic.php?t=77612
https://forum.freecad.org/viewtopic.php?t=78527
https://forum.freecad.org/viewtopic.php?p=682210#p682210

adrianinsaval, 2023-05-24:
maybe we need to call mat.setForm() before making the transformation?

Jolbas, 2023-05-24:
it doesn't help. It is changed back to non-orthogonal later in OCC. Also it has been and should be able to handle non-orthogonal matrices too.

adrianinsaval, 2023-05-24:
The line to throw an error if it was `gp_Other` was already there in 7.6, something must have changed in how the matrix is treated. Might have been introduced by https://github.com/Open-Cascade-SAS/OCCT/commit/33c8a72788661285860902f261062e39f32a2d46

it also looks to me like `mat` is not used directly but rather the values are copied over to a new `gp_GTrsf` so calling `setForm()` does noting anyway. I still don't quite get what's the sequence followed for the transformation but I'm only looking at this through the web interface now.

Jolbas, 2023-05-24:
If I run the following c++ code I don't get the error and the `gp_GTrsf.Form` does not change from `gp_Identity` to `gp_Other`:
```
    BRepPrimAPI_MakeBox aBox(10.0, 10.0, 10.0);
    TopoDS_Shape aShape = aBox.Shape();
    gp_GTrsf m;
    BRepBuilderAPI_GTransform aTransform(mat);
    aTransform.Perform(aShape, false);
```
So ther seem to be something in `_Shape` that is different.

adrianinsaval, 2023-05-24:
there's a couple of differences, we don't pass an identity matrix we use `setValue` many times on it, the first call already changes `Form` to `gp_Other` if I'm not mistaken https://github.com/FreeCAD/FreeCAD/blob/60d99ca57012bbdca93101feafe1b45372d81f01/src/Mod/Part/App/TopoShape.cpp#L3055-L3067

then we directly do
https://github.com/FreeCAD/FreeCAD/blob/60d99ca57012bbdca93101feafe1b45372d81f01/src/Mod/Part/App/TopoShape.cpp#L3070-L3071

we don't use `Perform`

Jolbas, 2023-05-24:
```
BRepBuilderAPI_GTransform aTransform(mat);
aTransform.Perform(aShape, false);
```
is equivalent to
```
BRepBuilderAPI_GTransform aTransform(aShape, mat, false);
```

I don't get the error for this code directly after `mat.SetValue(2,4,rclTrf[1][3]);
```
BRepPrimAPI_MakeBox aBox(10.0, 10.0, 10.0);
TopoDS_Shape aShape = aBox.Shape();
BRepBuilderAPI_GTransform aTransform(aShape, mat, false);
```
With this shape the execution never reaches the function where the error is thrown. It returns before because the triangulation is null. I don't understand what that means.
https://github.com/Open-Cascade-SAS/OCCT/blob/5d8b1a4076398556095c71c093ca517cde77d4d8/src/BRepTools/BRepTools_GTrsfModification.cxx#L278

Jolbas, 2023-05-24:
And the error is still thrown if I remove every `mat.setValue(...)` line.

adrianinsaval, 2023-05-24:
maybe it's related to tessellation?

adrianinsaval, 2023-05-24:
yeah it looks like it's about tessellation https://dev.opencascade.org/doc/refman/html/class_poly___triangulation.html#details

Jolbas, 2023-05-24:
In this code. the first call to `transformGeometry()` works but the second call fails
So the recompute on adocument object somehow creates tesselation?
```
boxshape = Part.makeBox(1,1,1)
transformedBoxshape = boxshape.transformGeometry(App.Matrix())
boxobj = App.ActiveDocument.addObject("Part::Feature", "Box")
boxobj.Shape = boxshape
boxobj.recompute()
transformedBoxshape2 = boxobj.Shape.transformGeometry(App.Matrix())
```

Jolbas, 2023-05-24:
The recompute isn't necessary. The assignment is enough

```
boxshape = Part.makeBox(1,1,1)
transformedBoxshape = boxshape.transformGeometry(App.Matrix())
# no error
boxobj = App.ActiveDocument.addObject("Part::Feature", "Box")
boxobj.Shape = boxshape
transformedBoxshape2 = boxobj.Shape.transformGeometry(App.Matrix())
# error
```

Jolbas, 2023-05-25:
This means we may work around this by cleaning the shape before transforming!

```
shape_a = Part.makeBox(1,1,1)
transformed_a = shape_a.transformGeometry(App.Matrix())
# no error
part_feature = App.ActiveDocument.addObject("Part::Feature", "Box")
part_feature.Shape = shape_a
shape_b = part_feature.Shape
transformed_b = shape_b.transformGeometry(App.Matrix())
# error
shape_c = shape_b.cleaned()
transformed_b = shape_c.transformGeometry(App.Matrix())
# no error!
```

adrianinsaval, 2023-05-25:
We could add a workaround for occt >=7.7 for the moment, we might need to submit a bug report upstream and if possible a patch to fix it

Jolbas, 2023-05-25:
A fix for OCCT may be to replace the linked lines with a simple
```
    theTriangulation->ComputeNormals();
```

https://github.com/Open-Cascade-SAS/OCCT/blob/5d8b1a4076398556095c71c093ca517cde77d4d8/src/BRepTools/BRepTools_TrsfModification.cxx#L174-L179

But I'm not sure at all that won't have side effects

adrianinsaval, 2024-01-08:
@bgbsww this is another occt 7.7 bug

bgbsww, 2024-01-08:
Looks to be fixed in 7.8.0  @adrianinsaval 

>>> obj = App.ActiveDocument.addObject("Part::Box","Box")
>>> obj.recompute()
True
>>> obj.Shape.transformGeometry(App.Matrix())
<Shape object at 0x55f19b6636f0>
>>> 
```
OS: Ubuntu 22.04.3 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.35500 (Git)
Build type: Debug
Branch: bgbsww-occt7.8.0
Hash: 2a6fa0d9832920dd50d8f661e639d14c4eff9a04
Python 3.10.12, Qt 5.15.3, Coin 4.0.0, Vtk 7.1.1, OCC 7.8.0
Locale: English/United States (en_US)
Installed mods: 
  * Curves 0.6.21
```

adrianinsaval, 2024-01-08:
Awesome, then once the 7.8 PR is merged is probably a good idea to skip 7.7 and move our distributions to that

adrianinsaval, 2024-03-03:
potentially fixed by https://github.com/Open-Cascade-SAS/OCCT/commit/bcfc5f0d5a9ea038d6e3bf93517e87d8f47a9b48

furgo16, 2024-05-02:
Reportedly this is the underlying issue for https://github.com/FreeCAD/FreeCAD/issues/13750 as well. Does it make sense to start shipping OCC 7.7.2 with the backported fix?

adrianinsaval, 2024-05-06:
do you have the same problem with the appimage? we are using 7.7.2 there already

furgo16, 2024-05-07:
@adrianinsaval thanks. Unfortunately, it does not solve the issue. I've used:

```
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.37213 (Git) AppImage
Build type: Release
Branch: main
Hash: 20e7deb86a8c6c2cd2378f09f8313760933f3a5c
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Installed mods: 
  * OpenTheme 2024.5.3
  * BIM 2021.12.0
  * parts_library
```

Does it mean https://github.com/FreeCAD/FreeCAD/issues/13750 is not a duplicate of this issue?

adrianinsaval, 2024-05-07:
I just tried this again and it looks like the issue is not actually fixed in occt 7.7.2

furgo16, 2024-05-07:
Bummer. Thanks for checking, though. I'm not familiar with OCC, and doing a quick search just returned this: https://dev.opencascade.org/node/84296#comment-25214, but I could not find any reference to the specific upstream commit or PR that addresses this.

furgo16, 2024-07-01:
Can this issue be considered as a target for [1.0](https://github.com/FreeCAD/FreeCAD/labels/1.0), by either upgrading OCCT to 7.8.2 or downgrading it to 7.6.3?

Jolbas, 2024-07-02:
> I just tried this again and it looks like the issue is not actually fixed in occt 7.7.2

I tested this and test case in issue #13750. And I get no error anymore. So are there differences between OCC 7.7.2 for Mac and OCC 7.7.2 for Linux? Does it work on Windows?

```
OS: macOS 14.5
Version: 0.22.0dev.37464 (Git)
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Locale: C/Default (C)
```

adrianinsaval, 2024-07-14:
hmm, testing now it seems to work on linux with 7.7.2 so I don't know what the hell I was testing before

furgo16, 2024-07-14:
> hmm, testing now it seems to work on linux with 7.7.2 so I don't know what the hell I was testing before

@adrianinsaval perhaps there's something else that might have fixed it. I could definitely [reproduce a duplicate issue](https://github.com/FreeCAD/FreeCAD/issues/9651#issuecomment-2097490787). I've tested it now, and I cannot reproduce it. The OCCT version has stayed the same (OCC 7.7.2) in both instances.

> I guess the only thing missing before we can forget about this issue is to update the snap to use occt 7.7.2

Unless this is a typo, why update the daily snap to OCCT 7.7.2 when the [daily PPA is being updated to 7.8.2](https://github.com/FreeCAD/FreeCAD/issues/14531#issuecomment-2227165003)? Would it not make sense to keep them in sync when it comes to OCCT versions?

Thanks.

```
OS: Ubuntu 22.04.4 LTS (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38043 (Git) AppImage
Build type: Release
Branch: main
Hash: 561e521817034ea7660de987d8ab7183ad583595
Python 3.11.9, Qt 5.15.13, Coin 4.0.2, Vtk 9.2.6, OCC 7.7.2
Installed mods: 
  * OpenTheme 2024.7.3
```

adrianinsaval, 2024-07-14:
true, snap should be updated to 7.8.1

furgo16, 2024-07-14:
If I read the steps to reproduce in the original issue description correctly, I cannot reproduce it in OCCT 7.7.1, either.

```python
>>> obj = App.ActiveDocument.addObject("Part::Box","Box")
>>> obj.recompute()
True
>>> obj.Shape.transformGeometry(App.Matrix())
<Shape object at 0x562dede4ce10>
```

```
OS: Ubuntu Core 22 (ubuntu:GNOME/ubuntu)
Word size of FreeCAD: 64-bit
Version: 0.22.0dev.38126 (Git) Snap 1074
Build type: Release
Branch: main
Hash: ac2b65c37b18617101822e5ece06f3c9b21375cd
Python 3.10.12, Qt 5.15.10, Coin 4.0.0, Vtk 7.1.1, OCC 7.7.1
Installed mods: 
  * OpenTheme 2024.7.3
  * Curves 0.6.40
  * ezydxf
  * Ondsel-Lens 2024.7.5.02
  * Assembly4 0.50.15
```

Regarding updating the snap's OCCT version, I looked at this myself a while ago, but I could not figure it out. It seems OCCT is built with the FreeCAD code itself, from whatever version it has at the build location. IIRC last time I built the snap locally, it used the OCCT version from the `-dev` packages I had installed, which were the ones from the daily PPA. When the snap is built on GitHub, I don't know how the build picks up the OCCT version.

https://github.com/FreeCAD/FreeCAD-snap/blob/3f38c962328177628ad69dc84fa0662c9e1c20e9/snap/snapcraft.yaml#L154-L167

closed 2024-09-28 by adrianinsaval