FreeCAD/FreeCAD #9659: [Problem] Arch Workbench Wall Fails

nelsonic, 2023-05-25:
### Is there an existing issue for this?

- [X] I have searched the existing issues

### Version

0.20 (Release)

### Full version info

```shell
OS: macOS 10.16
Word size of FreeCAD: 64-bit
Version: 0.20.2.29603 (Git)
Build type: Release
Branch: (HEAD detached at 0.20.2)
Hash: 930dd9a76203a3260b1e6256c70c1c3cad8c5cb8
Python 3.10.8, Qt 5.15.4, Coin 4.0.0, Vtk 9.1.0, OCC 7.6.3
Locale: C/Default (C)
Installed mods: 
  * BIM 2021.12.0
```

### Subproject(s) affected?

Arch

### Problem description

While attempting to follow @yorikvanhavre's "Presentation of FreeCAD Arch Workbench": https://youtu.be/CH_io4MHJ4U?t=100
and create a basic Wall in the Arch Workbench based on a Rectangle I get the following strange result:
[image]

In case it's easier to see the full sequence of events, I recorded a video of all the steps:
https://youtu.be/2n_JjEguOGw [71 seconds]

Really don't understand what I'm doing wrong and wanted to report it in case it's a bug. 🤷‍♂️ 

Created a forum topic: https://forum.freecad.org/viewtopic.php?t=78586 as it appears to be the faster way to get help. 

### Code of Conduct

- [X] I agree to follow this project's Code of Conduct

yorikvanhavre, 2023-05-25:
Can you attach the file here? (You might need to change the extension to .zip)

nelsonic, 2023-05-25:
Hi @yorikvanhavre thanks for checking this out. I'm literally just taking a `Rectangle` created in the `Arch` Workbench:

[image]

[image]

[image]

The file is on GitHub at: https://github.com/dwyl/learn-cad-cam-cnc/blob/master/freecad/20220524-freecad-arch-learning.FCStd

But also uploaded here as `.zip` for convenience:  [20220524-freecad-arch-learning.FCStd.zip](https://github.com/FreeCAD/FreeCAD/files/11563672/20220524-freecad-arch-learning.FCStd.zip)

Appreciate the help.
If it's just my Mac (M1) being silly, I could try a different computer. 
But given that this is the most recent machine I have with a decent graphics card I wanted to use it for `FreeCAD` so that I can build a reasonably complex/complete architectural model.

nelsonic, 2023-05-26:
On the advice of [**`@hko`**](https://forum.freecad.org/memberlist.php?mode=viewprofile&u=30966) (on the [forum](https://forum.freecad.org/viewtopic.php?p=684122#p684122)), updated the Preferences > Draft > Default Working plane from "None":
[image]

To "XY Top":
[image]

Then in a **`new document`**, created a fresh `Rectangle` with the dimensions I waned `3m x 4m`
and when creating a wall based on the `Rectangle` get the desired result:

[image]

Closing as "PEBCAK". ✅  (thanks for your patience with me ... 🙏)
Not a problem/issue/bug with `FreeCAD` but rather a lack of knowledge/experience of the **working plane**. 🙄 
But wish there was an easy way new `FreeCAD` people avoid this issue ... 💭

closed 2023-05-26 by nelsonic

nelsonic, 2023-05-26:
Follow-up note: In the Wall properties the `Length` is listed as `14.00m`: 

[image]

The `Wall` is based on a `3m x 4m` `Rectangle`: 
[image]

Therefore my assumption is that the "Length" of the wall is in fact the _perimeter_.
i.e. `3 + 4 + 3 + 4 = 14m` ...

yorikvanhavre, 2023-05-26:
Glad you have it sorted. Indeed the working plane is something to be aware of when starting. It is set by default to the XY plane. However, it is in "unattached" mode, which means if you change your view point it will adapt to the current view. I understand this might be confusing.

This is a useful behaviour though, but I'm not sure attaching it to the XY plane by default would make things better.

Thinking out loud now, we could add an additional "smart" auto mode to that working plane:

- if you are in top, front or side view, the plane adapts to your view
- in all other cases, the plane goes to XY

What would you think of that @Roy-043 ?

Roy-043, 2023-05-26:
The defaultWP is actually set to zero. Setting it to 1 would be a good idea IMO. I would not change the behavior of Auto mode.

It would also be good if Arch_Wall were not restricted to the XY plane. It could then be used as a more generic tool.